Copilot-Language/copilot #1: Can't build copilot-core 2.1.2 on GHC 7.10.2

octopuscabbage, 2015-09-12:
When I try to build copilot core I get this error:
https://gist.github.com/octopuscabbage/abf529a426ce8c6de1d2

I think it has to do with Applicative now being a superclass of monad.

agoodloe, 2015-09-12:
thanks we will take a look at the issue and see if we can resolve it as soon as possible

GAJaloyan, 2015-09-12:
Hi, 

I compiled it, and it seems to work modulo some changes that I pushed already.

I compiled it several times from scratch on a Debian wheezy with the following : 

cabal 1.22.6.0
ghc 7.10.2
caml 4.02.2
branch master for all submodules

And it works well everytime.

All installation details can be found there : https://github.com/Copilot-Language/copilot-discussion/tree/master/TutorialAndDevGuide/DevGuide

Could you please detail more about your configuration if it still does not work on your computer ?

Georges-Axel.

PS : Applicative is a superclass of monad since ghc 7.10, and it compiled well on 7.10.1 this summer. This bug may probably have another cause.

chathhorn, 2015-09-14:
Yeah, the version of copilot-core in hackage doesn't build on 7.10. We need to get an updated version into the repo. The version here should work, though.

t3476, 2015-10-20:
I can install it with github's copilot-core but why hasn't repo on hackage been updated?

leepike, 2015-10-20:
@agoodloe @chathhorn is master stable to push to hackage?

agoodloe, 2015-10-20:
We have been using this current master for a while here and I it should be
stable enough.

chathhorn, 2015-10-20:
Geh, the version of smtlib2 that works with GHC 7.10 still isn't in hackage. Hold off for a little bit and let me play with the dependencies.

agoodloe, 2015-10-20:
Good call I've just so used to building it by hand I had overlooked that
one.

GAJaloyan, 2015-10-20:
Do not forget to also change the dependecies for cabal (if not already done yet) and SBV 5.0 which has been released on hackage.

leepike, 2015-10-21:
Ok, it builds and the regression test works on my machine. Are we just waiting for smtlib2 to be updated now? @chathhorn, can you ping the maintainer?

chathhorn, 2015-10-21:
I think it should be alright now, there's a new version of smtlib2 up on hackage that works with 7.10.

Note that copilot-language depends on copilot-theorem, now, though (sorry), so it'll need to be put on hackage too...

leepike, 2015-10-23:
What about removing sbv and smtlib2 as submodules from Copilot? If they're up-to-date on Hackage, we shouldn't need them as submodules anymore (I'd like to keep submodules limited to Copilot packages).

chathhorn, 2015-10-23:
I removed them and stuff seems to still be working.

leepike, 2015-10-23:
Ok, the packages are all updated on Hackage, and I believe they're compatible with both 7.8.3 and 7.10.2. Sorry for the delay!

closed 2015-10-23 by leepike