Copilot-Language/copilot #4: `Stream a` doesn't implement `(RealFrac a) => RealFrac (Stream a)`

YPares, 2016-05-13:
Currently, it seems it is not possible to get either the integer (`truncate`) or decimal part (like `modf` does in C) of a Stream Double, or to truncate, say, a `Stream Float` into a `Stream Int32` with `unsafeCast`.

This makes for instance difficult the implementation of combination of oscillators dealing with floating-point values, like the following (which doesn't work):

```haskell
-- Program outputs analog values within the interval [0, maxAnalogOut]
maxAnalogOut = 255

type Millis = Word32

-- Milliseconds since the program has started
timeMillis :: Stream Millis
timeMillis = externFun "millis" [] (Just simu)
    where simu = [0] ++ (10 + simu)

sawtoothOsc, sinOsc :: Stream Double  -- ^ period (in milliseconds)
                   -> Stream Millis  -- ^ time since beginning
                   -> Stream Double

sawtoothOsc period millis = cast millis `modf` period

sinOsc period millis = sin $ 2*pi * cast millis / period

-- Modulate period of sawtoothOsc from output of sinOsc
intensity = maxAnalogOut * sawtoothOsc (250 * sinOsc 4000 timeMillis + 250) timeMillis
```

(sawtoothOsc could be expressed using only `Stream Word32` but sinOsc couldn't. So we need either be able to use Doubles everywhere or to force cast a `Stream Double` to a `Stream Word32` in order to compose the two oscillators)

YPares, 2016-05-17:
Trying to propose a PR. But there's a restriction on Cast right now:

`Cast     :: (Integral a, Num b) => Type a -> Type b -> Op1 a b`

Either this will have to be removed, or a new cast operator will have to be adder for floating-point values. @leepike do you have an idea?

agoodloe, 2016-05-17:
Yves,

    It's been a while since these design decisions were made, but I think I
remember we decided to disallow some of these unsafe casts, but Lee
probably remembers the details as to why better.

YPares, 2016-05-17:
@agoodloe Actually, I think that the fact that the unsafe casts is pretty sensible.
The problem is that the RealFrac functions are unavailable. (And they seem to be unavailable in Atom too...)

agoodloe, 2016-05-20:
Yves,

    You are correct the back ends are a constraint.

    Gorges-Axel who did a lot of work last summer on Copilot -->SBV sent me
a note reminding me that

>From what I remember, I did not implemented the cast float -> int because
>of the sbv backend.

>We have the promotion from integers to IEEE 974 by using "S.toSDouble
>S.sRoundNearestTiesToEven", but I do not know if there is something
>similar to go the other way. If there is, then it can be (almost) easily
>implemented.

>If there is no way to get it back, the best way remains calling an
>external function that will do the cast in C, but you will loose all the
>model checking power of SBV.

ivanperez-keera, 2019-11-15:
I know that this is old. We are looking into making some small changes and cleaning copilot overall, and we just discussed this.

In my opinion, it cannot be done. The interface of `RealFrac` is too permissive. Recall that:

```
class  (Real a, Fractional a) => RealFrac a  where
    properFraction   :: (Integral b) => a -> (b,a)
    truncate, round  :: (Integral b) => a -> b
    ceiling, floor   :: (Integral b) => a -> b
```

If you allow this, then you could try to `floor` a stream of `Float`s into a stream of `Integer`s, since the `floor` operation is polymorphic on the `b`. But we do not support unbounded ints in Copilot.

But it gets worse: I should be able to floor a stream of `Float`s into one single `Int`, as in:
```haskell
aStream :: Stream Float
aStream = ... --defined in some magical way

myNumber :: Int
myNumber = floor aStream
```

Obviously this does not make sense. `RealFrac` is too permissive. I think what needs to happen is that we support numeric operations to produce code as simple as you need it, but without using the actual `RealFrac` class.

closed 2019-11-15 by ivanperez-keera

ChrisPenner, 2021-05-16:
Hey folks, just checking in here.

> I think what needs to happen is that we support numeric operations to produce code as simple as you need it, but without using the actual RealFrac class.

Is there any movement in this area?

I don't really care about whether it's compatible with the `RealFrac` class or not, but I have some use-cases where I very much need to `floor` a `Stream Float`; and I'm okay with accepting the burden of unsafety it brings with it with regard to overflows 😄