Copilot-Language/copilot #22: Typos in README

ivanperez-keera, 2019-03-31:
There's a few typos in the readme, and the style could be clearer.

ivanperez-keera, 2019-03-31:
Minor question: would it be better to write in the readme:

_> the [Copilot website](https://copilot-language.github.io)._

Instead of writing:

_> the Copilot website: [https://copilot-language.github.io](https://copilot-language.github.io)_

?

ivanperez-keera, 2019-03-31:
@fdedden I merged some changes into master by mistake. Please, feel free to undo.

fdedden, 2019-04-01:
> Minor question: would it be better to write in the readme:
> 
> _> the [Copilot website](https://copilot-language.github.io)._
> 
> Instead of writing:
> 
> _> the Copilot website: https://copilot-language.github.io_
> 
> ?

In general it is better to not show the url. In this case I think we should, so that people will recognise and remember the website.

commit c3d7f001 (2019-03-31): Merge branch 'develop-readme-typos'. Closes #22.

[ci skip]
commit 8e9e9576 (2019-04-01): Merge branch 'develop-readme-style'. Closes #22
closed 2019-04-01 by fdedden