Copilot-Language/copilot #36: Defining Stream (Array 2 Double) from 2 Stream Double

ivanperez-nfm, 2019-07-08:
If I have 2 streams carrying doubles, such as:
```haskell
x :: Stream Double
x = extern "x" Nothing

y :: Stream Double
y = extern "y" Nothing
```
Is it possible to define, with some function (or maybe an `Op2`):
```haskell
xy :: Stream (Array 2 Double)
xy = ??? x y
```

fdedden, 2019-07-09:
Currently there is no way to construct an array from stream carrying scalars. Having a feature like this (and a similar one for structs) would indeed help a lot, and is worth looking into.

ivanperez-keera, 2019-07-10:
I wouldn't know how to go about it. I'm thinking about kinds and arities, but I'm not sure we could make it work.

ivanperez-keera, 2023-07-08:
Update: Someone is working on a feature atm that would allow us to implement this.

ivanperez-keera, 2023-08-06:
Update: @InnovativeInventor has implemented setter for arrays, which allows us to implement this. We'll be discussing integrating this into Copilot in the upcoming future.

ivanperez-keera, 2023-12-12:
Update: similar requests have come up recently related to structs. We are looking into porting the solution that @InnovativeInventor prepared to work for structs and adding both to Copilot. Stay tuned.

ivanperez-keera, 2024-07-07:
For awareness, we have an open issue #520 that we intend to merge today on updating fields of structs.

ivanperez-keera, 2024-07-08:
**Description**

Copilot currently supports reading values from streams of arrays, but not changing them.

We'd like to be able to modify an array by adjusting the value at a specific position.

**Type**

- Feature: new extension to the language.

**Additional context**

None.

**Requester**

- Ivan Perez

**Method to check presence of bug**

Not applicable (not a bug).

**Expected result**

Copilot provides a mechanism to update specific values in arrays in an stream of arrays.

**Desired result**

Copilot provides a mechanism to update specific values in arrays in an stream of arrays.

**Proposed solution**

Extend `copilot-language` to provide operations to modify values of arrays in streams.

Extend `copilot-core`, `copilot-c99` and other libraries accordingly.

**Further notes**

None.

ivanperez-keera, 2024-08-10:
Technical Lead: Issue scheduled for Copilot 4.

Fix assigned to: @ivanperez-keera.

ivanperez-keera, 2024-09-06:
Implementor: Solution implemetned, review requested.

ivanperez-keera, 2024-09-06:
Change Manager: Verified that:
- Solution is implemented:
  - [X] The code proposed compiles and passes all tests. Details:
    Build log: https://app.travis-ci.com/github/Copilot-Language/copilot/builds/272213129

  - [X] The solution proposed produces the expected result. Details:
    The following Dockerfile installs `copilot` and runs a file with struct update operations, which it then links and compares against its expected output, printing the message "Success" if the output matches the expectation:
    ```Dockerfile
    --- Dockerfile
    FROM ubuntu:focal
    
    ENV DEBIAN_FRONTEND=noninteractive
    RUN apt-get update
    
    RUN apt-get install --yes libz-dev
    RUN apt-get install --yes git
    
    RUN apt-get install --yes wget
    RUN mkdir -p $HOME/.ghcup/bin
    RUN wget https://downloads.haskell.org/~ghcup/0.1.19.2/x86_64-linux-ghcup-0.1.19.2 -O $HOME/.ghcup/bin/ghcup
    
    RUN chmod a+x $HOME/.ghcup/bin/ghcup
    ENV PATH=$PATH:/root/.ghcup/bin/
    ENV PATH=$PATH:/root/.cabal/bin/
    RUN apt-get install --yes curl
    RUN apt-get install --yes gcc g++ make libgmp3-dev
    RUN apt-get install --yes pkg-config
    
    SHELL ["/bin/bash", "-c"]
    
    RUN ghcup install ghc 9.4
    RUN ghcup install cabal 3.2
    RUN ghcup set ghc 9.4.8
    RUN cabal update
    
    ADD Arrays.hs /tmp/Arrays.hs
    ADD main.c /tmp/main.c
    ADD expected /tmp/expected
    
    CMD git clone $REPO \
        && cd $NAME \
        && git checkout $COMMIT \
        && cabal v1-sandbox init \
        && cabal v1-install alex happy \
        && cabal v1-install copilot**/ \
        && cabal v1-exec -- runhaskell /tmp/Arrays.hs \
        && mv /tmp/main.c . \
        && gcc main.c arrays.c -o main \
        && ./main > actual \
        && diff actual /tmp/expected \
        && echo Success

    --- Arrays.hs
    {-# LANGUAGE DataKinds         #-}
    {-# LANGUAGE GADTs             #-}
    {-# LANGUAGE NoImplicitPrelude #-}
    module Main (main) where
    
    import Copilot.Compile.C99
    import Data.Type.Equality  as DE
    import Language.Copilot
    
    -- Sample values
    
    arr :: Stream (Array 2 Float)
    arr = [ array [1.0, 2.5]
          , array [2.5, 1.0]
          ]
       ++ arr
    
    arr1 = (arr !! 1 =$ (+1)) !! 0 =$ (+100)
    
    arr2 = arr1 !! 1 =$ ([0.0] ++)
    
    spec :: Spec
    spec = do
      trigger "arrays" true [arg arr2]
    
    main :: IO ()
    main = do
      spec' <- reify spec
      compile "arrays" spec'

    --- main.c
    #include <stdio.h>
    #include <stdlib.h>
    #include <stdint.h>
    #include "arrays.h"
    #include "arrays_types.h"
    
    void arrays( float* arrays_arg0
               ) {
      printf("%f %f\n"
            , arrays_arg0[0]
            , arrays_arg0[1]);
    }
    
    int main() {
      step();
      step();
      step();
      step();
    }

    --- expected
    101.000000 0.000000
    102.500000 3.500000
    101.000000 2.000000
    102.500000 3.500000
    ```
    Command (substitute variables based on new path after merge):
    ```
    $ docker run -e "REPO=https://github.com/ivanperez-keera/copilot" -e "NAME=copilot" -e "COMMIT=191a75f096d5956f25737b354a1279b00a6e2cde" -it copilot-verify-36
    ```
- [X] Implementation is documented. Details:
  New operations and modules are documented.
- [X] Change history is clear.
- [X] Commit messages are clear.
- [X] Changelogs are updated.
- [X] Examples are updated. Details:
  Examples updated and new examples included.
- [X] Required version bumps are evaluated. Details:
  Bump needed (the definitions of both `Op3` have changed; a function has been hidden, a function has been deprecated, and operations have moved modules).

closed 2024-09-06 by ivanperez-keera