Copilot-Language/copilot #38: Copilot installation fails with ghc-8.6.5, cabal-2.4.0.0

JackTheEngineer, 2019-07-20:
Hello ! 
With an Archlinux computer, a fresh `$ rm -rf ~/.ghc`, and `$ rm -rf ~/.cabal`, 
the installations fails with both `$ cabal new-install copilot` and `$ cabal install copilot`, yielding 

```
Data/Bimap.hs:90:1: error:
    Could not find module ‘Control.Monad.Catch’
    There are files missing in the ‘exceptions-0.10.2’ package,
    try running 'ghc-pkg check'.
    Use -v to see a list of the files searched for.
   |
90 | import           Control.Monad.Catch
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
```
Other seemingly basic libraries aren't found either. 

Another rather generic cabal issue, which might correlate with this one, 
is that after the cabal installation, cabal trew the error:
```
$ cabal install copilot 
...
...
cabal: ghc-pkg dump failed: dieVerbatim: user error (cabal: '/bin/ghc-pkg'
exited with an error:
ghc-pkg: /home/jakov/.cabal/store/ghc-8.6.5/package.db:
getDirectoryContents:openDirStream: does not exist (No such file or directory)
)
```
which i fixed with: 
`cabal new-repl -w ghc-8.6.5 --build-dep fail`

I am not proficient with cabal package management, so simple tricks might also help. 
Thank you very much in advance, 
Jakov

fdedden, 2019-07-21:
Hi Jakov,

Thank you for opening an issue, in addition to your email, as well.

It seems that your problems are not Copilot related, but more specific to your GHC installation. I suspect you installed GHC and cabal-install through pacman? This provides you with a very basic installation: it does not contain the `exceptions` package (that provides `Control.Monad.Catch`) for example. Can you verify that `haskell-exceptions` is _not_ installed through pacman?

Surprisingly `exceptions` is a dependency of `bimap`, so it should be build correctly, unless it is already (incorrectly) installed, possibly by pacman. I assume that after deleting `~/.ghc` and `~/.cabal`, you did a `cabal update` as well?

In general I can recommend to just install GHC and cabal-install by the distro's package manager, and let cabal handle the rest of the packages. Please make sure that no other Haskell packages are installed through pacman, to avoid incompatibilities between versions installed by pacman and cabal-install. Cabal should be able to find and install the correct versions itself.

Please let me know if this helps for you.

Cheers,
Frank

References:
1. https://www.archlinux.org/packages/community/x86_64/ghc/
2. https://www.archlinux.org/packages/community/x86_64/cabal-install/
3. https://www.archlinux.org/packages/community/x86_64/haskell-exceptions/

JackTheEngineer, 2019-08-11:
Hi, 
Thank you for your quick response !
Oddly, when i do 
```
sudo pacman -S cabal-install       
resolving dependencies...
looking for conflicting packages...

Packages (22) haskell-async-2.2.2-1  haskell-base16-bytestring-0.1.1.6-14
              haskell-base64-bytestring-1.0.0.2-15  haskell-cryptohash-sha256-0.11.101.0-9
              haskell-digest-0.0.1.2-14  haskell-echo-0.1.3-11  haskell-ed25519-0.0.5.0-13
              haskell-edit-distance-0.2.2.1-14 haskell-exceptions-0.10.2-11
              haskell-hackage-security-0.5.3.0-56  haskell-hashable-1.2.7.0-8
              haskell-http-4000.3.14-18  haskell-network-2.8.0.0-19
              haskell-network-uri-2.6.1.0-18  haskell-random-1.1-17
              haskell-resolv-0.1.1.1-36  haskell-tar-0.5.1.0-8  haskell-temporary-1.3-37
              haskell-transformers-compat-0.6.5-1  haskell-zip-archive-0.4.1-13
              haskell-zlib-0.6.2-8  cabal-install-2.4.0.0-122
```
It still installs haskell-exceptions, even though it's not listed as a dependency on the archlinux website. I'll try to contact the maintainer of the package. 

Best regards, 
Jakov

fdedden, 2019-08-16:
I guess `haskell-exceptions` is a sub-dependency of `cabal-install`. Please let me know your findings.

fdedden, 2019-09-28:
No response for over 1.5 months, so I assume the issue has been solved. Please let me know if that's not the case.

closed 2019-09-28 by fdedden