Copilot-Language/copilot #47: Add newline character to the end of c and header files to remove c99 compiler warning

mattjuliec, 2019-11-26:
Generated code compiles with zero warnings if a newline character is added to generated c and header files

fdedden, 2019-11-27:
Hi Matt,
Copilot doesn't generate a newline at the end of the file, and this was never a problem for us. We will write a fix for this in the pretty printer of the code nonetheless.

Just to get an idea: which compiler and operating system are you using? Linux / MacOS + GCC never gave us any issues, but I can imagine other compilers do.

mattjuliec, 2019-11-27:
Frank, I am using MacOS + c99 (xcode gcc compiler). Not sure if this is a
broad problem. Just noticed it on my machine
Thanks!
Matt

fdedden, 2019-11-30:
The fix to this problem is not in Copilot, but rather the underlying `language-c99` we created. I have just fixed the issue and uploaded language-c99-0.1.2 to hackage (https://hackage.haskell.org/package/language-c99-0.1.2).

Assuming you have installed copilot through cabal globally, a simple `cabal new-install --upgrade-dependencies copilot` should force the new version of language-c99. Please let me know if you have any issues with this.

With regard to the occurrence of your problem: as far as I know, Xcode nowadays makes use of the Clang compiler. We mainly test using GCC, and this is probably the reason why we never had this problem before.

Please let me know if this fixes your issue, or if there are any other issues.

mattjuliec, 2019-11-30:
Thanks! I’ll try it

fdedden, 2019-12-13:
Have you been able to try the fix? Please let me know if there are any problems.

closed 2020-02-10 by fdedden