Copilot-Language/copilot #52: C99 backend crashes on constant struct args

wyager, 2019-11-29:
```haskell
module Lib where

import Language.Copilot as C
import Copilot.Compile.C99
import qualified Prelude as P

data W = W {w :: Field "w" Bool}
mkW :: Bool -> W
mkW = W . Field
instance Struct W where
    typename _ = "w"
    toValues x = [Value Bool (w x)]
instance Typed W where
    typeOf = Struct (mkW False)

ws1 :: Stream W
ws1 = [mkW True] ++ ws1

ws2 :: Stream W
ws2 = constant (mkW True)

spec = trigger "w" (constant True) [arg ws2]

go = reify spec >>= compile "test"
```

yields

```
*Main Lib C C99> go
*** Exception: src/Copilot/Compile/C99/Translate.hs:(113,14)-(124,23): Non-exhaustive patterns in case
```

`ws1` is fine.

fdedden, 2019-11-29:
Thank you for posting this bug, indeed constant arrays should work as expected.
This is a known issue to us
(https://github.com/Copilot-Language/copilot-c99/issues/13), but we decided to
focus on other issues and changes for the recent 3.1 release.

We are planning to fix this for the next release.

wyager, 2019-11-30:
For both this (constant struct) and the constant array problem you linked, could you emit a compound literal?

fdedden, 2019-12-23:
I have fixed the bug (Copilot-Language/copilot-c99#13), and released it as `copilot-c99-3.1.1`. After a updating cabals package list, you can fetch the newest dependencies to Copilot using (assuming a global install):
```
cabal install --upgrade-dependencies copilot
```
Please let me know if this works for you.

closed 2020-02-10 by fdedden