Copilot-Language/copilot #53: More modular C99 backend?

wyager, 2019-11-30:
Right now the C99 backend emits a bunch of decls into the header file that have to be bound by the linker. What if instead it did something like define two struct types:

1. A struct type `funs` containing function pointers to all the callbacks
2. A struct type `vals` containing (pointers to) all the external values

And then the step function would take in pointers to both of these.

This would allow for easily having multiple copilot instances in your C program and wouldn't rely on the linker as much.

fdedden, 2019-11-30:
That's an interesting view. Are there any reasons for you to use multiple instances of Copilot? Only one I could think of is having monitors at different rates. Copilot was never designed to be used in multiple instances, as having only one specification makes the monitor easier to understand and implement. Different rates would ideally be solved within Copilot.

We will be discussing this the possibility of implementing this idea. We will keep you updated on our findings.

wyager, 2019-12-02:
I could imagine someone building a system that connects some (possibly unknown at compile time) number of identical peripheral systems to a central controller and wanting to instantiate a copilot instance for each connected peripheral system, at the time of connection.

fdedden, 2019-12-04:
This is something that would ideally be handled within Copilot, but this is not trivial as dealing with unknown amounts of anything (be it peripherals or elements of an array) at compile time is something highly unwanted in aerospace systems.

Currently we are focusing on maturing Copilot and fixing bugs. Nonetheless we will take this into consideration, for possible future releases.

ivanperez-keera, 2021-08-06:
The issue of generating more modular code is being considered, although, at least currently, not using this approach. Different approaches are being considered, for example, using just a prefix for all names within a module, so that all functions generated, public or private, are called `prefix_<name>`.

This is related also to #64 , which is a relatively high priority at the moment.

Currently, the C code contains mainly three kinds of public declarations:
- Input global variables. Names are fixed, picked by the user, and should not be changed.
- Step function. Name customizable if #64 is implemented.
- Handlers. Names are fixed, picked by the user, and should not be changed.

So, at least for the time being, if #64 is implemented, the code should be relatively modular, since internal names need not be exposed and the linked should not confuse those.

I think it's safe to close this issue for now, and revisit the problem if it arises once #64 is implemented.

closed 2021-08-06 by ivanperez-keera