Copilot-Language/copilot #55: surprising behavior difference between interpreted and C

joeyh, 2020-01-24:
Adapting the counter.hs example to generate C code, I was surprised to see that the C counter did not generate the same stream of numbers as the interpreted counter. I am not 100% sure this is a bug, since I'm very new to copilot (and have not even finished reading the manual yet), but it seems like one to me.

I'm attaching a counter.hs, which can be interpreted by running "runghc.hs counter i" or generate C code by running "runghc counter.hs c". The attached main.c is a simple framework to run the C code, so "runghc counter.hs c; make main; ./main" will run the C code. (github for some reason required I gzip these attachments)

Rather than reset at 256, I made it reset at 4, so it's easier to see the difference in behavior between C and interpreted. Here's the output from the two side-by-side for comparison.

    C   interpreted
    1   (1)
    2   (2)
    3   (3)
    0   (0)
    2   (1)
    3   (2)
    4   (3)
    5   (4)

copilot-3.1 ghc-8.8.1

[counter.hs.gz](https://github.com/Copilot-Language/copilot/files/4109641/counter.hs.gz)
[main.hs.gz](https://github.com/Copilot-Language/copilot/files/4109661/main.hs.gz)

fdedden, 2020-02-09:
This is indeed a bug, and possibly connected to https://github.com/Copilot-Language/copilot/issues/54. We will be looking into this.

fdedden, 2020-03-25:
This issue and #57 seem to have the same origin, and are both fixed with modifications of mentioned in https://github.com/Copilot-Language/copilot/issues/57#issuecomment-604097202.

fdedden, 2020-03-28:
The issue is fixed by the new commits done to copilot-c99. In the next days an update to copilot-c99 will be pushed to hackage.