Copilot-Language/copilot #56: more type information for Stream

joeyh, 2020-01-27:
I'm quickly noticing that the low-level C type of a Stream is unsatisfying. It would be really nice to be able to attach my own type information to a Stream, so I can avoid eg, confusing pressure (Stream Float) with temperature (Stream Float).

Using a newtype of Stream is unsatisfactory, because my new type would not be able to be used within expessions in the copilot DSL.

A phantom type seems like a natural approach, eg:

  type Stream t = StreamP p t

And change copilot to use StreamP p everywhere it uses Stream currently. Then I could write eg:

    emergency_shutdown_required :: StreamP Pressure Float -> StreamP Temperaure Float -> Stream Bool

Would you be open to a change along these lines?

fdedden, 2020-02-09:
We have had thoughts about similar ways to make Copilot more type-safe. We will need to discuss this approach and the consequences it has on certain operations (casting of values comes to mind).

In general we are currently focused primarily on making Copilot more stable and mature, but having such a feature extension is definitely interesting for future releases.We will discuss the approach with our next meeting.

Thanks for your suggestion!

ivanperez-keera, 2021-08-06:
Although the approach is different, this issue is closely related to #58  and solving that one would solve this one too. I'm going to close this one in favor of #58 simply as a way to manage complexity, but this approach may be simpler and easier to start with.

closed 2021-08-06 by ivanperez-keera