Copilot-Language/copilot #57: Inconsistency between interpret and generated C

avieth, 2020-03-22:
This copilot program

```Haskell
{-# LANGUAGE RebindableSyntax #-}

module Bug where

import Language.Copilot
import Copilot.Compile.C99
import Prelude hiding ((++), drop)

time_ms :: Stream Word32
time_ms = extern "time_ms" (Just [1,2..])

times :: Stream Word32
times = [0] ++ time_ms

elapsed :: Stream Word32
elapsed = [0] ++ (elapsed + (time_ms - times))

spec :: Spec
spec = trigger "bug" true [arg $ elapsed]

main :: IO ()
main = reify spec >>= compile "bug"
```

works as expected in the interpeter

```
> interpret 10 spec
bug:
(0)
(1)
(2)
...
```

but when compiled to C and run with this main

```C
#include <unistd.h>
#include <stdint.h>
#include <stdio.h>

#include "bug.h"

#define TIME_STEP_MS 100

uint32_t time_ms = 0;

void bug(uint32_t t) {
  printf("elapsed is %i\n", t);
}

void main() {
  while (1) {
    time_ms += TIME_STEP_MS;
    step();
    usleep(TIME_STEP_MS*1000);
  }
}
```

the elapsed time never increases

```
$ ./a.out
elapsed is 0
elapsed is 0
elapsed is 0
...
```

Why could this be? The generated `bug.c` is here, with comments added by me

```C
#include <stdint.h>
#include <stdbool.h>
#include <string.h>

#include "bug.h"

static uint32_t time_ms_cpy;
static uint32_t s1[(1)] = {(0)};
static uint32_t s0[(1)] = {(0)};
static size_t s1_idx = (0);
static size_t s0_idx = (0);

// Correponds to the stream called "times" in the Haskell text.
uint32_t s1_gen(void) {
  return time_ms_cpy;
}

// Corresponds to the stream called "elapsed" in the Haskell text.
uint32_t s0_gen(void) {
  return ((s0)[s0_idx]) + ((time_ms_cpy) - ((s1)[s1_idx]));
}

bool bug_guard(void) {
  return true;
}

uint32_t bug_arg0(void) {
  return (s0)[s0_idx];
}

void step(void) {
  (time_ms_cpy) = (time_ms);
  if ((bug_guard)()) {
    (bug)(((bug_arg0)()));
  };
  // The "times" stream is updated before the "elapsed" stream, but that's no good
  // because the "elapsed" stream uses the difference between time_ms_cpy and
  // the "times" stream, which are made equal here.
  ((s1)[s1_idx]) = ((s1_gen)());
  ((s0)[s0_idx]) = ((s0_gen)());
  (s1_idx) = ((++(s1_idx)) % (1));
  (s0_idx) = ((++(s0_idx)) % (1));
}
```

avieth, 2020-03-22:
Potential fix here https://github.com/Copilot-Language/copilot-c99/pull/20

fdedden, 2020-03-25:
Hi Alexander,
Thanks for your report, this is indeed a bug. Instead of your solution (adding an extra element to the buffer), I think we can simply store the new value of the stream in a temporary variable and update the stream later:

```C
uint32_t s1_tmp = s1_gen();
uint32_t s0_tmp = s0_gen();

s1[s1_idx] = s1_tmp;
s0[s0_idx] = s0_tmp;
```

Is there any reason you opted for your solution?

fdedden, 2020-03-25:
I will fix this issue this week, and schedule for a new release during the weekend.

avieth, 2020-03-25:
> Is there any reason you opted for your solution?

It was just the most obvious/easy to me at the time :) Yours sounds good too.

avieth, 2020-03-25:
I've also made a patch to various copilot libraries to allow for optional naming of trigger arguments, so that the generated header file can be more useful as an interface to pass along to other developers who may not want to deal with the Haskell spec: instead of `trigger_name_arg0` it can be more descriptive. If you think that's a good idea to include I can put up pull requests.