Copilot-Language/copilot #59: Add ability to add names / labels to streams

fdedden, 2020-04-07:
Being able to give names or labels to streams has the following benefits:
* Traceability between the original Copilot specification and C code is improved greatly, which can be a really important point for certification purposes.
* The generated C code will be easier to read, again helping with certification.
* Same approach can possibly used for name arguments of triggers, making the generated header file readable for people that don't know about the Copilot specification.

There are some existing pull requests related to this enhancement:
https://github.com/Copilot-Language/copilot-core/pull/18
https://github.com/Copilot-Language/copilot-language/pull/9
https://github.com/Copilot-Language/copilot-c99/pull/27

These are denied and closed for now, as they implement a way which we deem to specific. They might come in handy later on.

robdockins, 2021-10-29:
I don't know if this is the right place to drop this idea, but I was thinking the other day that it might work out nicely to use the magic `HasCallStack` constraint to implement some form of traceability for stream and trigger definitions, etc.  CF: https://hackage.haskell.org/package/base-4.16.0.0/docs/GHC-Stack.html#t:HasCallStack

Basically, you can add the `HasCallStack` constraint to any Haskell declaration, and the compiler with automatically populate it with information about call sites.  That information is propagated to `error` calls and thrown exceptions, but can also be directly fetched using the `callStack` combinator.  Adding `HasCallStack` annotations into the user-facing APIs for copilot, and then grabbing those stacks and stashing them into high-level stream definitions would allow a pretty lightweight way to maintain traceability connections to specification code.

ivanperez-keera, 2024-03-06:
Tagging @RyanGlScott for awareness.