Copilot-Language/copilot #64: Allow customizing the name of he step function

ivanperez-keera, 2020-12-06:
The name of the `step` function generated by Copilot's C99 backend is fixed. This limits the ability to link Copilot-generated C code with object files that contain that name already, which may happen if the user has defined it, if there is more than one Copilot specification being linked together but compiled separately, or when it exists for other reasons.*

We could provide a way for people to customize the name of the step function. My immediate thought it to allow them to customize this at the highest level possible, I guess in the call to `Copilot.Compile.C99.Compile.compile`; that modules also fixes the name `step`.

*For an example of the latter: I recently compiled a Copilot module as a dynamically linked library and loaded it from Python. I used it to monitor a Python program, calling Copilot via the FFI. All went well, but I had to rename `step` for it to work.

fdedden, 2020-12-10:
This issue is related to #53. Although different of approach, they solve the same problem.

ivanperez-keera, 2021-03-31:
My immediate thought would be to close https://github.com/Copilot-Language/copilot/issues/53 in favour of this one, because 1) this is a relatively simple solution, requiring only a change in 4 functions in copilot-c99 (`Copilot.Compile.C99.Compile.compile`, `Copilot.Compile.C99.Compile.compilec`, `Copilot.Compile.C99.Compile.compileh`, and `Copilot.Compile.C99.CodeGen.mkstep`), and 2) even thought a struct-based interface may look like a relatively simple solution, it's likely not the interface that a C programmer would write by hand. It's just one level of indirection, but the simple fact that they'd just write the functions in a plain C file and we are giving them a struct with pointers to the functions can be enough for people to stop using it.

Instead, we should be aiming at giving code that conforms to the standard way of doing things, and at helping users of Copilot obtain guarantees that the code works as intended.

ivanperez-keera, 2021-08-06:
This change is now implemented in the branch https://github.com/Copilot-Language/copilot/tree/dev-step-function. it works, and the only question I have right now is which modules should export what. At first sight, it looks like the `copilot-c99` package exposes definitions that are not needed outside the package (e.g., auxiliary definitions).

commit 2394e4a0 (2021-08-12): Merge branch 'dev-step-function'. Close #64.

The name of the step function generated by Copilot's C99 backend is
fixed. This limits the ability to link Copilot-generated C code with
object files that contain that name already, which may happen if the
user has defined it, if there is more than one Copilot specification
being linked together but compiled separately, or when it exists for
other reasons.

This patch provides a new operation to compile the C code with
additional settings, defined in an auxiliary type, which include
the name of the step function. The name defaults to "step" for
bac [... cut]
commit 2dc10ca2 (2021-08-12): Merge branch 'dev-step-function'. Close #64.

The name of the step function generated by Copilot's C99 backend is
fixed. This limits the ability to link Copilot-generated C code with
object files that contain that name already, which may happen if the
user has defined it, if there is more than one Copilot specification
being linked together but compiled separately, or when it exists for
other reasons.

This patch provides a new operation to compile the C code with
additional settings, defined in an auxiliary type, which include
the name of the step function. The name defaults to "step" for
bac [... cut]
closed 2021-08-12 by ivanperez-keera