Copilot-Language/copilot #88: Cleanup and merge Copilot repositories

fdedden, 2021-03-28:
The current situation where the project is spread over multiple repositories doesn't server any real purpose and complicates development, additionally it is never really clear where an issue should be posted.

We should cleanup the layout of the project and merge the repositories under `lib/` into `copilot`. `copilot-discussion` shouldn't be part of this as does not contain any code.

Taking on this change is a big step, that has to executed perfectly to be consistent with the history of the project:
- Move milestones, issues, pull requests and issue labels over to `copilot`. Milestones sharing their name should be merged. Issues and pull requests will get new unique identifiers (as there will obviously be overlap, i.e. multiple issues with the same number). 
- Per merged repository, we need to keep track of the mapping between old issue numbers and new ones: We need to update the references in commit messages to these new issue numbers.
- Per repository, we need to update the issue numbers in changelogs as well.
- We need to merge the code of the repository into the `lib/` subdirectory of the `copilot` repository:
  * Remove the repository from the submodules (exact method still pending):
  ```$ git submodule deinit <repo>```
  * In the `copilot` directory, add a and fetch new remote:
  ```$ git remote add -f <repo> git@github.com:Copilot-Language/<repo>.git```
  * Create a local branch for the remote:
  ```$ git branch -t <repo>-master <repo>/master```
  * Checkout the new branch:
  ```$ git checkout <repo>-master```
  * Create the `lib/<repo>` directory in the new branch:
  ```$ mkdir lib/<repo>```
  * Move all files to lib/<repo>:
  ```$ git mv <allfiles> lib/<repo>```
  * Commit the move:
  ```$ git commit -m 'Moved files to subdirectory.'```
  * Rewrite all commit messages of this repository:
  ```$ (method still pending)```
  * Update changelog and commit changes.
  * Return to the master of `copilot`:
  ```$ git checkout master```
  * Merge the repository into `copilot`:
  ```$ git merge <repo>-master -no-ff --allow-unrelated-histories```

Note that with this method, the resulting directory structure will not change. All history, commit messages, issues, milestones etc. should all be transferred over the central `copilot` repository.

Afterwards we can opt to hide or remove the now obsolete repositories.

fdedden, 2021-05-03:
Progress:

- [ ] copilot-theorem
- [ ] copilot-language
- [ ] copilot-libraries
- [ ] copilot-core
- [ ] copilot-c99

fdedden, 2021-05-03:
For now, pull requests are skipped due to their complexity. Maybe we can find a proper solution later.

fdedden, 2021-05-06:
All master branches are merged. There might be some interesting branches left in the original repositories, these can be merged if necessary.

ivanperez-keera, 2021-05-07:
I think this issue can be closed, since there are issues for all pending tasks related to the merge.