Copilot-Language/copilot #223: generated C code cannot be compiled by C++ compiler (multi-dimensional array initialization)

joeyh, 2021-07-02:
arduino-copilot uses copilot-c99 to generate code, and then puts that into an arduino sketch, which gets compiled by a C++ compiler. Maybe that was a bad idea, but hey, it's worked great so far.

Here's some code that copilot-c99 generated that does not work that way:

    static bool s0[(1)][(8)] = {((bool [(8)]){(true), (false), (true), (true), (true), (true), (true), (true)})};

Which was generated from this haskell:

   test :: Stream (Array 8 Bool)
   test = [array [True, False, True, True, True, True, True, True]] ++ test

And the compiler complains:

    my.ino:21:102: error: array must be initialized with a brace-enclosed initializer
     static bool s0[1][8] = {(bool [(8)]){(true), (false), (true), (true), (true), (true), (true), (true)}};

As far as I can see, this would be equivilant, and it compiles as both C and C++:

    static bool s0[(1)][(8)] = {{(true), (false), (true), (true), (true), (true), (true), (true)}};

ivanperez-keera, 2021-08-06:
@fdedden is there any case in which that kind of explicit type casting(?) is necessary? I'm wondering if it's needed for nested or multi-dimensional arrays.

ivanperez-keera, 2021-08-12:
@joeyh Is this because the value of the array must be a statically known constant and the casting occurs in runtime?

joeyh, 2021-08-13:
I don't know why the C++ compiler doesn't like it, this is beyond my C(++) knowledge.

Would it be helpful to have a more complete test case showing how the array is used? I have one, but it does use arduino-copilot.

joeyh, 2021-08-13:
https://tmp.joeyh.name/Demo.hs -- haskell program using arduino-copilot, which generates the next file when run

https://tmp.joeyh.name/Blink.ino -- arduino sketch generated by that, compiling it with the arduino compiler will repproduce the error

ivanperez-keera, 2022-04-20:
@joeyh I've just uploaded a change (see #276) that I think was a (near-) duplicate of this issue. It's in the master branch. Could you see if that solves your issue?

joeyh, 2022-04-22:
@ivanperez-keera I've verified that this issue is fixed. Thanks!

ivanperez-keera, 2022-04-22:
Excellent! Closed in favor of #216 then. Thanks

closed 2022-04-22 by ivanperez-keera