Copilot-Language/copilot #238: Code generation issue?

robdockins, 2021-08-12:
I was playing around with some of the copilot examples, and ran into what appears to be a codegen issue.  I wanted to add some smoothing to the `Heater` example by computing a trailing 3-sample average, so I added these lines:

```
-- Compute a trailing average of the last 3 samples
avgTemp :: Stream Float
avgTemp = mean 3 (replicate 3 19.5 ++ ctemp)
```

And rewrote the triggers to use the average temp.

However, in the generated code, I'm seeing the following:

```
bool heaton_guard(void) {
  return (((((s0)[s0_idx]) + ((s0)[(s0_idx) + (1)])) + ((s0)[(s0_idx) + (2)])) / ((float)(3.0))) < ((float)(18.0));
}
```

I gather from the step function code that `s0` is intended to be a ring buffer and `s0_idx` points to the oldest element of the buffer.  These lookups, then, ought to be computed modulo the length of the buffer, I think. Here, it seems clear that we can get out-of-bounds access.

ivanperez-keera, 2021-08-12:
Hi @robdockins . I think you are absolutely right.

Just for completion, let me post the rest of the code generated for this example to show that the index would indeed go out of bounds:

```C
bool heaton_guard(void) {                                                       
  return (((((s0)[s0_idx]) + ((s0)[(s0_idx) + (1)])) + ((s0)[(s0_idx) + (2)])) / ((float)(3.0))) < ((float)(18.0));
}                                                                               
                                                                                
float heaton_arg0(void) {                                                       
  return ((((s0)[s0_idx]) + ((s0)[(s0_idx) + (1)])) + ((s0)[(s0_idx) + (2)])) / ((float)(3.0));
}                                                                               
                                                                                
void step(void) {                                                               
  float s0_tmp;                                                                 
  (temperature_cpy) = (temperature);                                            
  if ((heaton_guard)()) {                                                       
    (heaton)(((heaton_arg0)()));                                                
  };                                                                                                                                                          
  (s0_tmp) = ((s0_gen)());                                                      
  ((s0)[s0_idx]) = (s0_tmp);                                                    
  (s0_idx) = (((s0_idx) + (1)) % (3));                                          
}                                                                               
```

Clearly, accesses need to be modulo the length of the array.

robdockins, 2021-08-12:
OK, good to know I'm not missing something.  Looks like this code comes from here:

https://github.com/Copilot-Language/copilot/blob/c2d7bc0032c7fcae6693032c62c71c780c2511b2/lib/copilot-c99/src/Copilot/Compile/C99/Translate.hs#L29-L35

As far as I can see, information about the length of the buffer isn't available at this point in the code.  Either it will need to be piped into the `Drop` constructor, or fed in as a `Map Id Stream` or something.  Alternately, accessor functions could be emitted as part of the codegen process instead of constructing an indexing operation directly.

commit bf9ff76b (2021-08-16): Merge branch 'dev-ring-buffer-fix'. Close #238.

Indexing into ring buffers for delayed stream values was being
calculated incorrectly. The base pointer for the ring buffer was being
correctly reduced modulo the array length, but buffer access indexing
was not.

This patch causes code generation to introduce accessor functions that
uniformly compute the correct index to access a position in the stream
buffer.
closed 2021-08-16 by ivanperez-keera