Copilot-Language/copilot #246: atan2 not supported?

simondlevy, 2021-08-16:
Here's a self-explanatory, minimum working example showing the problem I'm having compiling Copilot code (using runhaskell) that contains the atan2 function:

`{-# LANGUAGE RebindableSyntax #-}
{-# LANGUAGE DataKinds        #-}

module Main where

import Language.Copilot
import Copilot.Compile.C99

x :: Stream Double
x = extern "x" Nothing

y :: Stream Double
y = extern "y" Nothing

spec = do

  let c = cos x

  let s = sin y

  -- Uncomment to get compiler error
  -- let a = atan2 y x

  trigger "fun" true [arg c]

-- Compile the spec
main = reify spec >>= compile "main"`

ivanperez-keera, 2021-08-16:
That is correct. I assume the reason is just that it was never needed it.

So long as it can be implemented safely, we can just add it.

There are three changes mainly that need to happen for this:
- The introduction of a new `Op2`, with its corresponding high level interface.
- The translation of that new `Op2` into C code.
- The interpretation of that new `Op2` in Haskell.

If other functions can be implemented as well, then we could give an instance of the type class `RealFloat`. This second point depends on whether operations are safe, and whether they can be implemented in C in hard-realtime (presumably yes).

simondlevy, 2021-08-17:
Thanks for looking into this, Ivan.  For what it's worth, I need this function to perform the conversion from quaternions to Euler angles in my flight-control firmware (which I am slowly migrating to Copilot).

agoodloe, 2021-08-17:
These sorts of coordinate transforms will  likely be needed as for us as we apply Copilot for more airspace work. I remember they came up when monitoring for well clear aircraft.

ivanperez-keera, 2021-08-18:
@simondlevy We'll soon have a meeting to plan the next version of Copilot. This feature definitely won't make it into the very next release (3.5, coming out in early September), but we'll see if we can put it in for the next one. I'll keep you posted.

simondlevy, 2021-08-18:
Thanks again, guys.  As I become more proficient at Copilot, I hope to be able to contribute in some helpful way.

ivanperez-keera, 2021-09-01:
@simondlevy We just had our planning meeting and I can confirm we'll be trying to put this in 3.6. If there are no obstacles, you may see it released 11/7/2021 but it should be in the repo earlier than that. I'll ping you once the branch with the new feature is ready so you can try it before it is merged.

ivanperez-keera, 2021-10-06:
@simondlevy I have just pushed a draft to https://github.com/Copilot-Language/copilot/tree/dev-atan2.

Can you please try it?

To use `atan2` on streams, you'll have to hide `atan2` from the `Prelude` because the class in which it is defined also requires some definitions that we cannot provide for streams (I think).

Support for `atan2` in `copilot-theorem` is *not* included in that draft branch.

ivanperez-keera, 2021-10-19:
I have updated the branch also with an implementation of `ceiling` and `floor`.

I am doubtful about the placement of these definitions in `Copilot.Language.Stream`. The reason to put them there is because that's the same module that exports the `Num` instances, but, for atan2/ceiling/floor, this frequently causes a clash with functions from the `Prelude`, which are re-exported by `Language.Copilot`, so the module `Copilot.Language.Stream` needs to be imported qualified.

This work is ready to be merged if the definitions should rename where they are currently.

commit 793fd983 (2021-10-19): Merge branch 'dev-atan2'. Close #246.

Introduce support for new Ops atan2, ceiling, floor.
closed 2021-10-19 by ivanperez-keera