Copilot-Language/copilot #249: Strange conversion of "I" => "J" by compiler

simondlevy, 2021-09-06:
Here's the familiar heater example, modified so that "temperature" is spelled "temperIture".  When I compile it, the C variable name comes out as "temperJture".

`--------------------------------------------------------------------------------
-- Copyright 2019 National Institute of Aerospace / Galois, Inc.
--------------------------------------------------------------------------------

-- This is a simple example with basic usage. It implements a simple home
-- heating system: It heats when temp gets too low, and stops when it is high
-- enough. It read temperature as a byte (range -50C to 100C) and translates
-- this to Celcius.

module Main where

import Language.Copilot
import Copilot.Compile.C99

import Prelude hiding ((>), (<), div)

-- External temperature as a byte, range of -50C to 100C
temp :: Stream Word8
temp = extern "temperIture" Nothing

-- Calculate temperature in Celcius.
-- We need to cast the Word8 to a Float. Note that it is an unsafeCast, as there
-- is no direct relation between Word8 and Float.
ctemp :: Stream Float
ctemp = (unsafeCast temp) * (150.0 / 255.0) - 50.0

spec = do
  -- Triggers that fire when the ctemp is too low or too high,
  -- pass the current ctemp as an argument.
  trigger "heaton"  (ctemp < 18.0) [arg ctemp]
  trigger "heatoff" (ctemp > 21.0) [arg ctemp]

-- Compile the spec
main = reify spec >>= compile "heater"
`

ivanperez-keera, 2021-09-06:
That is such an odd bug! Excellent catch.

ivanperez-keera, 2021-09-07:
@simondlevy This issue is caused by the C pretty printer. It was fixed in this commit (https://github.com/fdedden/language-c99/commit/fa5d557a90b89c4e6d4d4ba983df045ce300882e) but that version was not published on hackage.

If you install language-c99 from that repo, the heater example you wrote works.

Pinging @fdedden to see if he can publish a new version.

Since this is a bug in another package, I think we can close this issue.

closed 2021-09-07 by ivanperez-keera

fdedden, 2021-09-13:
I have uploaded a newer version of language-c99 (0.1.3) that fixes this issue. Doing a `cabal update` and reinstalling copilot by passing `--upgrade-dependencies` should force the newer dependency to be build.

Please let me know if you encounter any further problems.