Copilot-Language/copilot #255: destination directory for code generation

robdockins, 2021-09-21:
It would be convenient if we could add an option to the `CSettings` that would allow the user to select the destination directory for generated C code.

ivanperez-keera, 2021-09-21:
Agreed.

Out of curiosity, and I'm not saying it's a good design, what happens if you use `"directory/file"` as the file name prefix when compiling to C?

ivanperez-keera, 2021-09-21:
(assuming that `directory` exists)

robdockins, 2021-09-21:
That _almost_ works.  However, the generated C file now has

```
#include "temp/engine.h"
```

which doesn't get the relative directory paths right.

ivanperez-keera, 2021-09-28:
@robdockins I'm isolating each contribution (and adapting it to our process a bit too). I picked one commit from your PR.

Can you please check if this addresses this specific issue 255?

https://github.com/Copilot-Language/copilot/tree/develop-outputdir

There's a difference with yours: this one assumes that the directory is _not optional_. Of course, it defaults to the CWD, which I think is always equivalent.

By the way, why not just:

```
mkdir -p <newdir>; push $_; runhaskell $OLDPWD/<code>.hs; pop
```

Normally, command line tools (e.g., tee, wget, etc.) won't create directories for you if they do not exist. I know it's a small thing, but I wonder if it would be simpler to just assume it must exist.

robdockins, 2021-09-28:
As far as I know, that change should be fine.

The verifier system I'm working on allows the user to specify directories for final output and intermediate results via a configuration file and they might not be the same location.  I think it's nicer than forcing the user to manipulate their own CWD.

Personally, I'm always a bit annoyed if directories I specify aren't created as necessary, but reasonable people can differ.

commit 43e1efb9 (2021-10-01): Merge branch 'develop-outputdir'. Close #255.

It would be convenient if we could add an option to the CSettings that
would allow the user to select the destination directory for generated C
code.

This branch introduces a new field in the type CSettings (argument to
compileWith), to customize the target directory. The new directory is
created if necessary.
closed 2021-10-01 by ivanperez-keera