Copilot-Language/copilot #258: Special-case codegen handling for length 1 buffers

robdockins, 2021-09-21:
Currently, the code generator will produce uniform code for ring buffers of all lengths, including length 1.  This produces a length 1 array, and a buffer index global variable. This is correct, and works fine. However, at optimization level 1 and above, the `clang` compiler will deduce that the index variable is always 0, and that the array is only ever accessed at position 0.  Because these globals are declared `static`, the compiler feels free to rearrange these globals, and it does.  It will remove the index variable altogether and produce a new global for the buffer at the stream type (instead of a length 1 array).

This confuses the verification procedure I've been working on, as it expects to be able to make assertions about the variables as appearing in the generated source code.  We can suppress this behavior from `clang` by compiling at `-O0`, but obviously this also suppresses a variety of other optimizations we probably prefer to have done.

Another solution to this issue that I've prototyped is to perform this optimization ourselves, in code generation.  Then, we can reliably expect this special case regardless of the compiler transformations, and handle it properly in the verifier.  As a bonus, the generated code is rather simpler in this common special case.

ivanperez-keera, 2021-09-21:
I think it may be best that we have a quick chat about this.

robdockins, 2021-09-23:
Following discussion on this topic, it seems like we will not implement this special case. For posterity, and in case we reverse this decision, the implementation I was working with can be found here:

https://github.com/Copilot-Language/copilot/compare/master...GaloisInc:length-1-buffer

closed 2021-09-23 by robdockins