Copilot-Language/copilot #274: `what4-structs` example fails to compile to C code

RyanGlScott, 2021-11-02:
I was interested to know what C code the `what4-structs` example would compile to, so I tried making the following change:

```diff
diff --git a/copilot/copilot.cabal b/copilot/copilot.cabal
index dec9b1b..1b8c2ef 100644
--- a/copilot/copilot.cabal
+++ b/copilot/copilot.cabal
@@ -88,6 +88,7 @@ executable what4-structs
     hs-source-dirs:         examples/what4
     build-depends:          base
                           , copilot
+                          , copilot-c99
                           , copilot-theorem
     default-language:       Haskell2010
     if flag(examples)
diff --git a/copilot/examples/what4/Structs.hs b/copilot/examples/what4/Structs.hs
index 2fc1b25..83a7ab0 100644
--- a/copilot/examples/what4/Structs.hs
+++ b/copilot/examples/what4/Structs.hs
@@ -10,6 +10,7 @@ import qualified Prelude as P
 import Control.Monad (void, forM_)
 
 import Language.Copilot
+import Copilot.Compile.C99
 import Copilot.Theorem.What4
 
 
@@ -69,6 +70,7 @@ spec = do
 main :: IO ()
 main = do
   spec' <- reify spec
+  compile "structs" spec'
 
   -- Use Z3 to prove the properties.
   results <- prove Z3 spec'
```

If I do this, however, I get the following error message when compiling the code:

```
$ cabal run what4-structs -fexamples
Resolving dependencies...
Up to date
what4-structs: Empty BlockItemList is not allowed
CallStack (from HasCallStack):
  error, called at src/Language/C99/Util/IsList.hs:10:24 in language-c99-util-0.1.1-5c88c936679457490f9da31103e55d54f5b3e7222fb03697d5ac08b8412de52c:Language.C99.Util.IsList
```

RyanGlScott, 2021-11-02:
The culprit may be `language-c99-simple`'s use of `fromList`. In [this code](https://github.com/fdedden/language-c99-simple/blob/755ad2fc610a82888b9b64b3e6567bb86a7d7832/src/Language/C99/Simple/Translate.hs#L437-L441):

```hs
compound :: [Decln] -> [Stmt] -> C.CompoundStmt
compound ds ss = C.Compound (Just $ fromList items) where
  items = ds' ++ ss'
  ss' = map (C.BlockItemStmt . transstmt) ss
  ds' = map (C.BlockItemDecln . transdecln) ds
```

The `fromList` invocation is using the [`IsList` instance for `BlockItemList`](https://github.com/fdedden/language-c99-util/blob/e5a123b27867e5a235d6ab8ea5217b18b6faf621/src/Language/C99/Util/IsList.hs#L37-L40), which is defined in `language-c99-util`:

```hs
instance IsList BlockItemList where
  type Item BlockItemList = BlockItem
  fromList []     = error_emptylist "BlockItemList"
  fromList (x:xs) = foldl BlockItemCons (BlockItemBase x) xs
```

This fails when the input list is empty, which happens in the `what4-structs` example. One possible fix would be to check if the input list is empty, and if so, return `Nothing` instead:

```diff
diff --git a/src/Language/C99/Simple/Translate.hs b/src/Language/C99/Simple/Translate.hs
index 24f4555..45fb0f2 100644
--- a/src/Language/C99/Simple/Translate.hs
+++ b/src/Language/C99/Simple/Translate.hs
@@ -397,7 +397,7 @@ casestmt cs =
       Default s -> C.LabeledDefault (transstmt s)
 
 compound :: [Decln] -> [Stmt] -> C.CompoundStmt
-compound ds ss = C.Compound (Just $ fromList items) where
+compound ds ss = C.Compound (if null items then Nothing else Just $ fromList items) where
   items = ds' ++ ss'
   ss' = map (C.BlockItemStmt . transstmt) ss
   ds' = map (C.BlockItemDecln . transdecln) ds
```

That is enough to fix the `what4-structs` example, at least. There are some other suspicious uses of `Just . fromList` in `language-c99-simple` that perhaps ought to be warranted, however, such as [here](https://github.com/fdedden/language-c99-simple/blob/755ad2fc610a82888b9b64b3e6567bb86a7d7832/src/Language/C99/Simple/Translate.hs#L320) and [here](https://github.com/fdedden/language-c99-simple/blob/755ad2fc610a82888b9b64b3e6567bb86a7d7832/src/Language/C99/Simple/Translate.hs#L432).

RyanGlScott, 2021-11-02:
Here is a minimized version of `what4-structs` that also exhibits the same issue:

```hs
module Main (main) where

import Copilot.Compile.C99
import Language.Copilot

spec :: Spec
spec = return ()

main :: IO ()
main = do
  spec' <- reify spec
  compile "simple" spec'
```

fdedden, 2021-11-02:
Hi Ryan, thank you for letting us know! This is definitely a shortcoming in the language-c99-simple library, and shouldn't be arise at all.

The reason why you run into this problem is actually very simple: the spec does not contain any triggers. As triggers are the one thing that a spec consists of, a lack of triggers results in an empty spec, hence there is nothing to compile to C. Just for the interest of seeing the C99 code adding a trigger to the spec is enough, for example:

```haskell
spec :: Spec
spec = do
  let battery :: Stream Battery
      battery = extern "battery" Nothing

  trigger "testfun" true [arg battery]
```

@ivanperez-keera @agoodloe we might want to add a test in the `compile` function to check for empty specs.

ivanperez-keera, 2021-11-02:
In general and ideally, bugs we identify from now on should come with an independent way to automatically (if possible) reproduce the test, a fix, and the ability to re-run the test to see that the issue is fixed.

And as part of our (regression) tests, we should run these to make sure these bugs stay fixed.

RyanGlScott, 2021-11-03:
Even if I add the trigger from https://github.com/Copilot-Language/copilot/issues/274#issuecomment-958092970 into the example:

<details>

```diff
diff --git a/copilot/copilot.cabal b/copilot/copilot.cabal
index dec9b1b..1b8c2ef 100644
--- a/copilot/copilot.cabal
+++ b/copilot/copilot.cabal
@@ -88,6 +88,7 @@ executable what4-structs
     hs-source-dirs:         examples/what4
     build-depends:          base
                           , copilot
+                          , copilot-c99
                           , copilot-theorem
     default-language:       Haskell2010
     if flag(examples)
diff --git a/copilot/examples/what4/Structs.hs b/copilot/examples/what4/Structs.hs
index 2fc1b25..de04151 100644
--- a/copilot/examples/what4/Structs.hs
+++ b/copilot/examples/what4/Structs.hs
@@ -10,6 +10,7 @@ import qualified Prelude as P
 import Control.Monad (void, forM_)
 
 import Language.Copilot
+import Copilot.Compile.C99
 import Copilot.Theorem.What4
 
 
@@ -55,6 +56,8 @@ spec = do
   let battery :: Stream Battery
       battery = extern "battery" Nothing
 
+  trigger "testfun" true [arg battery]
+
   -- Check equality, indexing into nested structs and arrays. Note that this is
   -- trivial by equality.
   void $ prop "Example 1" $ forall $
@@ -69,6 +72,7 @@ spec = do
 main :: IO ()
 main = do
   spec' <- reify spec
+  compile "structs" spec'
 
   -- Use Z3 to prove the properties.
   results <- prove Z3 spec'
```
</details>

I still experience issues with the resulting C code. It produces this `structs.h` file:

```c
struct battery;
extern struct battery battery;
void testfun(struct battery testfun_arg0);
void step(void);
```

And this `structs.c` file:

```c
#include <stdint.h>
#include <stdbool.h>
#include <string.h>
#include <stdlib.h>
#include <math.h>

#include "structs.h"

struct battery
{ uint16_t temp;
  struct volts volts[(10)];
  uint32_t other[(10)][(5)];
};
static struct battery battery_cpy;

bool testfun_guard(void) {
  return true;
}

struct battery testfun_arg0(void) {
  return battery_cpy;
}

void step(void) {
  (battery_cpy) = (battery);
  if ((testfun_guard)()) {
    {(testfun)(((testfun_arg0)()));}
  };
}
```

If I try compiling this, I get the following error:

```
$ clang -c structs.c -o structs.o
structs.c:11:21: error: array has incomplete element type 'struct volts'
  struct volts volts[(10)];
                    ^
structs.c:11:10: note: forward declaration of 'struct volts'
  struct volts volts[(10)];
         ^
1 error generated.
```

For some reason, the C code has a declaration for `struct battery` but not `struct volts`, despite the fact that `Battery` and `Volts` both have `Struct` instances.

(I realize that this is a separate issue than the one raised above, so let me know if this would be better suited for a separate GitHub issue.)

ivanperez-keera, 2021-11-03:
Quick reply, as I haven't had time to read everything in detail: Yes, it the nature of the error is different, we should open a separate issue for it. It's best to document and fix each one separately, even if they are to some small (?) degree related or dependent, so long as they are not the same problem or fixing one necessarily fixes the other.

RyanGlScott, 2021-11-03:
I've opened #275 for the issue in https://github.com/Copilot-Language/copilot/issues/274#issuecomment-958987086. I've also changed the title of this issue to emphasize that the issue here is an empty specification.

ivanperez-keera, 2022-01-07:
So, irrespective of what `language-c99-simple` decides to do, Copilot should be able to handle this. One option would be to just detect this case and show an error message. The new alternative implementation of `compileWith`, the most general entry point to the compilation module, could be as follows (there's just a check and a new case if the list of triggers is empty):

```haskell
compileWith cSettings prefix spec
  | null (specTriggers spec)
  = do hPutStrLn stderr $
         "Copilot error: attempt at compiling empty specification.\n"
         ++ "You must define at least one trigger to generate C monitors."
       exitFailure

  | otherwise
  = do let cfile = render $ pretty $ C.translate $ compilec cSettings spec
           hfile = render $ pretty $ C.translate $ compileh cSettings spec

           -- TODO: find a nicer solution using annotated AST's
           -- Should figure out exactly which headers are needed, based on what
           -- is used.
           cmacros = unlines [ "#include <stdint.h>"
                             , "#include <stdbool.h>"
                             , "#include <string.h>"
                             , "#include <stdlib.h>"
                             , "#include <math.h>"
                             , ""
                             , "#include \"" ++ prefix ++ ".h\""
                             , ""
                             ]

       let dir = cSettingsOutputDirectory cSettings
       createDirectoryIfMissing True dir
       writeFile (dir </> prefix ++ ".c") $ cmacros ++ cfile
       writeFile (dir </> prefix ++ ".h") hfile
```

@fdedden @RyanGlScott

RyanGlScott, 2022-01-07:
I would happy with the approach in https://github.com/Copilot-Language/copilot/issues/274#issuecomment-1007397753.

closed 2022-01-07 by ivanperez-keera