Copilot-Language/copilot #275: Basic example involving structs generates invalid C code

RyanGlScott, 2021-11-03:
(Moved from a discussion starting at https://github.com/Copilot-Language/copilot/issues/274#issuecomment-958987086.)

Consider this variant of the [`what4-structs` example](https://github.com/Copilot-Language/copilot/blob/e8988de888b199a25ee54fe8a2d0fb9ed41412e4/copilot/examples/what4/Structs.hs), which adds a trigger:

```hs
{-# LANGUAGE DataKinds #-}

module Main where

import Language.Copilot
import Copilot.Compile.C99

data Volts = Volts
  { numVolts :: Field "numVolts" Word16
  , flag     :: Field "flag"     Bool
  }

instance Struct Volts where
  typename _ = "volts"
  toValues volts = [ Value Word16 (numVolts volts)
                   , Value Bool   (flag volts)
                   ]

instance Typed Volts where
  typeOf = Struct (Volts (Field 0) (Field False))

data Battery = Battery
  { temp  :: Field "temp"  Word16
  , volts :: Field "volts" (Array 10 Volts)
  , other :: Field "other" (Array 10 (Array 5 Word32))
  }

instance Struct Battery where
  typename _ = "battery"
  toValues battery = [ Value typeOf (temp battery)
                     , Value typeOf (volts battery)
                     , Value typeOf (other battery)
                     ]

instance Typed Battery where
  typeOf = Struct (Battery (Field 0) (Field undefined) (Field undefined))

spec :: Spec
spec = do
  let battery :: Stream Battery
      battery = extern "battery" Nothing

  trigger "testfun" true [arg battery]

main :: IO ()
main = do
  spec' <- reify spec
  compile "structs" spec'
```

If you run this, it will produce this `structs.h` file:

```c
struct battery;
extern struct battery battery;
void testfun(struct battery testfun_arg0);
void step(void);
```

And this `structs.c` file:

```c
#include <stdint.h>
#include <stdbool.h>
#include <string.h>
#include <stdlib.h>
#include <math.h>

#include "structs.h"

struct battery
{ uint16_t temp;
  struct volts volts[(10)];
  uint32_t other[(10)][(5)];
};
static struct battery battery_cpy;

bool testfun_guard(void) {
  return true;
}

struct battery testfun_arg0(void) {
  return battery_cpy;
}

void step(void) {
  (battery_cpy) = (battery);
  if ((testfun_guard)()) {
    {(testfun)(((testfun_arg0)()));}
  };
}
```

If I try compiling this, I get the following error:

```
$ clang -c structs.c -o structs.o
structs.c:11:21: error: array has incomplete element type 'struct volts'
  struct volts volts[(10)];
                    ^
structs.c:11:10: note: forward declaration of 'struct volts'
  struct volts volts[(10)];
         ^
1 error generated.
```

For some reason, the C code has a declaration for `struct battery` but not `struct volts`, despite the fact that `Battery` and `Volts` both have `Struct` instances.

RyanGlScott, 2021-11-09:
I think I see why this happens. Here is how `copilot-c99` determines what structs need to be declared:

https://github.com/Copilot-Language/copilot/blob/137a5b46485be2a7b23cf05f0aa37d02b9afe59a/copilot-c99/src/Copilot/Compile/C99/Compile/Internal.hs#L75-L82

The `exprtypes` function computes all `UType`s that are referenced in an expression. In turn, `exprtypes` invokes `typetypes`:

https://github.com/Copilot-Language/copilot/blob/137a5b46485be2a7b23cf05f0aa37d02b9afe59a/copilot-c99/src/Copilot/Compile/C99/CodeGen.hs#L165-L170

The `Struct` case here looks wrong. It returns all field types in a struct, but it does not do so recursively. For instances, on the example above, calling `typetypes` on the `Battery` struct would cause `Array 10 Volts` to be returned, but _not_ the `Volts` struct type itself. Moreover, `mkstructdeclns` will only produce a struct declaration if it matches against a `Struct` type directly, and since `Array 10 Volts` isn't directly a struct type, a struct declaration for `Volts` is never generated.

My first attempt to fix this was to recursively invoke `typetypes` in the `Struct` case:

```diff
diff --git a/copilot-c99/src/Copilot/Compile/C99/CodeGen.hs b/copilot-c99/src/Copilot/Compile/C99/CodeGen.hs
index 61bb68c..75a49f7 100644
--- a/copilot-c99/src/Copilot/Compile/C99/CodeGen.hs
+++ b/copilot-c99/src/Copilot/Compile/C99/CodeGen.hs
@@ -166,7 +166,7 @@ exprtypes e = case e of
 typetypes :: Typeable a => Type a -> [UType]
 typetypes ty = case ty of
   Array ty' -> [UType ty] `union` typetypes ty'
-  Struct x  -> [UType ty] `union` map (\(Value ty' _) -> UType ty') (toValues x)
+  Struct x  -> [UType ty] `union` concatMap (\(Value ty' _) -> typetypes ty') (toValues x)
   _         -> [UType ty]
 
 -- | Collect all expression of a list of streams and triggers and wrap them
```

However, that still isn't enough. If you do this, then the following C code will be generated for the code above:

```c
// structs.h
struct battery;
struct volts;
extern struct battery battery;
void testfun(struct battery testfun_arg0);
void step(void);
```
```c
// structs.c
#include <stdint.h>
#include <stdbool.h>
#include <string.h>
#include <stdlib.h>
#include <math.h>

#include "structs.h"

struct battery
{ uint16_t temp;
  struct volts volts[(10)];
  uint32_t other[(10)][(5)];
};
struct volts
{ uint16_t numVolts;
  bool flag;
};
static struct battery battery_cpy;

bool testfun_guard(void) {
  return true;
}

struct battery testfun_arg0(void) {
  return battery_cpy;
}

void step(void) {
  (battery_cpy) = (battery);
  if ((testfun_guard)()) {
    {(testfun)(((testfun_arg0)()));}
  };
}
```

This will fail to compile thusly:

```
$ clang -c structs.c -o structs.o
structs.c:11:21: error: array has incomplete element type 'struct volts'
  struct volts volts[(10)];
                    ^
./structs.h:2:8: note: forward declaration of 'struct volts'
struct volts;
       ^
1 error generated.
```

Indeed, because `struct battery` contains a non-pointer field that uses `struct volts`, the declaration for `struct volts` must come before the declaration for `struct battery`. One way to ensure this is to change `typetypes` to return types in the proper dependency order:

```diff
diff --git a/copilot-c99/src/Copilot/Compile/C99/CodeGen.hs b/copilot-c99/src/Copilot/Compile/C99/CodeGen.hs
index 61bb68c..27c219b 100644
--- a/copilot-c99/src/Copilot/Compile/C99/CodeGen.hs
+++ b/copilot-c99/src/Copilot/Compile/C99/CodeGen.hs
@@ -166,7 +166,7 @@ exprtypes e = case e of
 typetypes :: Typeable a => Type a -> [UType]
 typetypes ty = case ty of
   Array ty' -> [UType ty] `union` typetypes ty'
-  Struct x  -> [UType ty] `union` map (\(Value ty' _) -> UType ty') (toValues x)
+  Struct x  -> concatMap (\(Value ty' _) -> typetypes ty') (toValues x) `union` [UType ty]
   _         -> [UType ty]
 
 -- | Collect all expression of a list of streams and triggers and wrap them
```

With this change, the generated C code will finally compile:

```c
// structs.h
struct volts;
struct battery;
extern struct battery battery;
void testfun(struct battery testfun_arg0);
void step(void);
```
```c
// structs.c
#include <stdint.h>
#include <stdbool.h>
#include <string.h>
#include <stdlib.h>
#include <math.h>

#include "structs.h"

struct volts
{ uint16_t numVolts;
  bool flag;
};
struct battery
{ uint16_t temp;
  struct volts volts[(10)];
  uint32_t other[(10)][(5)];
};
static struct battery battery_cpy;

bool testfun_guard(void) {
  return true;
}

struct battery testfun_arg0(void) {
  return battery_cpy;
}

void step(void) {
  (battery_cpy) = (battery);
  if ((testfun_guard)()) {
    {(testfun)(((testfun_arg0)()));}
  };
}
```

RyanGlScott, 2021-11-10:
> If you have a pair of structs that reference each other, I think you will need a forward reference to break the cycle.

To be clear, `copilot-c99` is already generating forward references (see `structs.h` above). But even then, the only way you can have cyclic struct definitions in the first place in C (IIUC) is by having structs appear in pointer fields. As far as I am aware, however, `copilot-c99` never generates pointer fields in structs, however.

> Also, you'll need to remove duplicates from the list or you might get repeated declarations, I think.

That already happens, as `mkstructdeclns` uses `nub` to remove duplicates.

fdedden, 2021-12-17:
> To be clear, `copilot-c99` is already generating forward references (see `structs.h` above). But even then, the only way you can have cyclic struct definitions in the first place in C (if I understand correctly) is by having structs appear in pointer fields. As far as I am aware, however, `copilot-c99` never generates pointer fields in structs, however.

I think you are correct that structs with circular dependencies are only achievable by pointers. As dynamic memory is in general unwanted for realtime system, using pointers and `malloc` is something we avoided with Copilot. Therefore, Copilot doesn't generate any pointers.

commit 3c81d1d5 (2022-01-07): Merge branch 'dev-datatype-decl-order'. Close #275.

The function that generates C99 is not generating struct definitions for
all nested structs in a specification, and not in the right order. This
result in C code that is incomplete and/or incorrect. The same issue
applies to arrays.

This branch fixes that issue by changing the way in which types used in
a specification are calculated during the compilation to C. It also adds
an example that uses nested structs (which will fail to compile prior to
this fix and can therefore serve as a regression test of sorts).
closed 2022-01-07 by ivanperez-keera