Copilot-Language/copilot #276: Example involving arrays fails to compile with Clang

RyanGlScott, 2021-11-04:
Consider this variant of the [`Array.hs` example](https://github.com/Copilot-Language/copilot/blob/e8988de888b199a25ee54fe8a2d0fb9ed41412e4/copilot/examples/Array.hs):

```hs
{-# LANGUAGE RebindableSyntax #-}
{-# LANGUAGE DataKinds        #-}

module Main where

import Copilot.Compile.C99
import Language.Copilot

arr :: Stream (Array 2 Int32)
arr = [ array [0, 1]
      , array [2, 3]
      , array [4, 5]] ++ arr

spec :: Spec
spec = do
  trigger "func" true [arg arr]

main :: IO ()
main = do
  spec' <- reify spec
  compile "array" spec'
```

If you run this, it will produce this `array.h` file:

```c
void func(int32_t func_arg0[(2)]);
void step(void);
```

And this `array.c` file:

```c
#include <stdint.h>
#include <stdbool.h>
#include <string.h>
#include <stdlib.h>
#include <math.h>

#include "array.h"

static int32_t s0[(3)][(2)] = {((int32_t [(2)]){((int32_t)(0)), ((int32_t)(1))}), ((int32_t [(2)]){((int32_t)(2)), ((int32_t)(3))}), ((int32_t [(2)]){((int32_t)(4)), ((int32_t)(5))})};
static size_t s0_idx = (0);

int32_t * s0_get(size_t x) {
  return (s0)[((s0_idx) + (x)) % (3)];
}

int32_t * s0_gen(void) {
  return (s0_get)((0));
}

bool func_guard(void) {
  return true;
}

int32_t * func_arg0(void) {
  return (s0_get)((0));
}

void step(void) {
  int32_t s0_tmp[(2)];
  if ((func_guard)()) {
    {(func)(((func_arg0)()));}
  };
  (memcpy)((s0_tmp), ((s0_gen)()), (2));
  (memcpy)(((s0)[s0_idx]), (s0_tmp), (2));
  (s0_idx) = (((s0_idx) + (1)) % (3));
}
```

If I try compiling this with Clang, I get the following error:

```
$ clang --version
clang version 10.0.0-4ubuntu1 
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin

$ clang -c array.c -o array.o
array.c:9:32: warning: incompatible pointer to integer conversion initializing 'int32_t' (aka 'int') with an expression of type 'int32_t [2]' [-Wint-conversion]
static int32_t s0[(3)][(2)] = {((int32_t [(2)]){((int32_t)(0)), ((int32_t)(1))}), ((int32_t [(2)]){((int32_t)(2)), ((int32_t)(3))}), ((int32_t [(2)]){((int32_t)(4)), ((int32_t)(5))})};
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
array.c:9:83: warning: incompatible pointer to integer conversion initializing 'int32_t' (aka 'int') with an expression of type 'int32_t [2]' [-Wint-conversion]
static int32_t s0[(3)][(2)] = {((int32_t [(2)]){((int32_t)(0)), ((int32_t)(1))}), ((int32_t [(2)]){((int32_t)(2)), ((int32_t)(3))}), ((int32_t [(2)]){((int32_t)(4)), ((int32_t)(5))})};
                                                                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
array.c:9:134: warning: incompatible pointer to integer conversion initializing 'int32_t' (aka 'int') with an expression of type 'int32_t [2]' [-Wint-conversion]
static int32_t s0[(3)][(2)] = {((int32_t [(2)]){((int32_t)(0)), ((int32_t)(1))}), ((int32_t [(2)]){((int32_t)(2)), ((int32_t)(3))}), ((int32_t [(2)]){((int32_t)(4)), ((int32_t)(5))})};
                                                                                                                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
array.c:9:32: error: initializer element is not a compile-time constant
static int32_t s0[(3)][(2)] = {((int32_t [(2)]){((int32_t)(0)), ((int32_t)(1))}), ((int32_t [(2)]){((int32_t)(2)), ((int32_t)(3))}), ((int32_t [(2)]){((int32_t)(4)), ((int32_t)(5))})};
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
3 warnings and 1 error generated.
```

On the other hand, it does compile with GCC, although it produces warnings when `-Wpedantic` is enabled:

```
$ gcc -c array.c -o array.o -Wpedantic
array.c:9:32: warning: initializer element is not constant [-Wpedantic]
    9 | static int32_t s0[(3)][(2)] = {((int32_t [(2)]){((int32_t)(0)), ((int32_t)(1))}), ((int32_t [(2)]){((int32_t)(2)), ((int32_t)(3))}), ((int32_t [(2)]){((int32_t)(4)), ((int32_t)(5))})};
      |                                ^
array.c:9:32: note: (near initialization for ‘s0[0]’)
array.c:9:83: warning: initializer element is not constant [-Wpedantic]
    9 | static int32_t s0[(3)][(2)] = {((int32_t [(2)]){((int32_t)(0)), ((int32_t)(1))}), ((int32_t [(2)]){((int32_t)(2)), ((int32_t)(3))}), ((int32_t [(2)]){((int32_t)(4)), ((int32_t)(5))})};
      |                                                                                   ^
array.c:9:83: note: (near initialization for ‘s0[1]’)
array.c:9:134: warning: initializer element is not constant [-Wpedantic]
    9 | (2)] = {((int32_t [(2)]){((int32_t)(0)), ((int32_t)(1))}), ((int32_t [(2)]){((int32_t)(2)), ((int32_t)(3))}), ((int32_t [(2)]){((int32_t)(4)), ((int32_t)(5))})};
      |                                                                                                               ^

array.c:9:134: note: (near initialization for ‘s0[2]’)
```

robdockins, 2021-11-04:
I wonder if this is related to https://github.com/Copilot-Language/copilot/issues/223.  I think maybe the type casts are just unnecessary?

RyanGlScott, 2021-11-04:
Some of the casts are deliberate, if I've understood https://github.com/Copilot-Language/copilot-c99/pull/25 correctly. That being said, I don't know if the `(int32_t [(2)])` casts really need to be there. (I've been working around this issue locally by hacking `copilot-c99` to avoid generating those casts specifically.)

RyanGlScott, 2021-11-05:
Specifically, I'm working around the issue by applying this hack in `copilot-c99` to avoid the use of `InitVal` in the `Array` case of `constty`, as `InitVal`s get pretty-printed with casts (like the `(int32_t [(2)])` seen above):

```diff
diff --git a/copilot-c99/src/Copilot/Compile/C99/CodeGen.hs b/copilot-c99/src/Copilot/Compile/C99/CodeGen.hs
index c172660..2559244 100644
--- a/copilot-c99/src/Copilot/Compile/C99/CodeGen.hs
+++ b/copilot-c99/src/Copilot/Compile/C99/CodeGen.hs
@@ -47,7 +47,12 @@ mkbuffdecln sid ty xs = C.VarDecln (Just C.Static) cty name initvals where
   name     = streamname sid
   cty      = C.Array (transtype ty) (Just $ C.LitInt $ fromIntegral buffsize)
   buffsize = length xs
-  initvals = Just $ C.InitArray $ map (C.InitExpr . constty ty) xs
+  initvals = Just $ C.InitArray $ constarray ty xs
+
+  constarray :: Type a -> [a] -> [C.Init]
+  constarray ty xs = case ty of
+    Array ty' -> map (C.InitArray . constarray ty' . arrayelems) xs
+    _         -> map (C.InitExpr . constty ty) xs
 
 -- | Make a C index variable and initialise it to 0.
 mkindexdecln :: Id -> C.Decln
```

It's a bit heavy-handed, but it does result in fewer casts, which is enough to make Clang happy:

```c
static int32_t s0[(3)][(2)] = {{((int32_t)(0)), ((int32_t)(1))}, {((int32_t)(2)), ((int32_t)(3))}, {((int32_t)(4)), ((int32_t)(5))}};
```

RyanGlScott, 2021-11-11:
Curiously, `clang` does not reject the example above if an array of `Bool`s is used instead of an array of `Int32`. However, that doesn't necessarily mean that it works correctly at runtime. Some further digging reveals that arrays with `(bool [(2)]) { ... }` initializers have very strange behavior. Consider this program:

```c
#include <stdbool.h>
#include <stdio.h>

bool sa[3][2] = {((bool [2]){true, false}), ((bool [2]){true, true}), ((bool [2]){false, false})};
bool sb[3][2] = {           {true, false},             {true, true},             {false, false}};

int main() {
  printf("%d\n", sa[0][1]);
  printf("%d\n", sb[0][1]);
  return 0;
}
```

This demonstrates two streams, `sa` and `sb`. The `sa` stream is essentially what `copilot-c99` will currently produce for an `Array 2 Bool`. The `sb` stream removes the `(bool [2])` casts, but is otherwise identical. What happens when we access the `[0][1]`th elements of each stream? Shockingly, `clang` will return different values from each stream:

```
$ clang test.c -o test -Wall
test.c:4:18: warning: suggest braces around initialization of subobject [-Wmissing-braces]
bool sa[3][2] = {((bool [2]){true, false}), ((bool [2]){true, true}), ((bool [2]){false, false})};
                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                 {                                                  }
test.c:4:71: warning: suggest braces around initialization of subobject [-Wmissing-braces]
bool sa[3][2] = {((bool [2]){true, false}), ((bool [2]){true, true}), ((bool [2]){false, false})};
                                                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~
                                                                      {                         }
2 warnings generated.

$ ./test 
1
0
```

This happens regardless of which version of Clang (or what optimization levels) I use, so I don't think this is necessarily a Clang bug. Again, applying the patch in https://github.com/Copilot-Language/copilot/issues/276#issuecomment-961914824 works around the issue.

ivanperez-keera, 2021-12-01:
What you are seeing here:

```C
bool sa[3][2] = {((bool [2]){true, false}), ((bool [2]){true, true}), ((bool [2]){false, false})};
```

is a combination of two things:

First, the compiler is converting `((bool [2]){true, false})` into a `bool`. Since it's a non-null pointer (to an array with two elements), it becomes `true`.

Second, the array should have 6 elements, but only 3 are provided (3 arrays of `bool`s converted to `bool`s). So it's "underinitialized". By C's semantics, the rest are initialized to zero, which is `false`.

If you run a slightly modified version of this program with GDB, you see the following:

```C
(gdb) li
1   #include <stdbool.h>
2   #include <stdio.h>
3   
4   bool sa1[2] = (bool [2]){true,  false};
5   bool sa2[2] = (bool [2]){false, false};
6   bool sa3[2] = (bool [2]){false, false};
7   
8   bool sa[3][2] = {sa1, sa2, sa3};
9   
10  int main() {
(gdb) print sa1
$1 = {true, false}
(gdb) print sa2
$2 = {false, false}
(gdb) print sa3
$3 = {false, false}
(gdb) print sa
$4 = {{true, true}, {true, false}, {false, false}}
```

The behavior is the same even if you inline the definitions of `sa1`, `sa2`, `sa3` by hand. This also shows that the true that you are seeing in that print in your original program is just a pointer converted into a bool.

I don't think this is a bug with LLVM. It's just how array initialization works and LLVM's specific implementation.

GCC gives the same result as LLVM for my modified version, but if you inline  `sa1`, `sa2`, `sa3` then it interprets the arrays as "nested" and not implicitly converted pointers.

ivanperez-keera, 2022-04-04:
@fdedden any progress on this?

ivanperez-keera, 2022-04-15:
@RyanGlScott  so I have been working on this and I see that the problem is not exclusive to nested arrays.

If one introduces a delay in a stream of structs, GCC in pedantic mode also reports a problem (but Clang doesn't).

The compiler considers that casting to a struct is also not constant and therefore not a valid value for a static variable.

For example, consider the following spec:
```Haskell
{-# LANGUAGE DataKinds        #-}
{-# LANGUAGE RebindableSyntax #-}
module Main where

import Copilot.Compile.C99
import Language.Copilot

data SampleStruct = SampleStruct
  { sampleStructNum :: Field "sample_struct_num" Word16
  }

instance Struct SampleStruct where
  typename _ = "sample_struct_t"
  toValues sampleStruct = [ Value Word16 (sampleStructNum sampleStruct) ]

instance Typed SampleStruct where
  typeOf = Struct (SampleStruct (Field 0))

main :: IO ()
main = do
  spec' <- reify spec
  compile "staticarraynonconstant" spec'

spec :: Spec
spec = do
  trigger "handlerA" true [arg arr]
  trigger "handlerS" true [arg v1]

arr :: Stream (Array 1 Int32)
arr = [ array [1] ] ++ arr

v1 :: Stream SampleStruct
v1 = [ SampleStruct (Field 0) ] ++ v1
```

GCC still complains:
```
$ gcc -Wpedantic -Werror -c staticarraynonconstant.c 
staticarraynonconstant.c:13:42: error: initializer element is not constant [-Werror=pedantic]
   13 | static struct sample_struct_t s1[(1)] = {((struct sample_struct_t){((uint16_t)(0))})};
      |                                          ^
staticarraynonconstant.c:13:42: note: (near initialization for ‘s1[0]’)
cc1: all warnings being treated as errors
```

What I see from your PR is that you have treated nested arrays as a special case, but because structs are not dealt with, the arrays within structs also do not work, which are a use case of Copilot. As an example, https://github.com/Copilot-Language/copilot/blob/master/copilot/examples/Structs.hs, has a struct with an array of fixed length in it and, although it does not have a delay, it absolutely makes sense in some applications to have it.

A minimal reproducible example follows:
```haskell
{-# LANGUAGE DataKinds        #-}
{-# LANGUAGE RebindableSyntax #-}
module Main where

import Copilot.Compile.C99
import Language.Copilot

data SampleStruct = SampleStruct
  { sampleStructNum :: Field "sample_struct_num" (Array 1 Int32)
  }

instance Struct SampleStruct where
  typename _ = "sample_struct_t"
  toValues sampleStruct = [ Value typeOf (sampleStructNum sampleStruct) ]

instance Typed SampleStruct where
  typeOf = Struct (SampleStruct (Field (array [1])))

main :: IO ()
main = do
  spec' <- reify spec
  compile "staticarraynonconstant" spec'

spec :: Spec
spec = do
  trigger "handlerA" true [arg arr]
  trigger "handlerS" true [arg v1]

arr :: Stream (Array 1 Int32)
arr = [ array [1] ] ++ arr

v1 :: Stream SampleStruct
v1 = [ SampleStruct (Field (array [1])) ] ++ v1
```

ivanperez-keera, 2022-04-15:
@RyanGlScott let me know what you think. If it's not obvious how to solve it for structs, we could split this into two issues and deal with structs later.

RyanGlScott, 2022-04-15:
Good catch! I'd overlooked the fact that struct initializers pose the same challenges as array initializers. Luckily, I don't think it would be too difficult to accommodate structs. We'd just need to:

* Instead of generating `{((struct sample_struct_t){((uint16_t)(0))})}`, generate `{{(uint16_t)(0)}}`. That is, we need to the same initializer syntax for nested structs as for nested arrays. (This means we'd need to use `InitArray` for structs, somewhat confusingly, but that's just how `language-c99-simple` works.)
* Moreover, when we encounter a nested struct, we need to recursively apply the same treatment to each of its fields.

With those goals mind, here is an alternative patch:

```diff
diff --git a/copilot-c99/src/Copilot/Compile/C99/CodeGen.hs b/copilot-c99/src/Copilot/Compile/C99/CodeGen.hs
index 9b9ea8d..103d0b0 100644
--- a/copilot-c99/src/Copilot/Compile/C99/CodeGen.hs
+++ b/copilot-c99/src/Copilot/Compile/C99/CodeGen.hs
@@ -45,7 +45,7 @@ mkbuffdecln sid ty xs = C.VarDecln (Just C.Static) cty name initvals where
   name     = streamname sid
   cty      = C.Array (transtype ty) (Just $ C.LitInt $ fromIntegral buffsize)
   buffsize = length xs
-  initvals = Just $ C.InitArray $ map (C.InitExpr . constty ty) xs
+  initvals = Just $ C.InitArray $ constarray ty xs
 
 -- | Make a C index variable and initialise it to 0.
 mkindexdecln :: Id -> C.Decln
diff --git a/copilot-c99/src/Copilot/Compile/C99/Translate.hs b/copilot-c99/src/Copilot/Compile/C99/Translate.hs
index 1d6fc39..45aaf95 100644
--- a/copilot-c99/src/Copilot/Compile/C99/Translate.hs
+++ b/copilot-c99/src/Copilot/Compile/C99/Translate.hs
@@ -179,17 +179,27 @@ constty ty = case ty of
   Word64 -> explicitty ty . C.LitInt . fromIntegral
   Float  -> explicitty ty . C.LitFloat
   Double -> explicitty ty . C.LitDouble
-  Struct _ -> \v -> C.InitVal (transtypename ty) (map fieldinit (toValues v))
-    where
-      fieldinit (Value ty (Field val)) = C.InitExpr $ constty ty val
+  Struct _ -> \v -> C.InitVal (transtypename ty) (map constfieldinit (toValues v))
   Array ty' -> \v -> C.InitVal (transtypename ty) (vals v)
     where
       vals v = constarray ty' (arrayelems v)
 
-      constarray :: Type a -> [a] -> [C.Init]
-      constarray ty xs = case ty of
-        Array ty' -> constarray ty' (concatMap arrayelems xs)
-        _         -> map (C.InitExpr . constty ty) xs
+-- | Transform a Copilot Core literal, based on its value and type, into a C99
+-- initializer.
+constinit :: Type a -> a -> C.Init
+constinit ty val = case ty of
+  Array ty' -> C.InitArray $ constarray ty' $ arrayelems val
+  Struct _  -> C.InitArray $ map constfieldinit $ toValues val
+  _         -> C.InitExpr $ constty ty val
+
+-- | Transform a Copilot Core struct field into a C99 initializer.
+constfieldinit :: Value a -> C.Init
+constfieldinit (Value ty (Field val)) = constinit ty val
+
+-- | Transform a Copilot Array, based on the element values and their type,
+-- into a list of C99 initializer values.
+constarray :: Type a -> [a] -> [C.Init]
+constarray ty = map (constinit ty)
 
 
 -- | Explicitly cast a C99 value to a type.
```

The `constinit` function, which includes special cases for both nested `Array`s and `Struct`s, is where the action happens.

I've tested this patch on both of the test cases in https://github.com/Copilot-Language/copilot/issues/276#issuecomment-1100158705 by compiling the resulting C code with `clang -Wpedantic` and `gcc -Wpedantic`, and they both compile without warnings. Obviously, `constinit` needs some more comments, but do you think this approach will work?

ivanperez-keera, 2022-04-15:
This seems to work. I can accept a patch along these lines.

It seems like `language-c99` is lacking a specialized construct for struct initialization. In this case, it works because they are written the same way and the meaning carries across (i.e., enumeration of values in a product type).

However, structs can also be initialized by naming the fields, so the struct initialization expressions and an array initialization expressions are conceptually different. Pinging @fdedden about this; perhaps something must change in `language-c99` in the future.

Please leave a note in the code explaining why we are using `InitArray`, and make sure all still works for structs of structs and nested arrays.

Thanks!

RyanGlScott, 2022-04-15:
Sounds good to me. I've submitted the patch in https://github.com/Copilot-Language/copilot/issues/276#issuecomment-1100261056, along with some more notes about how it works, in #310.

ivanperez-keera, 2022-04-19:
Thanks. I've been working on integrating this. It should be in soon.

ivanperez-keera, 2022-04-20:
**Description**

The translation into C99 of specs with delays applied to streams of arrays or structs generates global C variables with non-constant initializations and possibly incorrect semantics.

This is due to an interplay of two facts: 1) `copilot-c99` produces code with explicit casts in the values of initialized buffers, needed to hold values of delayed streams, and 2) even thought the sizes match, casting to an array of fixed length, or a struct, in C99 is not considered a constant expression. As a result, the compiler considers that the global variable holding the buffer is being initialized to a non-constant value, and fails to compile.

For example, with a declaration such as:
```haskell
arr :: Stream (Array 2 Int32)
arr = [ array [0, 1] ] ++ arr
```
the generated C code includes:
```C
static int32_t s0[(1)][(2)] = {((int32_t [(2)]){((int32_t)(0)), ((int32_t)(1))})};
```
which, upon compilation, results in an error (and a warning):
```
$ clang array.c -c
array.c:9:32: warning: incompatible pointer to integer conversion initializing 'int32_t' (aka 'int') with an expression of type 'int32_t [2]' [-Wint-conversion]
static int32_t s0[(1)][(2)] = {((int32_t [(2)]){((int32_t)(0)), ((int32_t)(1))})};
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
array.c:9:32: error: initializer element is not a compile-time constant
static int32_t s0[(1)][(2)] = {((int32_t [(2)]){((int32_t)(0)), ((int32_t)(1))})};
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 warning and 1 error generated.
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

Note that this occurs also with structs in GCC with pedantic mode enabled, but not Clang.

Furthermore, in some cases, an explicit cast to a fixed-length array of values may being considered a cast to an integer (see warning above), and information may be lost.

**Type**

- Bug: the C99 backend generates incorrect array initialization expressions.

**Additional context**

None

**Requester**

- Ryan Scott (@RyanGlScott), Galois Inc.

**Method to check presence of bug**

The following Copilot specification will generate code that fails to compile with Clang, or with GCC if using `-Wpedantic -Werror`:

```Haskell
{-# LANGUAGE DataKinds        #-}
{-# LANGUAGE RebindableSyntax #-}
module Main where

import Copilot.Compile.C99
import Language.Copilot

data SampleStruct = SampleStruct
  { sampleStructNum :: Field "sample_struct_num" Word16
  }

instance Struct SampleStruct where
  typename _ = "sample_struct_t"
  toValues sampleStruct = [ Value Word16 (sampleStructNum sampleStruct) ]

instance Typed SampleStruct where
  typeOf = Struct (SampleStruct (Field 0))

main :: IO ()
main = do
  spec' <- reify spec
  compile "staticarraynonconstant" spec'

spec :: Spec
spec = do
  trigger "handlerA" true [arg arr]
  trigger "handlerS" true [arg v1]

arr :: Stream (Array 1 Int32)
arr = [ array [1] ] ++ arr

v1 :: Stream SampleStruct
v1 = [ SampleStruct (Field 0) ] ++ v1
```

**Expected result**

The specification above should generate C99 code with global variable initializers that compiles without errors.

**Desired result**

The specification above should generate C99 code with global variable initializers that compiles without errors.

**Proposed solution**

Modify the code that initializes expressions so that arrays and structs are not explicitly cast.

**Further notes**

This Change Request summary borrows from @RyanGlScott message, commit summaries, and pull request.

ivanperez-keera, 2022-04-20:
Change Manager: Confirmed that the bug is present by using method described and examining the result.

ivanperez-keera, 2022-04-20:
Technical Lead: Confirmed that the bug should be addressed.

ivanperez-keera, 2022-04-20:
Technical Lead: Bug scheduled for fixing in Copilot 3.9.

Fix assigned to: @ivanperez-keera.

ivanperez-keera, 2022-04-20:
Change Manager: Verified that:
- Solution is implemented:
  - [X] The code proposed compiles. Details:
    Build log: https://app.travis-ci.com/github/Copilot-Language/copilot/builds/249635363
  - [X] The code proposed passes the tests. Details:
    Docker image:
    ```Dockerfile
    FROM ubuntu:trusty

    RUN apt-get update

    RUN apt-get install --yes software-properties-common
    RUN add-apt-repository ppa:hvr/ghc
    RUN apt-get update

    RUN apt-get install --yes ghc-8.6.5 cabal-install-2.4
    RUN apt-get install --yes libz-dev

    ENV PATH=/opt/ghc/8.6.5/bin:/opt/cabal/2.4/bin:$PWD/.cabal-sandbox/bin:$PATH

    RUN cabal update
    RUN cabal v1-sandbox init
    RUN cabal v1-install alex happy
    RUN apt-get install --yes git
    
    CMD git clone $REPO && cd $NAME && git checkout $COMMIT && cd .. && cabal v1-install copilot/copilot**/ --enable-tests --run-tests
    ```
    Command:
    ```
    $ docker run -e "REPO=https://github.com/copilot-language/copilot" -e "NAME=copilot" -e "COMMIT=716089e9368566ec94c8504a91778331728b3845" -it copilot-test
    ```

  - [X] The solution proposed produces the expected result: it fails to compile with the prior version of Copilot, showing warnings and errors due to the explicit casts, and compiles correctly with the new version, both with Clang and GCC.
    Docker image and auxiliary files:
    ```Dockerfile
    --- Dockerfile-verify-276
    FROM ubuntu:trusty

    RUN apt-get update

    RUN apt-get install --yes software-properties-common
    RUN add-apt-repository ppa:hvr/ghc
    RUN apt-get update

    RUN apt-get install --yes ghc-8.6.5 cabal-install-2.4
    RUN apt-get install --yes libz-dev

    ENV PATH=/opt/ghc/8.6.5/bin:/opt/cabal/2.4/bin:$PWD/.cabal-sandbox/bin:$PATH

    RUN cabal update
    RUN cabal v1-sandbox init
    RUN cabal v1-install alex happy
    RUN apt-get install --yes git

    RUN apt-get install --yes clang

    ADD Stream.hs Stream.hs

    SHELL ["/bin/bash", "-c"]
    CMD git clone $REPO && cd $NAME && git checkout $COMMIT && cd .. \
        && cabal v1-install copilot/copilot**/ \
        && cabal v1-exec -- runhaskell Stream.hs \
        && set -x \
        && gcc -Werror -Wpedantic -c initialization.c \
        && rm initialization.o \
        && clang -Werror -Wpedantic -c initialization.c

    --- Stream.hs
    {-# LANGUAGE DataKinds        #-}
    {-# LANGUAGE RebindableSyntax #-}
    module Main where

    import Copilot.Compile.C99
    import Language.Copilot

    -- * Auxiliary types

    -- | Nested struct
    data SampleStruct = SampleStruct
      { sampleStructField :: Field "sample_struct_field" SampleStruct2
      }

    instance Struct SampleStruct where
      typename _ = "sample_struct_t"
      toValues sampleStruct = [ Value typeOf (sampleStructField sampleStruct) ]

    instance Typed SampleStruct where
      typeOf = Struct (SampleStruct (Field (SampleStruct2 (Field 1))))

    -- | Plain struct
    data SampleStruct2 = SampleStruct2
      { sampleStruct2Num :: Field "sample_struct2_num" Int32
      }

    instance Struct SampleStruct2 where
      typename _ = "sample_struct2_t"
      toValues sampleStruct2 = [ Value typeOf (sampleStruct2Num sampleStruct2) ]

    instance Typed SampleStruct2 where
      typeOf = Struct (SampleStruct2 (Field 1))

    -- | Struct with an array in it
    data StructWithArray = StructWithArray
      { structWithArrayNum :: Field "struct_with_array_num" (Array 1 Int32)
      }

    instance Struct StructWithArray where
      typename _ = "struct_with_array_t"
      toValues structWithArray =
        [ Value typeOf (structWithArrayNum structWithArray) ]

    instance Typed StructWithArray where
      typeOf = Struct (StructWithArray (Field (array [1])))

    main :: IO ()
    main = do
      spec' <- reify spec
      compile "initialization" spec'

    spec :: Spec
    spec = do
      trigger "handlerArray"        true  [arg strArray]
      trigger "handlerStruct"       true  [arg strStruct]
      trigger "handlerNestedStruct" true  [arg strNestedStruct]
      trigger "handlerNestedArray"  true  [arg strNestedArray]

    strArray :: Stream Int32
    strArray = [ 1 ] ++ strArray

    strNestedArray :: Stream (Array 2 (Array 2 Int32))
    strNestedArray = [ array [array [0, 1], array [2, 3]] ] ++ strNestedArray

    strStruct :: Stream SampleStruct2
    strStruct = [ SampleStruct2 (Field 1) ] ++ strStruct

    strNestedStruct :: Stream SampleStruct
    strNestedStruct =
      [ SampleStruct (Field (SampleStruct2 (Field 1))) ] ++ strNestedStruct

    strStructWithArray :: Stream StructWithArray
    strStructWithArray =
      [ StructWithArray (Field (array [55])) ] ++ strStructWithArray
    ```
    Command:
    ```
    $ docker run -e "REPO=https://github.com/copilot-language/copilot" -e "NAME=copilot" -e "COMMIT=716089e9368566ec94c8504a91778331728b3845" -it copilot-verify-276
    ```
- [X] Implementation is documented. Details:
  No changes to the public API; new implementation is documented well; all new functions include haddock documentation.
- [X] Change history is clear.
- [X] Commit messages are clear.

ivanperez-keera, 2022-04-20:
Change Manager: Implementation ready to be merged.

commit 76603ee8 (2022-04-20): Merge branch 'develop-fix-bufferInitnoExplicitCast'. Close #276.

**Description**

The translation into C99 of specs with delays applied to streams of
arrays or structs generates global C variables with non-constant
initializations and possibly incorrect semantics.

This is due to an interplay of two facts: 1) `copilot-c99` produces code
with explicit casts in the values of initialized buffers, needed to hold
values of delayed streams, and 2) even thought the sizes match, casting
to an array of fixed length, or a struct, in C99 is not considered a
constant expression. As a result, the compiler [... cut]
closed 2022-04-20 by ivanperez-keera