Copilot-Language/copilot #277: copilot-core-3.6 fails to build with GHC 9.2.1

sjakobi, 2021-11-13:
```
src/Copilot/Core/Type/Equality.hs:20:15: warning: [-Wstar-is-type]
    Using ‘*’ (or its Unicode variant) to mean ‘Data.Kind.Type’
    relies on the StarIsType extension, which will become
    deprecated in the future.
    Suggested fix: use ‘Type’ from ‘Data.Kind’ instead.
   |
20 | data Equal :: * -> * -> * where
   |               ^

src/Copilot/Core/Type/Equality.hs:20:20: warning: [-Wstar-is-type]
    Using ‘*’ (or its Unicode variant) to mean ‘Data.Kind.Type’
    relies on the StarIsType extension, which will become
    deprecated in the future.
    Suggested fix: use ‘Type’ from ‘Data.Kind’ instead.
   |
20 | data Equal :: * -> * -> * where
   |                    ^

src/Copilot/Core/Type/Equality.hs:20:25: warning: [-Wstar-is-type]
    Using ‘*’ (or its Unicode variant) to mean ‘Data.Kind.Type’
    relies on the StarIsType extension, which will become
    deprecated in the future.
    Suggested fix: use ‘Type’ from ‘Data.Kind’ instead.
   |
20 | data Equal :: * -> * -> * where
   |                         ^
[ 5 of 22] Compiling Copilot.Core.Type.Dynamic ( src/Copilot/Core/Type/Dynamic.hs, dist/build/Copilot/Core/Type/Dynamic.o, dist/build/Copilot/Core/Type/Dynamic.dyn_o )

src/Copilot/Core/Type/Dynamic.hs:32:18: warning: [-Wstar-is-type]
    Using ‘*’ (or its Unicode variant) to mean ‘Data.Kind.Type’
    relies on the StarIsType extension, which will become
    deprecated in the future.
    Suggested fix: use ‘Type’ from ‘Data.Kind’ instead.
   |
32 | data Dynamic :: (* -> *) -> * where
   |                  ^

src/Copilot/Core/Type/Dynamic.hs:32:23: warning: [-Wstar-is-type]
    Using ‘*’ (or its Unicode variant) to mean ‘Data.Kind.Type’
    relies on the StarIsType extension, which will become
    deprecated in the future.
    Suggested fix: use ‘Type’ from ‘Data.Kind’ instead.
   |
32 | data Dynamic :: (* -> *) -> * where
   |                       ^

src/Copilot/Core/Type/Dynamic.hs:32:29: warning: [-Wstar-is-type]
    Using ‘*’ (or its Unicode variant) to mean ‘Data.Kind.Type’
    relies on the StarIsType extension, which will become
    deprecated in the future.
    Suggested fix: use ‘Type’ from ‘Data.Kind’ instead.
   |
32 | data Dynamic :: (* -> *) -> * where
   |                             ^

src/Copilot/Core/Type/Dynamic.hs:36:19: warning: [-Wstar-is-type]
    Using ‘*’ (or its Unicode variant) to mean ‘Data.Kind.Type’
    relies on the StarIsType extension, which will become
    deprecated in the future.
    Suggested fix: use ‘Type’ from ‘Data.Kind’ instead.
   |
36 | data DynamicF :: (* -> *) -> (* -> *) -> * where
   |                   ^

src/Copilot/Core/Type/Dynamic.hs:36:24: warning: [-Wstar-is-type]
    Using ‘*’ (or its Unicode variant) to mean ‘Data.Kind.Type’
    relies on the StarIsType extension, which will become
    deprecated in the future.
    Suggested fix: use ‘Type’ from ‘Data.Kind’ instead.
[... 195 lines cut ...]

src/Copilot/Core/Locals.hs:21:1: warning: [-Wdodgy-imports]
    Module ‘Copilot.Core’ does not export ‘toList’
   |
21 | import Copilot.Core hiding (toList)
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[19 of 22] Compiling Copilot.Core.Interpret.Eval ( src/Copilot/Core/Interpret/Eval.hs, dist/build/Copilot/Core/Interpret/Eval.o, dist/build/Copilot/Core/Interpret/Eval.dyn_o )

src/Copilot/Core/Interpret/Eval.hs:239:3: error:
    Ambiguous occurrence ‘And’
    It could refer to
       either ‘Copilot.Core.And’,
              imported from ‘Copilot.Core’ at src/Copilot/Core/Interpret/Eval.hs:17:1-37
              (and originally defined in ‘Copilot.Core.Operators’
                 at src/Copilot/Core/Operators.hs:65:3-32)
           or ‘Data.Bits.And’,
              imported from ‘Data.Bits’ at src/Copilot/Core/Interpret/Eval.hs:26:1-16
    |
239 |   And          -> (&&)
    |   ^^^
```

v3.0 and up show the same errors.

As a Hackage trustee, I have revised these versions in order to prevent users from encountering these errors. See e.g. https://hackage.haskell.org/package/copilot-core-3.6/revisions/.

ivanperez-keera, 2021-11-13:
I see.

Thanks for reporting this!

As of currently, versions 9.2 and above are not supported. I'll modify the cabal packages to indicate so.

However, and even though the trustee role gives you that power at hackage level, we'd like to retain full control of all changes to all copilot code and packages, so I'd like to ask that you do not revise/edit the cabal packages and always let us know first. Thanks!

ivanperez-keera, 2021-11-13:
@sjakobi Just to clarify: I'd like to explicitly ask that you remove that revision. We'll fix it on our end and push an update. Thanks.

sjakobi, 2021-11-13:
@ivanperez-keera Apologies! If I had known that `copilot-core` should not be curated by Hackage trustees, I would not have applied these revisions.

I have undone my changes by applying another set of revisions that removes the constraints on `base` I had added in v3.0 to v3.6. See e.g. https://hackage.haskell.org/package/copilot-core-3.6/revisions/.

As far as I know, revisions cannot actually be "removed", similarly to how packages cannot be deleted once they have been uploaded. I will check back with the more experienced trustees on this though.

I'm not sure how to best prevent future edits by Hackage trustees in `copilot-core`. I had believed that the special `x-curation` field documented at https://hackage.haskell.org/upload could be used for this purpose, but on re-reading the section it seems that the `uncurated` setting does not guarantee this:

> Package uploaders may choose to exclude individual package uploads from curation, by setting the x-curation: field of the package's cabal file to uncurated. Packages which are uncurated have no expectations on them regarding versioning policy. Trustees or maintainers may adopt uncurated packages into the curated layer through metadata revisions. Metadata revisions must not set the value of the x-curation field to any variant of uncurated.

I will ask my more experienced colleagues for clarification on this.

ivanperez-keera, 2021-11-13:
@sjakobi thank you so much for your very quick response and for your understanding. I really appreciate it!

I sent an email around to hackage-admin so they know which packages we'd prefer not altered.

_More generally:_

I also took a quick look at the process and it seems that there is no current established mechanism to specify this.

If action to help package maintainers avoid these changes must be taken more widely and in the short term, I'm not sure if the cabal file itself should be the best place to specify it.

Additionally, it seems to me that adding and removing fields from the cabal package as we experiment with these decisions will lead to frequent breakage of backwards compatibility, which has a larger negative effect and severely diminishes Haskell position as a robust language.

While the proposal and its suitability more widely is still under evaluation, I'd recommend instead a setting we modify on Hackage directly via the web interface. Once we know this is a good mechanism and that it works well, we can flow that down to the right place knowing that the change is here to stay, be it to a new field in the cabal files or something else.

ivanperez-keera, 2021-11-13:
> I sent an email around to hackage-admin so they know which packages we'd prefer not altered.

I don't know if that's the right way to go, so I would nevertheless appreciate it if you could ask on your end too. Thanks!

andreasabel, 2021-11-13:
NB: The reported breakage seems to be due to the following addition to `base`: https://hackage.haskell.org/package/base-4.16.0.0/docs/Data-Bits.html#t:And

ivanperez-keera, 2021-11-13:
Indeed. Thanks @andreasabel . The right solution to this would be to use explicit imports (potentially qualifying some and hiding others), which is in the very near future of Copilot.

sjakobi, 2021-11-14:
@ivanperez-keera I've briefly discussed the issue with some other trustees. To clearly signal that a package should not be modified by trustees you can indeed add the stanza

```
x-curation: uncurated
``` 

to its .cabal file.

ivanperez-keera, 2021-11-14:
Thanks, @sjakobi. Is this shown to users on the hackage package page in any way? (_uncurated_ sounds bad and I'm concerned about the perception).

sjakobi, 2021-11-14:
@ivanperez-keera Judging by the hackage-server code: No, it's not shown.

closed 2022-01-07 by ivanperez-keera