Copilot-Language/copilot #278: `copilot-c99`: Codegen for `signum` has the wrong semantics for `0` and `NaN`

RyanGlScott, 2021-11-23:
This program:

```hs
{-# LANGUAGE NoImplicitPrelude #-}
module Main where

import Copilot.Compile.C99
import Language.Copilot

spec :: Spec
spec = do
  let stream :: Stream Int32
      stream = extern "stream" (Just [-2, -1, 0, 1, 2])

  trigger "func" true [arg (signum stream)]

main :: IO ()
main = do
  spec' <- reify spec
  compile "signum" spec'
  interpret 5 spec
```

Will produce this output:

```
func: 
(-1)  
(-1)  
(0)   
(1)   
(1)
```

On the other hand, the corresponding C code is:

```c
#include <stdint.h>
#include <stdbool.h>
#include <string.h>
#include <stdlib.h>
#include <math.h>

#include "signum.h"

static int32_t stream_cpy;

bool func_guard(void) {
  return true;
}

int32_t func_arg0(void) {
  return (copysign)((1.0), (stream_cpy));
}

void step(void) {
  (stream_cpy) = (stream);
  if ((func_guard)()) {
    {(func)(((func_arg0)()));}
  };
}
```

The use of `(copysign)((1.0), (stream_cpy))` is suspect, as it will return a different answer from the Copilot spec when `stream_cpy` is `0`. This C program demonstrates the difference:

```c
#include <math.h>
#include <stdio.h>
#include <stdint.h>

int32_t signum(int32_t x) {
  return (copysign)((1.0), (x));
}

int main() {
  printf("%d\n", signum(-2));
  printf("%d\n", signum(-1));
  printf("%d\n", signum(0));
  printf("%d\n", signum(1));
  printf("%d\n", signum(2));
  return 0;
}
```

This will output:

```
-1
-1
1
1
1
```

Note that it outputs `1` when the argument is `0`, whereas the Haskell version outputs `0` when the output is `0`.

A similar bug exists for `-0.0` and `NaN` values when `signum` is invoked on `Stream`s of `Float`s or `Double`s. In Haskell, `signum (-0.0) = -0.0` and `signum NaN = NaN`. With the `copysign(1.0, x)` approach above, however, the C code will return `-1.0` when the argument to `signum` is `-0.0`, and `±1.0` when the argument is `NaN` (depending on the sign bit of the `NaN`).

ivanperez-keera, 2022-01-13:
So, the reason why I did not merge this straight away is because it's not immediately clear to me which one of the two semantics is The Correct One™.

Do we want Copilot's semantics to follow what `copysign` does, or do we want the C99 backend to implement what the interpreter is doing (even with an ad hoc function)?

RyanGlScott, 2022-01-13:
I don't have a strong opinion on which approach to take. If Copilot's `Sign` operator is intended to reflect the semantics of C's `copysign` function, however, it seems like the implementation of `signum` should be adjusted.

ivanperez-keera, 2022-01-13:
Exactly!

ivanperez-keera, 2022-01-13:
If you want, just wait a sec until we confirm. You may save yourself some time.

ivanperez-keera, 2022-01-13:
Picking the C99 copysign as reference will seem familiar to C99 programmers. As we add more backends, the importance will be "diluted".

I've just spent 1h down the rabbit hole of sign implementation functions in C and there is no agreement. Some people like copysign. Some hate it. Some say it's slow. Even code that is often quoted as equivalent, like: `x > 0 ? 1 : (x < 0 ? -1 : x)` and `(x > 0) - (0 < x)` isn't equivalent for NaN. I wonder if anyone ever needs signum of NaN.

I think I'm OK accepting this implementation as provided in the PR (meaning that Copilot/Haskell determines the semantics). I suspect we'll need more knowledge to find any corner cases for which this is inconvenient, and, if that happens, we may need to change the language.

ivanperez-keera, 2022-01-14:
By the way: why are you preferring `genericLength` over just `length`?

ivanperez-keera, 2022-01-14:
The following should serve as an independent, reproducible regression test that both confirms the issue and shows that it is solved by @RyanGlScott 's PR.

```
--- Dockerfile-278
FROM ubuntu:trusty

RUN apt-get update

RUN apt-get install --yes software-properties-common
RUN add-apt-repository ppa:hvr/ghc
RUN apt-get update

RUN apt-get install --yes ghc-8.6.5 cabal-install-2.4
RUN apt-get install --yes libz-dev

ENV PATH=/opt/ghc/8.6.5/bin:/opt/cabal/2.4/bin:$PWD/.cabal-sandbox/bin:$PATH

RUN cabal update
RUN cabal v1-sandbox init
RUN cabal v1-install alex happy
RUN apt-get install --yes git

ADD main.c main.c
ADD test-278 test-278
ADD Signum.hs Signum.hs
RUN chmod a+x test-278

CMD git clone $COMMIT && cabal v1-install copilot/copilot**/ && ./test-278

--- main.c
#include <stdint.h>
#include <stdio.h>
#include "signum.h"

int32_t nums[5] = { -2, -1, 0, 1, 2 };

int32_t stream;

void func (int32_t sign)
{
  printf ("%d\n", sign);
}

int main () {
  int i = 0;
  for (i=0; i<5; i++)
  {
     stream = nums[i];
     step();
  }
  return 0;
}

--- Signum.hs
{-# LANGUAGE CPP               #-}
{-# LANGUAGE NoImplicitPrelude #-}
-- Copyright © 2021 National Institute of Aerospace / Galois, Inc.

-- | An example of a specification using the 'signum' function.
module Main where

import Data.List (genericLength)

import Copilot.Compile.C99
import Language.Copilot

nums :: [Int32]
nums = [-2, -1, 0, 1, 2]

stream :: Stream Int32
stream = extern "stream" (Just nums)

spec :: Spec
spec =
  trigger "func" true [arg (signum stream)]

main :: IO ()
main = do
  spec' <- reify spec
#ifdef INTERPRET
  csv (genericLength nums) spec
#else
  compile "signum" spec'
#endif

--- test-278
#!/bin/bash
HANDLER_NAME=func
COPILOT_SPEC=Signum.hs
COPILOT_C=signum

REG_EXP="^${HANDLER_NAME},"
SUBS_EXP="s/${REG_EXP}//g"

set -x
set -e
cabal v1-exec -- runhaskell -DINTERPRET $COPILOT_SPEC | grep -e $REG_EXP | sed -e $SUBS_EXP > output-interpreter
cabal v1-exec -- runhaskell $COPILOT_SPEC
gcc -o main main.c ${COPILOT_C}.c
./main > output-compiled
diff output-interpreter output-compiled
echo Success
```

It can be prepared with:
```bash
$ docker build -t copilot-test-278 -f Dockerfile-278  .
```

To confirm the bug, one can run:

```
$ docker run -e "COMMIT=https://github.com/copilot-language/copilot" -it copilot-test-278
...a lot of output from cabal...
+ set -e
+ cabal v1-exec -- runhaskell -DINTERPRET Signum.hs
+ grep -e '^func,'
+ sed -e 's/^func,//g'
+ cabal v1-exec -- runhaskell Signum.hs
+ gcc -o main main.c signum.c
+ ./main
+ diff output-interpreter output-compiled
3c3
< 0
---
> 1
```

To confirm the fix, one can run:

```
$ docker run -e "COMMIT=--branch dev-signum-fixCornerCases https://github.com/copilot-language/copilot" -it copilot-test-278
...a lot of output from cabal...
Completed    copilot-3.7
+ set -e
+ cabal v1-exec -- runhaskell -DINTERPRET Signum.hs
+ grep -e '^func,'
+ sed -e 's/^func,//g'
+ cabal v1-exec -- runhaskell Signum.hs
+ gcc -o main main.c signum.c
+ ./main
+ diff output-interpreter output-compiled
+ echo Success
Success
```

In the last line, `dev-signum-fixCornerCases` names a branch that identifies the fix. That line could instead clone a commit hash, etc.

ivanperez-keera, 2022-01-14:
Side note, no action needed: I don't like keeping these in comments. They are harder to use/replicate/etc. I'm considering either to create a directory in our repo to store these tests or, if it can be too much, to create a separate repo for it.

ivanperez-keera, 2022-04-12:
Change Manager: Confirmed that the bug is present by using method described and examining the result.

ivanperez-keera, 2022-04-12:
Technical Lead: Bug scheduled for fixing in Copilot 3.9.

Fix assigned to: @ivanperez-keera , borrowing from solution submitted by @RyanGlScott in #280 .

ivanperez-keera, 2022-04-14:
Implementor: solution implemented, review requested.

ivanperez-keera, 2022-04-14:
Change Manager: Verified that:
- Solution is implemented:
  - [X] The code proposed compiles. Details:
    Build log: https://app.travis-ci.com/github/Copilot-Language/copilot/builds/249437739
  - [X] The code proposed passes the tests. Details:
    Docker image:
    ```
    FROM ubuntu:trusty

    RUN apt-get update

    RUN apt-get install --yes software-properties-common
    RUN add-apt-repository ppa:hvr/ghc
    RUN apt-get update

    RUN apt-get install --yes ghc-8.6.5 cabal-install-2.4
    RUN apt-get install --yes libz-dev

    ENV PATH=/opt/ghc/8.6.5/bin:/opt/cabal/2.4/bin:$PWD/.cabal-sandbox/bin:$PATH

    RUN cabal update
    RUN cabal v1-sandbox init
    RUN cabal v1-install alex happy
    RUN apt-get install --yes git
    
    CMD git clone $REPO && cd $NAME && git checkout $COMMIT && cd .. && cabal v1-install copilot/copilot**/ --enable-tests --run-tests
    ```
    Command:
    ```
    $ docker run -e "REPO=https://github.com/copilot-language/copilot" -e "NAME=copilot" -e "COMMIT=9528d9c303043d3857d9f5b4e3557b67e6f81ead" -it copilot-test-278
    ```

  - [X] The solution proposed produces the expected result: the output of running a specification with signum is the same for the interpreter and the C99 code, especially in corner cases:
    Docker image and auxiliary files:
    ```
    --- Dockerfile-verify-278
    FROM ubuntu:trusty

    RUN apt-get update

    RUN apt-get install --yes software-properties-common
    RUN add-apt-repository ppa:hvr/ghc
    RUN apt-get update

    RUN apt-get install --yes ghc-8.6.5 cabal-install-2.4
    RUN apt-get install --yes libz-dev

    ENV PATH=/opt/ghc/8.6.5/bin:/opt/cabal/2.4/bin:$PWD/.cabal-sandbox/bin:$PATH

    RUN cabal update
    RUN cabal v1-sandbox init
    RUN cabal v1-install alex happy
    RUN apt-get install --yes git

    ADD main.c main.c
    ADD test-278 test-278
    ADD Signum.hs Signum.hs
    RUN chmod a+x test-278

    CMD git clone $REPO && cd $NAME && git checkout $COMMIT && cd .. && cabal v1-install copilot/copilot**/ && ./test-278

    --- main.c
    #include <stdint.h>
    #include <stdio.h>
    #include "signum.h"

    uint8_t numsWord8[5] = { 0, 1, 2, 3, 4 };
    uint8_t streamWord8;

    void funcWord8 (uint8_t sign)
[... 195 lines cut ...]
    MY_DIR="$( cd "$( dirname "${BASH_SOURCE[0]}" )" && pwd )"

    COPILOT_SPEC="${MY_DIR}/Signum.hs"
    COPILOT_C="${MY_DIR}/signum"
    MAIN_C="${MY_DIR}/main.c"
    MAIN="${MY_DIR}/main"

    set -x
    set -e
    cabal v1-exec -- runhaskell -DINTERPRET ${COPILOT_SPEC} | grep -ve '^Note' > output-interpreter
    cabal v1-exec -- runhaskell ${COPILOT_SPEC}
    gcc -o ${MAIN} ${MAIN_C} ${COPILOT_C}.c
    # We modify the output to limit trailing zeroes since the Haskell interpreter does that
    ./main | sed -e 's/\.0\+/.0/g' > output-compiled
    diff output-interpreter output-compiled
    echo Success
    ```
    Command:
    ```
    $ docker run -e "REPO=https://github.com/copilot-language/copilot" -e "NAME=copilot" -e "COMMIT=9528d9c303043d3857d9f5b4e3557b67e6f81ead" -it copilot-verify-278
    ```
- [X] Implementation is documented. Details:
  No need for additional documentation; it is an internal detail that does not affect the interface.
- [X] Change history is clear.
- [X] Commit messages are clear.

ivanperez-keera, 2022-04-14:
Change Manager: Implementation ready to be merged.

commit f2fd2b23 (2022-04-14): Merge branch 'develop-fix-signum'. Close #278.

**Description**

The translation of the function `signum` to C99 differs from the meaning
of the same function in the interpreter. The C99 backend uses the
function `copysign`, whose meaning differs from Haskell's `signum`.

For example, as evaluated by the interpreter, `signum 0` returns `0`,
but the C99 code generated for it (which executes `copysign(1.0, 0)`),
returns `1`.

Moreover, when dealing with floating-point numbers, `signum (-0)` and
`signum NaN` return `-0` and `NaN`, respectively, in the interpreter,
but the C99-based implementation [... cut]