Copilot-Language/copilot #294: Add support for GHC 9.x

fdedden, 2022-02-08:
As GHC 8.x is getting used less, we want to move over to 9.x, as this should really help and make it easier for people to install Copilot. We currently depend on what4 1.1, which itself isn't compatible with 9.x (https://github.com/GaloisInc/what4/pull/110).

Simply bumping the dependency of what4 a minor release up, allows Copilot to be build on at least 9.0.

ivanperez-keera, 2022-05-03:
As of currently, copilot installs with versions of what4 greater than 1.1.

It does not install with what4 1.3, but I think that's a separate issue (separate from the GHC version).

It installs just fine with cabal 3.4 and GHC 9.0 on my machine. We can still add that combination to travis.

ivanperez-keera, 2022-05-03:
And at least one of the blockers to go above GHC 9.0 is `language-c99-util`, which has a constraint on base of `<4.16`. I have not gone past that point.

ivanperez-keera, 2022-05-04:
Ok, given the fact that 9.2 and greater is not yet supported, I'd say the way to go for now is to add GHC 9.0 to the CI build.

We don't mention which GHC versions are supported anywhere in the README, so there's no need to change that.

If anything needs to change in older published cabal packages for `what4`, that's on the maintainers of that package. We don't need to disable a specific version of `what4` just because it had incorrect bounds on `base`. Instead, the right way to fix this is for the `what4` maintainers to publish a revision on hackage modifying the constraints on `base` (if they really want to -- it's totally up to them).

That would be the right way to indicate that you need a more modern version of `what4` when using GHC 9.0 or greater.

The same goes for `language-c99-util`: the current version won't compile with GHC 9.2, but that doesn't mean that we need to constrain anything on our side. Instead, the `language-c99-util` package should add a fix.

ivanperez-keera, 2022-05-07:
**Description**

A new version of GHC has been available for some time and Copilot is compatible with it. We can add it to our CI setup so that we officially support it.

**Type**

- Management: conformance with new GHC version.

**Additional context**

None.

**Requester**

- Frank Dedden.

**Method to check presence of bug**

Not applicable (not a bug).

**Expected result**

- CI script successfully compiles copilot with GHC 9.0.
 
**Desired result**

- CI script successfully compiles copilot with GHC 9.0.

**Proposed solution**

- Modify CI script to include GHC 9.0 and a suitable version of Cabal. 

**Further notes**

None.

ivanperez-keera, 2022-05-07:
Change Manager: Confirmed that the code does not meet with the style guide.

ivanperez-keera, 2022-05-07:
Technical Lead: Issue scheduled for fixing in Copilot 3.9.

Fix assigned to: @ivanperez-keera.

ivanperez-keera, 2022-05-07:
Implementor: solution implemented, review requested.

ivanperez-keera, 2022-05-07:
Change Manager: Verified that:
- Solution is implemented:
  - [X] The code proposed compiles. Details:
    Build log: https://app.travis-ci.com/github/Copilot-Language/copilot/builds/250344009
  - [X] The code proposed passes the tests. Details:
    Docker image:
    ```
    FROM ubuntu:trusty

    RUN apt-get update

    RUN apt-get install --yes software-properties-common
    RUN add-apt-repository ppa:hvr/ghc
    RUN apt-get update

    RUN apt-get install --yes ghc-8.6.5 cabal-install-2.4
    RUN apt-get install --yes libz-dev

    ENV PATH=/opt/ghc/8.6.5/bin:/opt/cabal/2.4/bin:$PWD/.cabal-sandbox/bin:$PATH

    RUN cabal update
    RUN cabal v1-sandbox init
    RUN cabal v1-install alex happy
    RUN apt-get install --yes git
    
    CMD git clone $REPO && cd $NAME && git checkout $COMMIT && cd .. && cabal v1-install copilot/copilot**/ --enable-tests --run-tests
    ```
    Command:
    ```
    $ docker run -e "REPO=https://github.com/copilot-language/copilot" -e "NAME=copilot" -e "COMMIT=dbe1e3858ef68fcfdd266870f3024fc05f8fd9d1" -it copilot-test
    ```

  - [X] The solution proposed addresses the necessary modification to the CI setup, as evidenced by the CI log mentioned at the top of this comment. Due to the nature of the issue (maintenance issue, not a bug), no docker file is provided to verify that the issue is "fixed".
- [X] Implementation is documented. Details:
  No need for documentation, only CI change.
- [X] Change history is clear.
- [X] Commit messages are clear.

ivanperez-keera, 2022-05-07:
Change Manager: Implementation ready to be merged.

commit 8f54f7f6 (2022-05-07): Merge branch 'develop-ghc9'. Close #294.

**Description**

A new version of GHC has been available for some time and Copilot is
compatible with it. We can add it to our CI setup so that we officially
support it.

**Type**

- Management: conformance with new GHC version.

**Additional context**

None.

**Requester**

- Frank Dedden.

**Method to check presence of bug**

Not applicable (not a bug).

**Expected result**

- CI script successfully compiles copilot with GHC 9.0.

**Solution implemented**

- Modify CI script to include GHC 9.0 in the compilation matrix, picking
  a suitable version o [... cut]
closed 2022-05-07 by ivanperez-keera