Copilot-Language/copilot #305: Pass structs by reference, not by value, in trigger functions

RyanGlScott, 2022-04-05:
Currently, `copilot-c99`–generated trigger functions with struct arguments look approximately like this:

```c
void trigger(struct s s1, struct s s2, ...) { ... }
```

That is, each struct argument is passed by value. While this works, it has some drawbacks:

* This is somewhat inconsistent with `copilot-c99`'s treatment of array arguments, which is the other form of aggregate data structure that `copilot-c99` supports. In C, array arguments are passed by reference. As a result, passing arrays to trigger functions does not require copying the entirety of the array. This is not the case for structs, however, which require copying all fields to the stack.
* This presents a challenge for verification tools which analyze the `copilot-c99`–generated code, as they must deal with the subtleties of System V's conventions for passing structs. Section 3.2.3 of the [System V ABI](https://raw.githubusercontent.com/wiki/hjl-tools/x86-psABI/x86-64-psABI-1.0.pdf) has the full story, but the short version is that the ABI requirements for passing struct arguments can often look quite different from how the structs are defined in C. An example to demonstrate this is:

  ```c
  struct s1 {
    int x1;
    double y1;
  };

  struct s2 {
    int x2;
    double y2;
    char z2;
  };

  struct s3 {
    int x3;
    int y3;
  };

  void f1(struct s1 ss) {}
  void f2(struct s2 ss) {}
  void f3(struct s3 ss) {}
  ```

  Here, `f1`, `f2`, and `f3` are three trigger functions with struct arguments of different types. Although the trigger functions look very similar in C, they are handled quite differently at the ABI level. A good way to demonstrate this is to compile this program to LLVM bitcode, which adheres to the System V ABI conventions:

  ```ll
  %struct.s2 = type { i32, double, i8 }

  ; Function Attrs: norecurse nounwind readnone uwtable
  define dso_local void @f1(i32 %ss.coerce0, double %ss.coerce1) local_unnamed_addr #0 {
  entry:
    ret void
  }

  ; Function Attrs: norecurse nounwind readnone uwtable
  define dso_local void @f2(%struct.s2* nocapture byval(%struct.s2) align 8 %ss) local_unnamed_addr #0 {
  entry:
    ret void
  }

  ; Function Attrs: norecurse nounwind readnone uwtable
  define dso_local void @f3(i64 %ss.coerce) local_unnamed_addr #0 {
  entry:
    ret void
  }
  ```

  The important parts of this bitcode are the arguments to each function:

  1. `f1` takes two arguments rather than a single struct argument.
  2. `f2` takes a struct pointer argument.
  3. `f3` takes a single 64-bit integer argument.

  This example demonstrates that faithfully adhering to the conventions in System V is tricky in general, as there are many different rules that must be considered.

To address these issues, we propose that `copilot-c99` instead generate code which passes structs by reference, not by value. That is, instead generate trigger functions that look like this:

```c
void trigger(const struct s *s1, const struct s *s2, ...) { ... }
```

Making this change in `copilot-c99` proves relatively straightforward. Currently, a trigger function is invoked roughly like so from within `copilot-c99`–generated `step` function:

```c
if (trig_guard()) {
  trigger(trig_arg1(), trig_arg2(), ...);
}
```

With the proposed pass-structs-by-reference scheme, this would instead look like:

```c
struct s1 trig_argtemp1;
struct s2 trig_argtemp2;
...

if (trig_guard()) {
  trig_argtemp1 = trig_arg1();
  trig_argtemp2 = trig_arg2();
  ...
  trigger(&trig_argtemp1, &trig_argtemp2, ...);
}
```

Here, we use C's `&` operator to obtain references to each of the struct argument values.

This would represent a breaking change to the generated code, but one that would be relatively straightforward to adapt to in downstream code. If there is any Copilot documentation that discusses how struct arguments are compiled, that will also need to be updated, although we have not found any documentation which does so (neither [this tutorial](https://copilot-language.github.io/downloads/copilot_tutorial.pdf) nor [this technical report](https://ntrs.nasa.gov/api/citations/20200003164/downloads/20200003164.pdf) discuss this aspect of compilation).

ivanperez-keera, 2022-04-05:
Thanks for documenting the issue.

@fdedden any thoughts?

Questions/comments:
- Can this present any challenge when it comes to responsibility of memory management and memory de-allocations?

- What happens when you pass an array of structs or nested structs? Does it require any special treatment, or is the top-level value the only one that is passed by reference?

- Some arguments passed to handlers will be internal variables, or even input variables. Should each of those have a separate copy for the purposes of passing them as arguments? If not, does this expose the internals of the monitoring module to calling functions in a way that they could modify the behavior of the monitors by altering memory needed in future samples?

RyanGlScott, 2022-04-05:
> * Can this present any challenge when it comes to responsibility of memory management and memory de-allocations?

I wouldn't anticipate any challenges. In the proposal above, we have:

```c
struct s1 trig_argtemp1;
struct s2 trig_argtemp2;
...

if (trig_guard()) {
  trig_argtemp1 = trig_arg1();
  trig_argtemp2 = trig_arg2();
  ...
  trigger(&trig_argtemp1, &trig_argtemp2, ...);
}
```

This simply allocates each argument to the stack before passing it by reference, so there's no need for any manual memory management on the Copilot side. If any downstream users wish to invoke the trigger functions in their own code, they would need to choose how they allocate the structs that they pass. That being said, if anyone is currently using code that manually invokes a trigger function with a struct argument, they are almost assuredly stack-allocating it. In that case, migrating that code is as simple as adding a `&` character to the right places.

> * What happens when you pass an array of structs or nested structs? Does it require any special treatment, or is the top-level value the only one that is passed by reference?

No special treatment is required. Really, "passing by reference" in C just means "pass as a pointer", and the only thing that needs a pointer is the top-level value being referenced.

> * Some arguments passed to handlers will be internal variables, or even input variables. Should each of those have a separate copy for the purposes of passing them as arguments? If not, does this [...]

If I understand how `copilot-c99` works correctly, it is already passing a separate copy of each struct argument to trigger functions. Here is my understanding of how things work today:

```c
static struct s1 arg1_orig;
static struct s2 arg2_orig;
...

struct battery trig_arg1() {
  return arg1_orig;
}
struct battery trig_arg2() {
  return arg2_orig;
}
...

void step() {
  if (trig_guard()) {
    trigger(trig_arg1(), trig_arg2(), ...);
  }
}
```

Because the `trig_arg` functions all return struct values directly instead of returning pointers, the values of `arg1_orig`, `arg2_orig`, etc. will be copied at the call sites of the functions. That is to say, modifying the values that the struct arguments point to in the `trigger` function wouldn't change the underlying values of `arg1_orig`, `arg2_orig`, etc. As a result, altering the memory wouldn't really accomplish anything.

The same holds true in the proposed new design:

```c
void step() {
  if (trig_guard()) {
    trig_argtemp1 = trig_arg1();
    trig_argtemp2 = trig_arg2();
    ...
    trigger(&trig_argtemp1, &trig_argtemp2, ...);
  }
}
```

The only difference is that if you modified the values that the struct arguments point to in this version of the `trigger` function, you would be changing the values of `trig_argtemp1`, `trig_argtemp2`, etc. Again, this wouldn't really accomplish anything, since these values are distinct from the values in `arg1_orig`, `arg2_orig`, etc., and the `trig_argtemp` values aren't used anywhere else except as arguments to `trigger`. Moreover, the `trig_argtemp` values are local to the `step` function, so users wouldn't be able to access them outside of the `trigger` function.

If you wanted, you could be defensive and mark the struct arguments as `const` to prevent users from modifying the values that they point to:

```c
void trigger(const struct s *s1, const struct s *s2, ...) { ... }
```

This is not strictly required, however. If consistency with arrays is important, note that Copilot does not mark its array arguments as `const`.

RyanGlScott, 2022-04-06:
On the topic of consistency with arrays, I just remembered the Copilot's `arg()` functions handle arrays slightly differently. That is, `arg()` functions return _references_ to arrays, e.g.,

```
static int16_t s[3][2] = ...;
static size_t s_idx = 0;

int16_6 * s_get(size_t x) {
  return s[(s_idx + x) % 3];
}

int16_t * trig_arg() {
  return s_get(0);
}
```

So a downstream user could, in theory, modify the memory that `trig_arg()` points to in a trigger function, and those changes would be reflected in the static `s` array. (I'm not sure if this is ever desirable to allow, but if not, array arguments could be marked as `const`.) If we wanted to keep the handling of structs consistent with arrays, we could simply change the `arg()` functions to return references to static structs.

RyanGlScott, 2022-04-06:
> Just a note though: that would be a separate issue and commit (no need to open it yet; let's do one at a time).

Certainly.

Do you have an opinion on whether arguments that are passed by reference should be marked as `const` or not? If you don't wish to do this, then we wouldn't need to change anything about the way arrays are handled, just structs.

ivanperez-keera, 2022-04-06:
> Do you have an opinion on whether arguments that are passed by reference should be marked as const or not? If you don't wish to do this, then we wouldn't need to change anything about the way arrays are handled, just structs.

If that doesn't affect this issue, then I'd really prefer to discuss it separately at a later point.

RyanGlScott, 2022-04-06:
> If that doesn't affect this issue, then I'd really prefer to discuss it separately at a later point.

OK. In that case, I've edited the original proposal to remove mentions of `const`.

ivanperez-keera, 2022-04-06:
I thought you meant about arrays.

Please don't change anything from as it is right now (if you've made that change, then keep it as is). Re-evaluating issues/requests and PRs is generally harder if they keep changing.

If I need to ask for changes, I will do so.

robdockins, 2022-04-06:
Before we submit a PR for this, I think it is worth answering the question "should trigger functions use defensive copies for their aggregate arguments"? That will inform what the patch regarding structs should look like, i.e., if we copy the struct value to the stack before taking its address, or if we just pass the address of the struct directly.  Separately, if needed, we can make changes to the code that handles array arguments.

I am of the opinion that defensive copies are probably not needed, but I can see the argument the other way.

ivanperez-keera, 2022-04-06:
@robdockins That's fine; yes. I think that should be discussed.

> defensive copies are probably not needed

What's the reason behind that? (e.g., does it not offer any protection, does it make your side of the work harder, does it require big changes).

My thinking was that, as a tool that helps build correct monitors, Copilot should help people do the right thing, by design. If something is not meant to be changed, then it's good to make the compiler check for us, and to also make it such that users can't miss it by mistake or because they didn't read the documentation.

That being said, if I understand correctly, you would always declare a separate variable struct to hold the argument being passed to the handler. The result of getting or determining the value to pass as argument would be copied to that variable, and that variable would never be used in the calculation of the status of the monitors (not even if we have delays). It would simply be a different memory location. If that is correct, then I think you are right: it offers no protection because the value is always re-written before calling the handler, and it is never read in the generated code.

So, that last paragraph is correct, it would be more of a message to the user than offer any effectual protection.

robdockins, 2022-04-06:
What you are describing in your last paragraph is what I mean by "defensive copy." In that scenario you copy the values of the struct or array into a temporary location (usually on the stack) and then pass the address of that temporary location to the trigger function. The advantage of this is that the called function cannot (by accident or malice) alter the values you care about. The disadvantage is largely the overhead (time, but mostly memory) for making the copy.  This might be an issue for constrained devices with small stack space, if the arrays or structs involved are large.

The alternative approach is simply to pass the trigger function a direct pointer to the place where the struct or array lives (e.g., inside the ring buffer). This avoids the copy, but now your caller might clobber your values.  Let's call this the "direct" scenario.

Somewhat orthogonal to this is the issue of adding a `const` qualifier to the type signature of the trigger functions. Such a qualifier will tell implementers that they _should_ not modify through a pointer they are given.  Doing this makes sense in either the "direct" or "defensive copy" situations.  However, the protection provided by a `const` qualifier is  only advisory; users can cast away a `const` qualifier and write through the pointer anyway and the compiler will not stop them.  It does make it much harder to do by accident, however.

In my mind, defensive copies are warranted when the code being called is untrusted, either in an adversarial sense, or from a defense-in-depth damage containment sense.  I don't think an adversarial assumption applies in this instance, but one can certainly argue for defense-in-depth.

My initial stance is that the "direct" scenario (together with the `const` qualifier and documentation indicating that users should not modify the passed values) is probably sufficient in this instance; but I could certainly be convinced by an appeal via defense-in-depth that defensive copies are worthwhile.

ivanperez-keera, 2022-06-02:
Change Manager: Confirmed via discussions with parties affected that the issue exists and the new behavior is needed.

ivanperez-keera, 2022-06-02:
Technical Lead: Confirmed after discussion with project lead and parties interested (Galois) that the issue must be addressed.

ivanperez-keera, 2022-06-02:
Technical Lead: Issue scheduled for fixing in Copilot 3.10.

Fix assigned to: @RyanGlScott.

RyanGlScott, 2022-06-02:
I've submitted a patch in #341.

RyanGlScott, 2022-06-02:
Implementor: solution implemented, review requested.

ivanperez-keera, 2022-06-15:
Change Manager: Verified that:
- Solution is implemented:
  - [X] The code proposed compiles and passes all tests. Details:
    Build log: https://app.travis-ci.com/github/Copilot-Language/copilot/builds/252004980

  - [X] The solution proposed produces the expected result. Details:
    The following dockerfile, spec and C file attached, check that the new work generates code that compiles, passes structs as a pointer, and handles delays in streams of structs correctly. When executed, it produces some debugging messages and finishes with the message `Success` if execution worked as expected.
      ```Dockerfile
    --- Dockerfile-verify-305
    FROM ubuntu:trusty
    
    RUN apt-get update
    
    RUN apt-get install --yes software-properties-common
    RUN add-apt-repository ppa:hvr/ghc
    RUN apt-get update
    
    RUN apt-get install --yes ghc-8.6.5 cabal-install-2.4
    RUN apt-get install --yes libz-dev
    
    ENV PATH=/opt/ghc/8.6.5/bin:/opt/cabal/2.4/bin:$PWD/.cabal-sandbox/bin:$PATH
    
    RUN cabal update
    RUN cabal v1-sandbox init
    RUN cabal v1-install alex happy
    RUN apt-get install --yes git
    
    ADD Spec.hs /Spec.hs
    ADD main.c /main.c
    
    SHELL ["/bin/bash", "-c"]
    CMD git clone $REPO && cd $NAME && git checkout $COMMIT && cd .. \
        && cabal v1-install $NAME/copilot**/ \
        && cabal v1-exec -- runhaskell Spec.hs \
        && gcc main.c structs.c -o main \
        && ./main \
        && echo "Success"    
    
    --- Spec.hs
    -- | An example showing how specifications involving structs (in particular,
    -- nested structs) are compiled to C using copilot-c99.
    
    {-# LANGUAGE DataKinds #-}
    
    module Main where
    
    import qualified Prelude as P
    import Control.Monad (void, forM_)
    
    import Language.Copilot
    import Copilot.Compile.C99
    
    -- | Definition for `Volts`.
    data Volts = Volts
      { numVolts :: Field "numVolts" Word16
      , flag     :: Field "flag"     Bool
      }
    
    -- | `Struct` instance for `Volts`.
    instance Struct Volts where
      typename _ = "volts"
      toValues volts = [ Value Word16 (numVolts volts)
                       , Value Bool   (flag volts)
                       ]
    
    -- | `Volts` instance for `Typed`.
    instance Typed Volts where
      typeOf = Struct (Volts (Field 0) (Field False))
    
    -- | Input voltage information
[... 100 lines cut ...]
    
      // Check that sumLast has added all past values. In this case,
      // they will be 21 + 28 + 36 + 45 = 130
      if (sumLast != 130)
      {
        printf("Failure: the last value sum is incorrect: %d\n", sumLast);
        result = 1;
      }
    
      return result;
    }
    ```
    Command (substitute variables based on new path after merge):
    ```
    $ docker run -e "REPO=https://github.com/GaloisInc/copilot-1" -e "NAME=copilot-1" -e "COMMIT=4c29ea5915a7d91f6c103e891e3c99e73cf15e50" -it copilot-verify-305
    ```
- [X] Implementation is documented. Details:
  No changes to the documentation needed.
- [X] Change history is clear.
- [X] Commit messages are clear.
- [X] Changelogs are updated.
- [X] Examples are updated. Details:
  No changes needed: The example that uses structs does not provide a C main that would be affected by this change.
- [X] Required version bumps are evaluated. Details:
  Bump needed: Change to the interface of the code produced.

ivanperez-keera, 2022-06-15:
Change Manager: Implementation ready to be merged.

commit aec23dcf (2022-06-15): Merge branch 'develop-pass-structs-by-reference'. Close #305.

**Description**

Currently, calls to `copilot-c99`–generated trigger functions with
struct arguments pass the structs by value:

```c
void trigger(struct s s1, struct s s2, ...) { ... }
```

Verification tools which analyze the `copilot-c99`–generated code must
deal with the subtleties of System V's conventions for passing structs,
as there are many different rules that must be considered. In the final
LLVM form, struct arguments passed to handlers may have different types,
even if their C type (struct) definitions are relatively s [... cut]
closed 2022-06-15 by ivanperez-keera