Copilot-Language/copilot #331: `copilot-language`: typo in comment

ivanperez-keera, 2022-05-12:
**Description**

The module `Copilot.Language.Interpret` contains a typo in a comment. In the module declaration, it says: `which may be used to simulated or executed Copilot specifications`, when it should say `which may be used to simulate or execute Copilot specifications`.

**Type**

- Bug: typo in comment.

**Additional context**

None.

**Requester**

- Ivan Perez

**Method to check presence of bug**

Not applicable (grammatical error in comment, not possible to check automatically with existing tools). 

**Expected result**

The comment should read: `which may be used to simulate or execute Copilot specifications`.

**Desired result**

The comment should read: `which may be used to simulate or execute Copilot specifications`.

**Proposed solution**

Fix typo in comment.

**Further notes**

None.

ivanperez-keera, 2022-05-31:
Change Manager: Confirmed that the typo exists.

ivanperez-keera, 2022-06-02:
Technical Lead: Issue scheduled for fixing in Copilot 3.10.

willpog, 2022-06-13:
Implementor: Solution implemented, review requested.

ivanperez-keera, 2022-06-13:
Change Manager: Verified that:
- Solution is implemented:
  - [X] The code proposed compiles and passes the unit tests. Details:
    Build log: https://app.travis-ci.com/github/copilot-language/copilot/builds/251881131
  - [X] The solution proposed produces the expected result. Details:
    Evaluated via manual inspection. Justification: not (easily) scriptable, as it is a typo correction in a comment and we have no good mechanism to extract that portion of the text and spell check it without producing numerous false positives.
- [X] Implementation is documented. Details:
  No need for additional documentation (typo fix in comment).
- [X] Change history is clear.
- [X] Commit messages are clear.
- [X] Changelogs are updated.
- [X] Required version bumps are evaluated. Details:
  Bump not required (minor change to comment, not API-breaking).

ivanperez-keera, 2022-06-13:
Change Manager: Implementation ready to be merged.

commit a6a21830 (2022-07-04): Merge branch 'develop-typo-comment'. Close #331.

**Description**

The module `Copilot.Language.Interpret` contains a typo in a comment. In
the module declaration, it says: `which may be used to simulated or
executed Copilot specifications`, when it should say `which may be used
to simulate or execute Copilot specifications`.

**Type**

- Bug: typo in comment.

**Additional context**

None.

**Requester**

- Ivan Perez

**Method to check presence of bug**

Not applicable (grammatical error in comment, not possible to check
automatically with existing tools).

**Expected result**

The comment s [... cut]