Copilot-Language/copilot #335: Extend range of versions of dependencies

ivanperez-keera, 2022-05-16:
**Description**

In order to keep Copilot effectively working in the current Haskell ecosystem, as well as new versions of GHC, we need to extend the versions of dependencies that Copilot can be installed with. For example, `bimap` has just seen a new release 0.5.0, but `copilot-theorem` needs versions strictly lower than 0.4. The same is true for `optparse-applicative`, of which we depend on versions prior to 0.16 but versions 0.16, 0.16.1 and 0.17 have been released.

For those libraries that have introduced no breaking changes affecting Copilot, we should extend the version bounds.

**Type**

- Management: update versions of dependencies.

**Additional context**

None.

**Requester**

- Ivan Perez

**Method to check presence of bug**

Not applicable (not a bug).

**Expected result**

Copilot can be installed with the most recent version of all dependencies published on hackage that do not require a change to the Haskell code.
 
**Desired result**

Copilot can be installed with the most recent version of all dependencies published on hackage that do not require a change to the Haskell code.

**Proposed solution**

Copilot can be installed with the most recent version of all dependencies published on hackage that do not require a change to the Haskell code.

**Further notes**

The library `bimap` was not deduced by GHC as Safe in version 0.4 due to a new unsafe import, but that has now been resolved. For that dependency, version 0.4 should be avoided, versions strictly lower than 0.4 and higher than 0.5 are safe.

See: https://github.com/joelwilliamson/bimap/issues/14

ivanperez-keera, 2022-06-02:
Change Manager: Confirmed that versions are overly constrained and upper bounds should be relaxed. Final determination of upper bounds to be made during implementation.

ivanperez-keera, 2022-06-02:
Technical Lead: Issue scheduled for fixing in Copilot 3.10.

ivanperez-keera, 2022-07-05:
Technical Lead: Fix assigned to @ivanperez-keera .

ivanperez-keera, 2022-07-05:
I'm about to push an update for this (this will be included in Copilot 3.10, to be released in two days).

A temporary WIP is here: https://github.com/Copilot-Language/copilot/compare/master...develop-test-ghc9.2

@RyanGlScott @robdockins The fix for this issue supports `what4 < 1.4`. Is there a version of what4 that is expected to be released soon? If so, are there any breaking changes?

RyanGlScott, 2022-07-05:
I can reproduce this bug independently of Copilot, so I've opened fdedden/language-c99-simple#8 to track this on the `language-c99-simple` side.

ivanperez-keera, 2022-07-07:
Thanks, Ryan.

If the release of that fix for `language-c99-simple` happened soon, there exist mechanisms on that package's side to prevent users from installing the broken version. However, seeing that `language-c99-simple` has not accepted (or responded to) that yet, it is unclear when that fix will be released (unlikely to be today).

Let's prevent users from using `language-c99*` 0.2 for now and widen the versions of all other packages.

ivanperez-keera, 2022-07-07:
The change is ready now here: https://github.com/ivanperez-keera/copilot.

ivanperez-keera, 2022-07-07:
Change Manager: Verified that:
- Solution is implemented:
  - [X] The code proposed compiles and passes all tests. Details:
    Build log: https://app.travis-ci.com/github/Copilot-Language/copilot/builds/252925719

  - [X] The solution proposed produces the expected result. Details:
    The following Dockerfile checks that copilot can be installed with the latest versions of all dependencies modified by the commits, except for mtl and transformers (details explained below). When executed, it prints the message `Success` if installation completes successfully, and an error message if not.
      ```Dockerfile
    FROM ubuntu:focal
    
    RUN apt-get update
    
    RUN apt-get install --yes libz-dev
    RUN apt-get install --yes git
    
    # RUN curl --proto '=https' --tlsv1.2 -sSf https://get-ghcup.haskell.org | sh
    RUN apt-get install --yes wget
    RUN mkdir -p $HOME/.ghcup/bin
    RUN wget https://downloads.haskell.org/~ghcup/0.1.17.7/x86_64-linux-ghcup-0.1.17.7 -O $HOME/.ghcup/bin/ghcup
    
    RUN chmod a+x $HOME/.ghcup/bin/ghcup
    ENV PATH=$PATH:/root/.ghcup/bin/
    ENV PATH=$PATH:/root/.cabal/bin/
    RUN apt-get install --yes curl
    RUN apt-get install --yes gcc make libgmp3-dev
    RUN ghcup install ghc 9.0
    RUN ghcup install cabal 3.4
    RUN ghcup set ghc 9.0.2
    RUN cabal update
    
    SHELL ["/bin/bash", "-c"]
    CMD git clone $REPO \
        && cd $NAME \
        && git checkout $COMMIT \
        && cd .. \
        && cabal install alex happy \
        && cd copilot \
        && cabal install --lib copilot**/ \
             --constraint='bimap==0.5.0' \
             --constraint='random==1.2.1.1' \
             --constraint='what4==1.3' \
             --constraint='optparse-applicative==0.17.0.0' \
        && echo "Success"
    ```
    Command (substitute variables based on new path after merge):
    ```
    $ docker run -e "REPO=https://github.com/ivanperez-keera/copilot" -e "NAME=copilot" -e "COMMIT=e0a2de77e0717f815245ef69fc5c606043a35559" -it copilot-verify-335
    ```
    Copilot is technically compatible with more modern versions of mtl and transformers than those installed with GHC 9.0, but we cannot install that until we support language-c99 >= 0.2 and therefore GHC 9.0. This dockerfile therefore does not enforce using the latest versions of mtl and transformers.
- [X] Implementation is documented. Details:
  No changes to the documentation needed (change to dependencies).
- [X] Change history is clear.
- [X] Commit messages are clear.
- [X] Changelogs are updated.
- [X] Examples are updated. Details:
  No examples affected.
- [X] Required version bumps are evaluated. Details:
  No bump needed.

ivanperez-keera, 2022-07-07:
Change Manager: Implementation ready to be merged.

commit af4c457e (2022-07-07): Merge branch 'develop-version-bounds'. Close #335.

**Description**

In order to keep Copilot effectively working in the current Haskell
ecosystem, as well as new versions of GHC, we need to extend the
versions of dependencies that Copilot can be installed with. For
example, `bimap` has just seen a new release 0.5.0, but
`copilot-theorem` needs versions strictly lower than 0.4. The same is
true for `optparse-applicative`, of which we depend on versions prior to
0.16 but versions 0.16, 0.16.1 and 0.17 have been released.

For those libraries that have introduced no breaking changes affecting
Co [... cut]