Copilot-Language/copilot #336: Why is there no `Typed` instance for Char or String?

IQubic, 2022-05-26:
I'm looking through the docs for Copilot and noticing that there's no instance for `Typed Char` or for `Typed String`. From my understanding, this means it's impossible to create a Trigger that is invoked with a Char or with a String as one of its arguments. This seems like a major shortcoming for this library, seeing as these are two very common types in C. Is there some reason for why these two instances don't exist?

ivanperez-keera, 2022-05-26:
Wrt. `Char`, we have Word8. Does that cover your needs?

Wrt. `String`, we require all arrays to be of fixed length so we can generate hard realtime code with no loops or recursion. You could have a string of up-to X characters. Support for modifying and creating new arrays has been discussed and may be added (TBD).

IQubic, 2022-05-26:
Actually, that works just fine for me. However, I think I might be using an outdated version of Copilot, because `Word8` doesn't seem to have a `Typeable` instance for me. `Int8` has one. Am I using an outdated version of the library?

closed 2022-05-27 by ivanperez-keera