Copilot-Language/copilot #352: `copilot`: typo in comments (Celcius)

ivanperez-keera, 2022-06-13:
**Description**

The README includes an example that uses the word _Celcius_ multiple times, as opposed to _Celsius_ (spelled with an 's'). The former spelling is a typo.

The same error appears in the heater example under `copilot/examples/Heater.hs`.

Both files should be corrected.

**Type**

- Bug: typo in comment.

**Additional context**

None.

**Requester**

- Ivan Perez

**Method to check presence of bug**

```sh
$ grep -nIHre 'Celcius' .
./copilot/README.md:86:-- this to Celcius.
./copilot/README.md:99:-- Calculate temperature in Celcius.
./copilot/examples/Heater.hs:6:-- this to Celcius.
./copilot/examples/Heater.hs:19:-- Calculate temperature in Celcius.
```

**Expected result**

The grep command executed above should return no results.

**Desired result**

The grep command executed above should return no results.

**Proposed solution**

Fix typos in the comments in both files.

**Further notes**

None.

ivanperez-keera, 2022-06-13:
Change Manager: Confirmed that the typo exists.

ivanperez-keera, 2022-06-13:
Technical Lead: Issue scheduled for fixing in Copilot 3.10.

Fix assigned to: @willpog

willpog, 2022-06-17:
Implementor: Solution implemented, review requested.

ivanperez-keera, 2022-06-17:
Change Manager: Verified that:
- Solution is implemented:
  - [X] The code proposed compiles and passes all tests. Details:
    Build log: https://app.travis-ci.com/github/Copilot-Language/copilot/builds/252102974

  - [X] The solution proposed produces the expected result. Details:
    The following Dockerfile checks that the misspelling _Celcius_ has been completely removed from the repo. When executed, it prints the message `Success` if the misspelled word cannot be found, and the word `Failure` otherwise.
      ```Dockerfile
    FROM ubuntu:focal
    
    RUN apt-get update
    RUN apt-get install --yes git moreutils
    RUN git config --global advice.detachedHead false
    
    SHELL ["/bin/bash", "-c"]
    CMD git clone --quiet $REPO \
        && cd $NAME \
        && git checkout --quiet $COMMIT \
        && (grep -rie 'celcius' copilot** | ifne false) \
        && echo "Success" || echo "Failure"
    ```
    Command (substitute variables based on new path after merge):
    ```
    $ docker run -e "REPO=https://github.com/willpog/copilot" -e "NAME=copilot" -e "COMMIT=db8e77caf8a0cf96a7e59f4720dad1bbef3c0e89" -it copilot-verify-352
    ```
- [X] Implementation is documented. Details:
  No changes to the documentation needed due to this change (but some changes affect the documentation).
- [X] Change history is clear.
- [X] Commit messages are clear.
- [X] Changelogs are updated.
- [X] Examples are updated. Details:
  Typos in examples also fixed.
- [X] Required version bumps are evaluated. Details:
  No bump needed: Change to spelling in comments, documentation.

ivanperez-keera, 2022-06-17:
Change Manager: Implementation ready to be merged.

commit 21d94b9d (2022-06-17): Merge branch 'develop-typo-celsius'. Close #352.

**Description**

The README includes an example that uses the word _Celcius_ multiple
times, as opposed to _Celsius_ (spelled with an 's'). The former
spelling is a typo.

The same error appears in the heater example under
`copilot/examples/Heater.hs`.

Both files should be corrected.

**Type**

- Bug: typo in comment.

**Additional context**

None.

**Requester**

- Ivan Perez

**Method to check presence of bug**

```sh
$ grep -nIHre 'Celcius' .
./copilot/README.md:86:-- this to Celcius.
./copilot/README.md:99:-- Calculate temperature in Celci [... cut]
closed 2022-06-17 by ivanperez-keera