Copilot-Language/copilot #380: `copilot-c99`: Remove unnecessary compilation flag

ivanperez-keera, 2022-10-13:
**Description**

The cabal file of `copilot-c99` specifies a compiler flag `--no-deprecations` that is no longer necessary. It should have been removed when modules were hidden, but the flag was left behind. 

**Type**

- Bug: unnecessary code included in compilation options.

**Additional context**

None.

**Requester**

- Ivan Perez

**Method to check presence of bug**

The following line prints `found` if the unnecessary flag is present in the cabal file, and `nothing` otherwise.
```sh
$ if [[ $(grep -re 'no-deprecations' copilot-c99/copilot-c99.cabal | grep -ve '--') ]]; then echo "found"; else echo "nothing"; fi
```

**Expected result**

The above line prints `nothing`.
 
**Desired result**

The above line prints `nothing`.

**Proposed solution**

Remove flag and preceding comment from cabal file.

**Further notes**

See issues:
- #289 
- #272

ivanperez-keera, 2022-10-18:
Change Manager: Confirmed that the flag is no longer needed as per the description provided in the comment.

ivanperez-keera, 2022-10-18:
Technical Lead: Issue scheduled for fixing in Copilot 3.12.

Fix assigned to @willpog .

willpog, 2022-10-18:
Implementor: Solution implemented, review requested.

ivanperez-keera, 2022-10-19:
Change Manager: Verified that:
- Solution is implemented:
  - [X] The code proposed compiles and passes all tests. Details:
    Build log: https://app.travis-ci.com/github/Copilot-Language/copilot/builds/256824776

  - [X] The solution proposed produces the expected result. Details:
    The following dockerfile checks that the flag is no longer present:
    ```Dockerfile
    FROM ubuntu:focal
    
    RUN apt-get update
    
    RUN apt-get install --yes git
    
    SHELL ["/bin/bash", "-c"]
    CMD git clone $REPO \
        && cd $NAME \
        && git checkout $COMMIT \
        && if [[ $(grep -re 'no-deprecations' copilot-c99/copilot-c99.cabal | grep -ve '--') ]]; then echo "Found"; else echo "Nothing"; fi
    ```
    When executed over the new commit, it prints the text "Nothing", indicating that the flag is absent.
- [X] Implementation is documented. Details:
  - No changes needed.
- [X] Change history is clear.
- [X] Commit messages are clear.
- [X] Changelogs are updated.
- [X] Examples are updated. Details:
  No updates needed.
- [X] Required version bumps are evaluated. Details:
  Bump not needed. API is extended by existing definitions do not change.

ivanperez-keera, 2022-10-19:
Change Manager: Implementation ready to be merged.

commit 1456f12c (2022-10-19): Merge branch 'develop-c99-remove-flag'. Close #380.

**Description**

The cabal file of `copilot-c99` specifies a compiler flag
`--no-deprecations` that is no longer necessary. It should have been
removed when modules were hidden, but the flag was left behind.

**Type**

- Bug: unnecessary code included in compilation options.

**Additional context**

None.

**Requester**

- Ivan Perez

**Method to check presence of bug**

The following line prints `found` if the unnecessary flag is present in
the cabal file, and `nothing` otherwise.
```sh
$ if [[ $(grep -re 'no-deprecations' copilot-c99/copil [... cut]
closed 2022-10-19 by ivanperez-keera