Copilot-Language/copilot #386: `copilot-c99`: Streams of arrays result in C code with undefined behavior

RyanGlScott, 2022-11-01:
_(A continuation of a discussion started in https://github.com/Copilot-Language/copilot/issues/314#issuecomment-1225858862 and https://github.com/Copilot-Language/copilot/issues/373#issuecomment-1265255432.)_

Given this Copilot spec:

```hs
-- Delay.hs
{-# LANGUAGE DataKinds #-}
module Main where

import Copilot.Compile.C99
import Language.Copilot
import Prelude             hiding ((++))

spec :: Spec
spec = trigger "f" true [ arg $ stream]
  where
    stream :: Stream (Array 2 Int16)
    stream = [array [3,4]] ++ rest

    rest :: Stream (Array 2 Int16)
    rest = constant $ array [5,6]

main :: IO ()
main = reify spec >>= compile "delay"
```

If you run this, it will generate the following C monitor files:

```c
// delay_types.h
```
```c
// delay.h
void f(int16_t f_arg0[(2)]);
void step(void);
```
```c
// delay.c
#include <stdint.h>
#include <stdbool.h>
#include <string.h>
#include <stdlib.h>
#include <math.h>

#include "delay_types.h"
#include "delay.h"

static int16_t s0[(1)][(2)] = {{((int16_t)(3)), ((int16_t)(4))}};
static size_t s0_idx = (0);

int16_t * s0_get(size_t x) {
  return (s0)[((s0_idx) + (x)) % (1)];
}

int16_t * s0_gen(void) {
  return (int16_t [(2)]){((int16_t)(5)), ((int16_t)(6))};
}

bool f_guard(void) {
  return true;
}

int16_t * f_arg0(void) {
  return (s0_get)((0));
}

void step(void) {
  int16_t s0_tmp[(2)];
  int16_t * f_arg_temp0;
  if ((f_guard)()) {
    {(f_arg_temp0) = ((f_arg0)());
     (f)((f_arg_temp0));}
  };
  (memcpy)((s0_tmp), ((s0_gen)()), ((2) * (sizeof(int16_t))));
  (memcpy)(((s0)[s0_idx]), (s0_tmp), ((2) * (sizeof(int16_t))));
  (s0_idx) = (((s0_idx) + (1)) % (1));
}
```

If you link this code against the following program:

```c
// delay_driver.c
#include <stdint.h>
#include <stdio.h>

#include "delay_types.h"
#include "delay.h"

void f (int16_t f_arg0[(2)]) {
  printf("([%d,%d])\n", f_arg0[0], f_arg0[1]);
}

int main(void) {
  int i = 0;

  printf("f:\n");

  for (i=0; i<5; i++) {
    step();
  }
  return 0;
}
```

Then it will produce different results depending on what optimization level you compile with! On my machine:

```
$ gcc -O0 delay_driver.c delay.c -o main && ./main # Without optimizations
f:
([3,4])
([5,6])
([5,6])
([5,6])
([5,6])

$ gcc -O delay_driver.c delay.c -o main && ./main # Without optimizations
f:
([3,4])
([0,0])
([0,0])
([0,0])
([0,0])
```

This is because the generated code in `delay.c` exhibits undefined behavior. In particular, the `s0_gen` function:

```c
int16_t * s0_gen(void) {
  return (int16_t [(2)]){((int16_t)(5)), ((int16_t)(6))};
}
```

This will stack-allocate an array and return a reference to the array. Because it is stack allocated, however, the lifetime of the array is limited to the scope of the `s0_gen()` function, so the pointer that `s0_gen()` returns will be dangling.

In https://github.com/Copilot-Language/copilot/issues/314#issuecomment-1226059544, @ivanperez-keera suggests addressing the issue by changing this line:

```c
  (memcpy)((s0_tmp), ((s0_gen)()), ((2) * (sizeof(int16_t))));
```

To something like this:

```c
  s0_gen(s0_tmp);
```

Where `s0_gen` has been modified to take an array as an output parameter. Here is one possible way to accomplish this:

```c
void s0_gen(int16_t s0_arg[2]) {
  int16_t s0_tmp[2] = {((int16_t)(5)), ((int16_t)(6))};
  memcpy(s0_arg, s0_tmp, 2 * sizeof(int16_t));
}
```

This version of the program eliminates the source of undefined behavior, producing the same results regardless of optimization level:

```
$ gcc -O0 delay_driver.c delay.c -o main && ./main # Without optimizations
f:
([3,4])
([5,6])
([5,6])
([5,6])
([5,6])

$ gcc -O delay_driver.c delay.c -o main && ./main # Without optimizations
f:
([3,4])
([5,6])
([5,6])
([5,6])
([5,6])
```

ivanperez-keera, 2022-12-10:
**Description**

C99 backend generates incorrect code. In the generated code, the functions that generate values passed as arguments to trigger functions may return pointers to stack-allocated arrays, so the memory may be used for something else (or freed).

Instead, the structure of the implementation should guarantee that the arrays are always allocated.

**Type**

- Bug: generated C code may be incorrect.

**Additional context**

None.

**Requester**

- Ryan Scott (Galois)

**Method to check presence of bug**

Given this Copilot spec:

```hs
-- Delay.hs
{-# LANGUAGE DataKinds #-}
module Main where

import Copilot.Compile.C99
import Language.Copilot
import Prelude             hiding ((++))

spec :: Spec
spec = trigger "f" true [ arg $ stream]
  where
    stream :: Stream (Array 2 Int16)
    stream = [array [3,4]] ++ rest

    rest :: Stream (Array 2 Int16)
    rest = constant $ array [5,6]

main :: IO ()
main = reify spec >>= compile "delay"
```

If you link the generated code against the following program:

```c
// delay_driver.c
#include <stdint.h>
#include <stdio.h>

#include "delay_types.h"
#include "delay.h"

void f (int16_t f_arg0[(2)]) {
  printf("([%d,%d])\n", f_arg0[0], f_arg0[1]);
}

int main(void) {
  int i = 0;

  printf("f:\n");

  for (i=0; i<5; i++) {
    step();
  }
  return 0;
}
```

Then it will produce different results depending on what optimization level you compile with. The observed behavior of the program should be independent of the optimization level.

```
$ gcc -O0 delay_driver.c delay.c -o main && ./main # Without optimizations
f:
([3,4])
([5,6])
([5,6])
([5,6])
([5,6])

$ gcc -O delay_driver.c delay.c -o main && ./main # Without optimizations
f:
([3,4])
([0,0])
([0,0])
([0,0])
([0,0])
```

**Expected result**

The two executions above after compiling code with and without optimizations should return the same results.

**Desired result**

The two executions above after compiling code with and without optimizations should return the same results.

**Proposed solution**

Modify generated C99 backend so that the output array in a `*_gen` function takes the array as input. That way, all memory is preallocated and there's no need for mallocs either (which we want to avoid).

**Further notes**

None.

ivanperez-keera, 2023-01-05:
Technical Lead: Issue scheduled for fixing in Copilot 3.13.

Fix assigned to: @ivanperez-keera .

ivanperez-keera, 2023-01-07:
Implementor: Solution implemented, review requested.

ivanperez-keera, 2023-01-08:
Change Manager: Verified that:
- Solution is implemented:
  - [X] The code proposed compiles and passes all tests. Details:
    Build log: https://app.travis-ci.com/github/Copilot-Language/copilot/builds/259474304

  - [X] The solution proposed produces the expected result. Details:
    The following Dockerfile checks that passing a (stream of) arrays as an argument to a trigger, when compiled with and without optimizations, produces the same results, in which case it prints the message `Success`. The docker image produces a diff of the two outputs if not.
    ```Dockerfile
      --- Dockerfile-verify-386
      FROM ubuntu:trusty
      
      RUN apt-get update
      
      RUN apt-get install --yes software-properties-common
      RUN add-apt-repository ppa:hvr/ghc
      RUN apt-get update
      
      RUN apt-get install --yes ghc-8.6.5 cabal-install-2.4
      RUN apt-get install --yes libz-dev
      
      ENV PATH=/opt/ghc/8.6.5/bin:/opt/cabal/2.4/bin:$PWD/.cabal-sandbox/bin:$PATH
      
      RUN cabal update
      RUN cabal v1-sandbox init
      RUN cabal v1-install alex happy
      RUN apt-get install --yes git
      
      ADD Delay.hs /tmp/Delay.hs
      ADD delay_driver.c /tmp/delay_driver.c
      
      SHELL ["/bin/bash", "-c"]
      CMD git clone $REPO && cd $NAME && git checkout $COMMIT && cd .. \
        && cabal v1-install $NAME/copilot**/ \
        && cabal v1-exec -- runhaskell /tmp/Delay.hs \
        && gcc -O0 /tmp/delay_driver.c delay.c -I. -o main-no-op && ./main-no-op > no-optimizations \
        && gcc -O /tmp/delay_driver.c delay.c -I. -o main-op && ./main-op > with-optimizations \
        && diff no-optimizations with-optimizations \
        && echo "Success"
      
      --- Delay.hs
      {-# LANGUAGE DataKinds #-}
      module Main where
      
      import Copilot.Compile.C99
      import Language.Copilot
      import Prelude             hiding ((++))
      
      spec :: Spec
      spec = trigger "f" true [ arg $ stream]
        where
          stream :: Stream (Array 2 Int16)
          stream = [array [3,4]] ++ rest
      
          rest :: Stream (Array 2 Int16)
          rest = constant $ array [5,6]
      
      main :: IO ()
      main = reify spec >>= compile "delay"
      
      --- delay_driver.c
      #include <stdint.h>
      #include <stdio.h>
      
      #include "delay_types.h"
      #include "delay.h"
      
      void f (int16_t f_arg0[(2)]) {
        printf("([%d,%d])\n", f_arg0[0], f_arg0[1]);
      }
      
      int main(void) {
        int i = 0;
      
        printf("f:\n");
      
        for (i=0; i<5; i++) {
          step();
        }
        return 0;
      }
    ```
    Command (adjust repo as necessary):
    ```
    $ docker run -e "REPO=https://github.com/ivanperez-keera/copilot" -e "NAME=copilot" -e "COMMIT=37797aa15a3b8582564023a7106f8062fd4f6ffe" -it copilot-verify-386
    ```
- [X] Implementation is documented. Details:
  Two new internal functions are introduced; the functions have haddock documentation.
- [X] Change history is clear.
- [X] Commit messages are clear.
- [X] Changelogs are updated.
- [X] Examples are updated. Details:
  No updates needed.
- [X] Required version bumps are evaluated. Details:
  Bump not needed. This change is internal (bug fix in function in hidden module; new functions not exposed).

ivanperez-keera, 2023-01-08:
Change Manager: Implementation ready to be merged.

commit 9bc8fa16 (2023-01-08): Merge branch 'develop-stream-array'. Close #386.

**Description**

C99 backend generates incorrect code. In the generated code, the functions that
generate values passed as arguments to trigger functions may return pointers to
stack-allocated arrays, so the memory may be used for something else (or
freed).

Instead, the structure of the implementation should guarantee that the arrays
are always allocated.

**Type**

- Bug: generated C code may be incorrect.

**Additional context**

None.

**Requester**

- Ryan Scott (Galois)

**Method to check presence of bug**

The following Dockerfile checks [... cut]
closed 2023-01-08 by ivanperez-keera