Copilot-Language/copilot #400: Cannot declare `extern` stream of nested arrays

RyanGlScott, 2023-01-03:
This example:

```hs
{-# LANGUAGE DataKinds #-}
module Main (main) where

import qualified Prelude as P

import Language.Copilot
import Copilot.Compile.C99

e :: Stream (Array 2 (Array 3 Int16))
e = extern "e" Nothing

spec :: Spec
spec = trigger "t" ((e .!! 0) .!! 1 == 42) [arg e]

main :: IO ()
main = do
  spec' <- reify spec
  compile "externNestedArrays" spec'
```

Triggers a Copilot error:

```
$ runghc ExternNestedArrays.hs
ExternNestedArrays.hs: Copilot error: The external symbol 'e' has been declared to have two different types!
CallStack (from HasCallStack):
  error, called at src/Copilot/Language/Error.hs:24:16 in copilot-language-3.12-inplace:Copilot.Language.Error
```

I don't think this should happen, however, as `e` has only been declared once.

Curiously, this issue only arises when the type of the stream involves a nested array like `Array 2 (Array 3 Int16)`. If I change the type of `e` to a non-nested array such as `Array 2 Int16`:

```hs
{-# LANGUAGE DataKinds #-}
module Main (main) where

import qualified Prelude as P

import Language.Copilot
import Copilot.Compile.C99

e2 :: Stream (Array 2 Int16)
e2 = extern "e2" Nothing

spec :: Spec
spec = trigger "t" (e2 .!! 0 == 42) [arg e2]

main :: IO ()
main = do
  spec' <- reify spec
  compile "externArrays" spec'
```

Then the error does not occur.

RyanGlScott, 2023-01-03:
A similar issue occurs when the `extern` contains arrays of structs:

```hs
{-# LANGUAGE DataKinds #-}
module Main (main) where

import qualified Prelude as P

import Language.Copilot
import Copilot.Compile.C99

data S = S { field :: Field "field" Int16 }

instance Struct S where
  typename _ = "s"
  toValues s = [Value Int16 (field s)]

instance Typed S where
  typeOf = Struct (S (Field 0))

e :: Stream (Array 2 S)
e = extern "e" Nothing

spec :: Spec
spec = trigger "t" ((e .!! 0)#field == 42) [arg e]

main :: IO ()
main = do
  spec' <- reify spec
  compile "externArrayOfStructs" spec'
```
```
$ runghc ExternArrayOfStructs.hs
ExternArrayOfStructs.hs: Copilot error: The external symbol 'e' has been declared to have two different types!
CallStack (from HasCallStack):
  error, called at src/Copilot/Language/Error.hs:24:16 in copilot-language-3.12-inplace:Copilot.Language.Error
```

RyanGlScott, 2023-01-03:
I think I know why this happens. Looking at the `Eq` instance for `SimpleType`:

https://github.com/Copilot-Language/copilot/blob/e1355bbbd5c32a6932490c5b00edc38d1cd54329/copilot-core/src/Copilot/Core/Type.hs#L173-L188

We see that when checking for array equality, the types of the elements are compared using `testEquality`. Here is the `TestEquality` instance for `Type`:

https://github.com/Copilot-Language/copilot/blob/e1355bbbd5c32a6932490c5b00edc38d1cd54329/copilot-core/src/Copilot/Core/Type.hs#L138-L150

Note that there are no cases checking for `Array` or `Struct`, so calling `testEquality` on arrays or structs will always return `Nothing`. This also explains why `extern` can be used on a non-nested array, as there are cases for every other sort of `Type`.

One possible way to fix this is to add `Array` and `Struct` cases to this `TestEquality` instance:

```diff
diff --git a/copilot-core/src/Copilot/Core/Type.hs b/copilot-core/src/Copilot/Core/Type.hs
index 7e4305c1..cb330d90 100644
--- a/copilot-core/src/Copilot/Core/Type.hs
+++ b/copilot-core/src/Copilot/Core/Type.hs
@@ -6,6 +6,7 @@
 {-# LANGUAGE KindSignatures            #-}
 {-# LANGUAGE Safe                      #-}
 {-# LANGUAGE ScopedTypeVariables       #-}
+{-# LANGUAGE TypeOperators             #-}
 {-# LANGUAGE UndecidableInstances      #-}

 -- The following flag is disabled in this module so that the import of
@@ -46,9 +47,10 @@ import Data.Int           (Int16, Int32, Int64, Int8)
 import Data.List          (intercalate)
 import Data.Proxy         (Proxy (..))
 import Data.Type.Equality as DE
-import Data.Typeable      (Typeable, typeRep)
+import Data.Typeable      (Typeable, eqT, typeRep)
 import Data.Word          (Word16, Word32, Word64, Word8)
-import GHC.TypeLits       (KnownNat, KnownSymbol, Symbol, natVal, symbolVal)
+import GHC.TypeLits       (KnownNat, KnownSymbol, Symbol, natVal, sameNat,
+                           symbolVal)

 -- Internal imports
 import Copilot.Core.Type.Array    (Array)
@@ -147,6 +149,21 @@ instance TestEquality Type where
   testEquality Word64 Word64 = Just DE.Refl
   testEquality Float  Float  = Just DE.Refl
   testEquality Double Double = Just DE.Refl
+  testEquality (Array t1) (Array t2) =
+      testArrayEquality t1 t2
+    where
+      testArrayEquality :: forall n1 a1 n2 a2.
+                           (KnownNat n1, KnownNat n2)
+                        => Type a1
+                        -> Type a2
+                        -> Maybe (Array n1 a1 :~: Array n2 a2)
+      testArrayEquality ty1 ty2
+        | Just DE.Refl <- sameNat (Proxy :: Proxy n1) (Proxy :: Proxy n2)
+        , Just DE.Refl <- testEquality ty1 ty2
+        = Just DE.Refl
+        | otherwise
+        = Nothing
+  testEquality (Struct _) (Struct _) = eqT
   testEquality _ _ = Nothing

 -- | A simple, monomorphic representation of types that facilitates putting
```

This suffices to fix the two failing examples above. It might also be worth adding similar cases to the `EqualType Type` instance here:

https://github.com/Copilot-Language/copilot/blob/e1355bbbd5c32a6932490c5b00edc38d1cd54329/copilot-core/src/Copilot/Core/Type.hs#L124-L136

ivanperez-keera, 2023-01-05:
**Description**

The implementation of type equality in `copilot-core` does not have cases for arrays or structs, and thus assumes that nested arrays or arrays with struct types are different (even if they are the exact same).

As a consequence of this lack, the C99 backend fails to complete. Since it cannot tell that an extern with a nested array or array with structs is used twice, it "assumes" that the spec tries to declare the same variable multiple times different types and refuses to compile the specification.

**Type**

- Bug: function is under-defined and results incorrect values for some inputs.

**Additional context**

None.

**Requester**

- Ryan Scott (Galois)

**Method to check presence of bug**

The following two examples should produce correct C99 code, but don't:

```haskell
{-# LANGUAGE DataKinds #-}
module Main (main) where

import qualified Prelude as P

import Language.Copilot
import Copilot.Compile.C99

e :: Stream (Array 2 (Array 3 Int16))
e = extern "e" Nothing

spec :: Spec
spec = trigger "t" ((e .!! 0) .!! 1 == 42) [arg e]

main :: IO ()
main = do
  spec' <- reify spec
  compile "externNestedArrays" spec'
```

```haskell
{-# LANGUAGE DataKinds #-}
module Main (main) where

import qualified Prelude as P

import Language.Copilot
import Copilot.Compile.C99

data S = S { field :: Field "field" Int16 }

instance Struct S where
  typename _ = "s"
  toValues s = [Value Int16 (field s)]

instance Typed S where
  typeOf = Struct (S (Field 0))

e :: Stream (Array 2 S)
e = extern "e" Nothing

spec :: Spec
spec = trigger "t" ((e .!! 0)#field == 42) [arg e]

main :: IO ()
main = do
  spec' <- reify spec
  compile "externArrayOfStructs" spec'
```

Both cases result in similar error messages:
```
$ runghc ExternNestedArrays.hs
ExternNestedArrays.hs: Copilot error: The external symbol 'e' has been declared to have two different types!
CallStack (from HasCallStack):
  error, called at src/Copilot/Language/Error.hs:24:16 in copilot-language-3.12-inplace:Copilot.Language.Error
```

**Expected result**

The specs above should the compiled correctly by Copilot and generate valid C99 code.

**Desired result**

The specs above should the compiled correctly by Copilot and generate valid C99 code.

**Proposed solution**

Modify the instance of `TestEquality` in `copilot-core/src/Copilot/Core/Type.hs` to include cases for arrays and structs.

**Further notes**

None.

ivanperez-keera, 2023-01-05:
Technical Lead: Issue scheduled for fixing in Copilot 3.13.

Fix assigned to: @RyanGlScott .

RyanGlScott, 2023-01-05:
Implementor: Solution implemented, review requested.

ivanperez-keera, 2023-01-05:
Change Manager: Verified that:
- Solution is implemented:
  - [X] The code proposed compiles and passes all tests. Details:
    Build log: https://app.travis-ci.com/github/Copilot-Language/copilot/builds/259392071

  - [X] The solution proposed produces the expected result. Details:
    The following Dockerfile checks that an example with an extern with nested arrays, and one with an extern with an array of structs, can both be compiled and the resulting C code can also be compiled, in which case it prints the message `Success`.
    ```Dockerfile
      --- Dockerfile-verify-400
      FROM ubuntu:trusty
      
      RUN apt-get update
      
      RUN apt-get install --yes software-properties-common
      RUN add-apt-repository ppa:hvr/ghc
      RUN apt-get update
      
      RUN apt-get install --yes ghc-8.6.5 cabal-install-2.4
      RUN apt-get install --yes libz-dev
      
      ENV PATH=/opt/ghc/8.6.5/bin:/opt/cabal/2.4/bin:$PWD/.cabal-sandbox/bin:$PATH
      
      RUN cabal update
      RUN cabal v1-sandbox init
      RUN cabal v1-install alex happy
      RUN apt-get install --yes git
      
      ADD Arrays.hs /tmp/Arrays.hs
      ADD Structs.hs /tmp/Structs.hs
      
      SHELL ["/bin/bash", "-c"]
      CMD git clone $REPO && cd $NAME && git checkout $COMMIT && cd .. \
        && cabal v1-install $NAME/copilot**/ \
        && cabal v1-exec -- runhaskell /tmp/Arrays.hs \
        && gcc -Wall -c externNestedArrays.c \
        && cabal v1-exec -- runhaskell /tmp/Structs.hs \
        && gcc -Wall -c externArrayOfStructs.c \
        && echo "Success"
      
      --- Arrays.hs
      {-# LANGUAGE DataKinds #-}
      module Main (main) where
      
      import qualified Prelude as P
      
      import Language.Copilot
      import Copilot.Compile.C99
      
      e :: Stream (Array 2 (Array 3 Int16))
      e = extern "e" Nothing
      
      spec :: Spec
      spec = trigger "t" ((e .!! 0) .!! 1 == 42) [arg e]
      
      main :: IO ()
      main = do
        spec' <- reify spec
        compile "externNestedArrays" spec'
      
      --- Structs.hs
      {-# LANGUAGE DataKinds #-}
      module Main (main) where
      
      import qualified Prelude as P
      
      import Language.Copilot
      import Copilot.Compile.C99
      
      data S = S { field :: Field "field" Int16 }
      
      instance Struct S where
        typename _ = "s"
        toValues s = [Value Int16 (field s)]
      
      instance Typed S where
        typeOf = Struct (S (Field 0))
      
      e :: Stream (Array 2 S)
      e = extern "e" Nothing
      
      spec :: Spec
      spec = trigger "t" ((e .!! 0)#field == 42) [arg e]
      
      main :: IO ()
      main = do
        spec' <- reify spec
        compile "externArrayOfStructs" spec'
    ```
    Command:
    ```
    $ docker run -e "REPO=https://github.com/GaloisInc/copilot-1" -e "NAME=copilot-1" -e "COMMIT=8a6d32f31d5976eefd0a312f2f07d522c61e219e" -it copilot-verify-400
    ```
- [X] Implementation is documented. Details:
  No documentation needed. This change is internal (a fix to an underimplemented function).
- [X] Change history is clear.
- [X] Commit messages are clear.
- [X] Changelogs are updated.
- [X] Examples are updated. Details:
  No updates needed.
- [X] Required version bumps are evaluated. Details:
  Bump not needed. This change is internal (a fix to an underimplemented function).

ivanperez-keera, 2023-01-05:
Change Manager: Implementation ready to be merged.

commit 63589908 (2023-01-05): Merge branch 'develop-extern-nested-arrays'. Close #400.

**Description**

The implementation of type equality in `copilot-core` does not have cases for
arrays or structs, and thus assumes that nested arrays or arrays with struct
types are different (even if they are the exact same).

As a consequence of this lack, the C99 backend fails to complete. Since it
cannot tell that an extern with a nested array or array with structs is used
twice, it "assumes" that the spec tries to declare the same variable multiple
times different types and refuses to compile the specification.

**Type**

- Bug: fu [... cut]
closed 2023-01-05 by ivanperez-keera