Copilot-Language/copilot #401: Local stream of arrays produces C initializer that `gcc` rejects

RyanGlScott, 2023-01-03:
If you run this example:

```hs
{-# LANGUAGE DataKinds #-}
module Main (main) where

import qualified Prelude as P

import Language.Copilot
import Copilot.Compile.C99

arrs :: Stream (Array 2 Int16)
arrs = constant $ array [5,6]

spec :: Spec
spec = trigger "local" (local arrs $ \s -> (s .!! 0) == 5) []

main :: IO ()
main = do
  spec' <- reify spec
  compile "local" spec'
```

It will generate this C code:

```c
// local_types.h
```
```c
// local.h
void local(void);
void step(void);
```
```c
// local.c
#include <stdint.h>
#include <stdbool.h>
#include <string.h>
#include <stdlib.h>
#include <math.h>

#include "local_types.h"
#include "local.h"

bool local_guard(void) {
  int16_t local_0[(2)] = ((int16_t [(2)]){((int16_t)(5)), ((int16_t)(6))});
  return ((local_0)[(uint32_t)(0)]) == ((int16_t)(5));
}

void step(void) {
  if ((local_guard)()) {
    {(local)();}
  };
}
```

This will not compile when passed to `gcc`:

```
$ gcc local.c -c -o local.o
local.c: In function ‘local_guard’:
local.c:11:26: error: array initialized from non-constant array expression
   11 |   int16_t local_0[(2)] = ((int16_t [(2)]){((int16_t)(5)), ((int16_t)(6))});
      |                          ^
```

The error would go away if the following change were made to the generated code in `local.c`:

```diff
--- local.c     2023-01-02 10:02:46.365120252 -0600
+++ local_fixed.c       2023-01-02 10:03:02.919228741 -0600
@@ -8,7 +8,7 @@
 #include "local.h"
 
 bool local_guard(void) {
-  int16_t local_0[(2)] = ((int16_t [(2)]){((int16_t)(5)), ((int16_t)(6))});
+  int16_t local_0[(2)] = {((int16_t)(5)), ((int16_t)(6))};
   return ((local_0)[(uint32_t)(0)]) == ((int16_t)(5));
 }

```

This issue is similar to #276. While #276 specifically concerns global variables representing streams, this issue is about _temporary_ variables, which are created through different code paths than global variables.

fdedden, 2023-01-04:
Unfortunately C99 contains a some language features that share a similar syntax, but differ in their semantics. There is a mistake in the generated C code that mixes two of those. The comments here refer to the C99 committee draft (http://www.open-std.org/jtc1/sc22/wg14/www/docs/n1256.pdf), page numbers point to the printed numbering:
```C
int  a[2] =         {1,2}; // Initializer list: section 6.7.8, page 125.
int *b    = (int[2]){1,2}; // Compound literal for array: section 6.5.2.5, page 75.
```
The rational behind this is, the initializer syntax assigns two values to the block defined with `a[2]`. In contrary the compound literal syntax creates a block of memory and assigns the start of the block to `*b`.
In contrast the code written by Copilot is a mix, and is invalid.
```C
int c[2] = (int[2]){1,2}; // error: array initialized from non-constant array expression
```

Note that the `{}` in the initializer list _are not_ part of an expression, but `1` and `2` are expressions, that is `{1, 2}` is not an expression!
The compound literal `(int[2]){1,2}` however is an expression:
```C
int a =         {1,2}[0]; // Invalid, {1,2} is not an expression.
int b = (int[2]){1,2}[0]; // Valid, (int[2]){1,2} is an expression!
```

Note that the `()` notation differs from the unrelated typecast:
```C
int i = 10;
float f = (float)i; // section 6.5.4, page 81.
```

Copilot tries to initialize a value from a literal value, but also creates memory for this new piece of literal value. We have two options for fixing this:

1. Use the initializer list syntax.
2. Use a pointer instead of a proper array type for `local_0`.

RyanGlScott, 2023-01-04:
Good point—I had forgotten about C's compound literals. I agree that either option should suffice to fix the issue.

A handful of stray thoughts I had when reading this:

* Option 2 (the compound literal option) is interesting because we have to add a special case to give `local_0` a pointer type (e.g., `int *local_0 = ...`) instead of an array type (e.g., `int local_0[2] = ...`). Moreover, the special treatment isn't recursively applied. Namely, if you have a compound literal for a nested array like `(int[2][2][2]){{{1,2}, {3,4}}, {{5,6}, {7,8}}}`, then you would need to use the type `int (*local_0)[2][2] = ...`, not `int ***local_0`.
* Although option 2 requires some additional special treatment, it would have the nice benefit of fixing #402. This is because generator functions for arrays actually return pointers, so making local array variables have pointer types would make their types match up.
* We have a separate code path for global array variables that represent streams. Currently, global arrays use option 1 (the initializer list syntax option). Would we want global and local arrays to use the same option? It isn't strictly required to do so, but it might be nice to do for consistency's sake.

fdedden, 2023-01-04:
I implemented a prototype that uses the initializer list syntax: https://github.com/fdedden/copilot/commit/3623d2c2ccbb4f2185677602564873df3902eb63. I wouldn't trust it to not break other things though.

ivanperez-keera, 2023-01-05:
**Description**

Copilot's c99 backed is producing code for variable initializers when `local` is used is invalid. More specifically, the code produced for temporary variables of type array indicates their length in the type, but the value they are initialized to is a _compound literal_, which is not a valid initialization expression.

GCC rejects the generated code.

**Type**

- Bug: C99 backend produces invalid code.

**Additional context**

None.

**Requester**

- Ryan Scott (Galois)

**Method to check presence of bug**

The following Copilot spec:
```haskell
{-# LANGUAGE DataKinds #-}
module Main (main) where

import qualified Prelude as P

import Language.Copilot
import Copilot.Compile.C99

arrs :: Stream (Array 2 Int16)
arrs = constant $ array [5,6]

spec :: Spec
spec = trigger "local" (local arrs $ \s -> (s .!! 0) == 5) []

main :: IO ()
main = do
  spec' <- reify spec
  compile "local" spec'
```

Generates code that cannot be compiled by GCC:

```
$ gcc local.c -c -o local.o
local.c: In function ‘local_guard’:
local.c:11:26: error: array initialized from non-constant array expression
   11 |   int16_t local_0[(2)] = ((int16_t [(2)]){((int16_t)(5)), ((int16_t)(6))});
      |                          ^
```
**Expected result**

Using arrays in local definitions should generate valid code that either does not specify the length in the type, or does not add the type before the brackets (thus denoting a compound literal).

**Desired result**

Using arrays in local definitions should generate valid code that uses a pointer for the outermost type of the array.

**Proposed solution**

Modify the translation of `local` by the C99 backend such that it uses a pointer for the outermost type of an array.

**Further notes**

When a nested array is used, the inner lengths should still be made explicit, even if the "outer" length is not made explicit in the type and a pointer is used instead.

ivanperez-keera, 2023-01-05:
Technical Lead: Issue scheduled for fixing in Copilot 3.13.

Fix assigned to: @ivanperez-keera .

ivanperez-keera, 2023-01-07:
Implementor: Solution implemented, review requested.

ivanperez-keera, 2023-01-07:
Change Manager: Verified that:
- Solution is implemented:
  - [X] The code proposed compiles and passes all tests. Details:
    Build log: https://app.travis-ci.com/github/Copilot-Language/copilot/builds/259472863

  - [X] The solution proposed produces the expected result. Details:
    The following Dockerfile checks that a use of `local` with an array generates correct C99 code, in which case it prints the message `Success`:
    ```Dockerfile
      --- Dockerfile-verify-401
      FROM ubuntu:trusty
      
      RUN apt-get update
      
      RUN apt-get install --yes software-properties-common
      RUN add-apt-repository ppa:hvr/ghc
      RUN apt-get update
      
      RUN apt-get install --yes ghc-8.6.5 cabal-install-2.4
      RUN apt-get install --yes libz-dev
      
      ENV PATH=/opt/ghc/8.6.5/bin:/opt/cabal/2.4/bin:$PWD/.cabal-sandbox/bin:$PATH
      
      RUN cabal update
      RUN cabal v1-sandbox init
      RUN cabal v1-install alex happy
      RUN apt-get install --yes git
      
      ADD Local.hs /tmp/Local.hs
      
      SHELL ["/bin/bash", "-c"]
      CMD git clone $REPO && cd $NAME && git checkout $COMMIT && cd .. \
        && cabal v1-install $NAME/copilot**/ \
        && cabal v1-exec -- runhaskell /tmp/Local.hs \
        && gcc -Wall -c local.c \
        && echo "Success"
      
      --- context/Local.hs
      {-# LANGUAGE DataKinds #-}
      module Main (main) where
      
      import qualified Prelude as P
      
      import Language.Copilot
      import Copilot.Compile.C99
      
      arrs :: Stream (Array 2 Int16)
      arrs = constant $ array [5,6]
      
      spec :: Spec
      spec = trigger "local" (local arrs $ \s -> (s .!! 0) == 5) []
      
      main :: IO ()
      main = do
        spec' <- reify spec
        compile "local" spec'
    ```
    Command:
    ```
    $ docker run -e "REPO=https://github.com/ivanperez-keera/copilot" -e "NAME=copilot" -e "COMMIT=ab85ab0fbf84224edc59d19bf7c076e8cd63f488" -it copilot-verify-401
    ```
- [X] Implementation is documented. Details:
  A new function is introduced; the function has haddock documentation.
- [X] Change history is clear.
- [X] Commit messages are clear.
- [X] Changelogs are updated.
- [X] Examples are updated. Details:
  No updates needed.
- [X] Required version bumps are evaluated. Details:
  Bump not needed. This change is internal (bug fix in function in hidden module; new function not exposed).

ivanperez-keera, 2023-01-07:
Change Manager: Implementation ready to be merged.

commit b6e854fe (2023-01-07): Merge branch 'develop-local-array'. Close #401.

**Description**

Copilot's c99 backend is producing code for variable initializers when `local`
is used is invalid. More specifically, the code produced for temporary
variables of type array indicates their length in the type, but the value they
are initialized to is a _compound literal_, which is not a valid initialization
expression.

GCC rejects the generated code.

**Type**

- Bug: C99 backend produces invalid code.

**Additional context**

None.

**Requester**

- Ryan Scott (Galois)

**Method to check presence of bug**

The following Docker [... cut]
closed 2023-01-07 by ivanperez-keera