Copilot-Language/copilot #402: Local streams generate invalid C code when combined with generator functions

RyanGlScott, 2023-01-03:
This code:

```hs
{-# LANGUAGE DataKinds #-}
module Main (main) where

import qualified Prelude as P

import Language.Copilot
import Copilot.Compile.C99

arrs :: Stream (Array 2 Int16)
arrs = [array [5,6]] ++ arrs

spec :: Spec
spec = trigger "local_gen" (local arrs $ \s -> (s .!! 0) == 5) []

main :: IO ()
main = do
  spec' <- reify spec
  compile "local_gen" spec'
```

Generates the following C code:

```c
// local_gen_types.h
```
```c
// local_gen.h
void local_gen(void);
void step(void);
```
```c
// local_gen.c
#include <stdint.h>
#include <stdbool.h>
#include <string.h>
#include <stdlib.h>
#include <math.h>

#include "local_gen_types.h"
#include "local_gen.h"

static int16_t s1[(1)][(2)] = {{((int16_t)(5)), ((int16_t)(6))}};
static size_t s1_idx = (0);

int16_t * s1_get(size_t x) {
  return (s1)[((s1_idx) + (x)) % (1)];
}

int16_t * s1_gen(void) {
  return (s1_get)((0));
}

bool local_gen_guard(void) {
  int16_t local_0[(2)] = ((s1_get)((0)));
  return ((local_0)[(uint32_t)(0)]) == ((int16_t)(5));
}

void step(void) {
  int16_t s1_tmp[(2)];
  if ((local_gen_guard)()) {
    {(local_gen)();}
  };
  (memcpy)((s1_tmp), ((s1_gen)()), ((2) * (sizeof(int16_t))));
  (memcpy)(((s1)[s1_idx]), (s1_tmp), ((2) * (sizeof(int16_t))));
  (s1_idx) = (((s1_idx) + (1)) % (1));
}
```

This will not compile when passed to `gcc` or `clang`:

```
$ gcc local_gen.c -c -o local_gen.o
local_gen.c: In function ‘local_gen_guard’:
local_gen.c:22:26: error: invalid initializer
   22 |   int16_t local_0[(2)] = ((s1_get)((0)));
      |                          ^

$ clang local_gen.c -c -o local_gen.o
local_gen.c:22:11: error: array initializer must be an initializer list
  int16_t local_0[(2)] = ((s1_get)((0)));
          ^
1 error generated.
```

The error would go away if the following change were made to the generated code in `local_gen.c`:

```diff
--- local_gen.c 2023-01-01 22:29:45.256863934 -0600
+++ local_gen_fixed.c   2023-01-01 22:31:31.416553670 -0600
@@ -19,7 +19,7 @@
 }

 bool local_gen_guard(void) {
-  int16_t local_0[(2)] = ((s1_get)((0)));
+  int16_t * local_0 = ((s1_get)((0)));
   return ((local_0)[(uint32_t)(0)]) == ((int16_t)(5));
 }

```

This issue is similar to #401 in nature, although I believe the code path that generates the C code in #401 is a separate code path from what is causing this issue.

ivanperez-keera, 2023-01-08:
Change Manager: This error cannot be reproduced with the just released version Copilot 3.13. This is expected; based on discussions with @RyanGlScott and @fdedden , the conclusion is that the solution proposed for #401 also fixed #402.

@RyanGlScott please check on your side and, if you can't reproduce it either, go ahead and close. Thanks!

RyanGlScott, 2023-01-08:
Indeed, I cannot reproduce this after #404. Thanks!

closed 2023-01-08 by RyanGlScott