Copilot-Language/copilot #408: `majority` produces C code with unused local variables

RyanGlScott, 2023-01-11:
If you run this Copilot spec:

```hs
module Main where

import Language.Copilot
import Copilot.Compile.C99
import qualified Copilot.PrettyPrint as PP

spec :: Spec
spec = trigger "maj" (majority [true, true]) []

main :: IO ()
main = do
  spec' <- reify spec
  putStrLn $ PP.prettyPrint spec'
  compile "maj" spec'
```

It will generate the following C code:

```c
#include <stdint.h>
#include <stdbool.h>
#include <string.h>
#include <stdlib.h>
#include <math.h>

#include "maj_types.h"
#include "maj.h"

bool maj_guard(void) {
  bool local_0 = (false);
  bool local_1 = ((local_0) ? true : (true));
  uint32_t local_2 = ((uint32_t)(2));
  return local_1;
}

void step(void) {
  if ((maj_guard)()) {
    {(maj)();}
  };
}
```

Note that `local_2` is entirely unused. This will produce warnings when compiled with `-Wall`:

```
$ gcc -Wall maj.c -c -o maj.o
maj.c: In function ‘maj_guard’:
maj.c:13:12: warning: unused variable ‘local_2’ [-Wunused-variable]
   13 |   uint32_t local_2 = ((uint32_t)(2));
      |            ^~~~~~~
```

Note that this issue isn't the fault of the C backend, but rather, it's a consequence of the way that the spec is reified. This can be seen from the pretty-printed spec output that the Copilot program produces:

```
$ runghc Maj.hs 
trigger "maj" = local "local_0" = false
                in local "local_1" = (if var "local_0" then true else true)
                   in local "local_2" = 2
                      in var "local_1" [
  ]
```

RyanGlScott, 2023-01-11:
For the C99 backend, this is easy enough to work around: just don't pass `-Wall`. I originally encountered this issue while working on a different Copilot backend, however, where the compiler _always_ warns about unused local variables. This means that compiling any program that uses `majority` will produce warnings, and in the case of the `Engine` and `Voting` examples from `copilot/examples`, the number of warnings can quickly fill up your terminal window.

ivanperez-keera, 2023-03-09:
@RyanGlScott Would it be as simple as doing something like the following?

Note: I don't know entirely if this patch would be correct, but it produces a set without all those locals. I hope I haven't butchered the C99 backend:

```diff
diff --git a/copilot-c99/src/Copilot/Compile/C99/Compile.hs b/copilot-c99/src/Copilot/Compile/C99/Compile.hs
index c76d0bc2..fc80978b 100644
--- a/copilot-c99/src/Copilot/Compile/C99/Compile.hs
+++ b/copilot-c99/src/Copilot/Compile/C99/Compile.hs
@@ -6,8 +6,9 @@ module Copilot.Compile.C99.Compile
   ) where
 
 import Text.PrettyPrint     (render)
-import Data.List            (nub)
+import Data.List            (nub, union)
 import Data.Maybe           (catMaybes)
+import Data.Typeable        (Typeable)
 import System.Directory     (createDirectoryIfMissing)
 import System.Exit          (exitFailure)
 import System.FilePath      ((</>))
@@ -73,8 +74,9 @@ compile = compileWith mkDefaultCSettings
 -- * Generator functions for streams, guards and trigger arguments.
 -- * Declaration of the @step()@ function.
 compilec :: CSettings -> Spec -> C.TransUnit
-compilec cSettings spec = C.TransUnit declns funs
+compilec cSettings specO = C.TransUnit declns funs
   where
+    spec     = cleanSpec specO
     streams  = specStreams spec
     triggers = specTriggers spec
     exts     = gatherexts streams triggers
@@ -119,8 +121,9 @@ compilec cSettings spec = C.TransUnit declns funs
 
 -- | Generate the .h file from a 'Spec'.
 compileh :: CSettings -> Spec -> C.TransUnit
-compileh cSettings spec = C.TransUnit declns []
+compileh cSettings specO = C.TransUnit declns []
   where
+    spec     = cleanSpec specO
     streams  = specStreams spec
     triggers = specTriggers spec
     exts     = gatherexts streams triggers
@@ -195,3 +198,59 @@ compileTypeDeclns _cSettings spec = C.TransUnit declns []
 safeCRender :: C.TransUnit -> String
 safeCRender (C.TransUnit [] []) = ""
 safeCRender transUnit           = render $ pretty $ C.translate transUnit
+
+-- | Remove unused variables from a Spec.
+cleanSpec :: Spec -> Spec
+cleanSpec spec = spec
+  { specStreams  = fmap cleanStream (specStreams spec)
+  , specTriggers = fmap cleanTrigger (specTriggers spec)
+  }
+
+-- | Remove unused variables from a Stream.
+cleanStream :: Stream -> Stream
+cleanStream (Stream id buf expr ty) = Stream
+  { streamId       = id
+  , streamBuffer   = buf
+  , streamExpr     = cleanExpr expr
+  , streamExprType = ty
+  }
+
+-- | Remove unused variables from a Trigger.
+cleanTrigger :: Trigger -> Trigger
+cleanTrigger t = t
+  { triggerGuard = cleanExpr (triggerGuard t)
+  , triggerArgs  = fmap cleanUExpr (triggerArgs t)
+  }
+
+-- | Remove unused variables from a UExpr.
+cleanUExpr :: UExpr -> UExpr
+cleanUExpr (UExpr ty expr) = UExpr ty (cleanExpr expr)
+
+-- | Remove unused variables from an Expr.
+cleanExpr :: Typeable a => Expr a -> Expr a
+cleanExpr e = case e of
+  Local ta tb name exprLocal exprIn
+    | smaller1 <- cleanExpr exprLocal
+    , smaller2 <- cleanExpr exprIn
+    , name `elem` exprNames smaller1 || name `elem` exprNames smaller2
+    -> Local ta tb name smaller1 smaller2
+
+    | otherwise
+    -> cleanExpr exprIn
+
+  Op1 ty e1       -> Op1 ty (cleanExpr e1)
+  Op2 ty e1 e2    -> Op2 ty (cleanExpr e1) (cleanExpr e2)
+  Op3 ty e1 e2 e3 -> Op3 ty (cleanExpr e1) (cleanExpr e2) (cleanExpr e3)
+  Label ty n e    -> Label ty n (cleanExpr e)
+  _ -> e
+
+-- | List all variables in an expression, returns items uniquely.
+exprNames :: Typeable a => Expr a -> [String]
+exprNames e = case e of
+  Var _ n               -> [n]
+  Local ty1 ty2 _ e1 e2 -> exprNames e1 `union` exprNames e2
+  Op1 _ e1              -> exprNames e1
+  Op2 _ e1 e2           -> exprNames e1 `union` exprNames e2
+  Op3 _ e1 e2 e3        -> exprNames e1 `union` exprNames e2 `union` exprNames e3
+  Label _ _ e           -> exprNames e
+  _                     -> []
```

If that's correct, it might have to live somewhere else (`copilot-language`?).

There may be cases that I haven't considered. For example, should labels also be cleaned especially?

RyanGlScott, 2023-03-09:
`cleanExpr` is implementing a dead-code elimination pass for stream expressions, right? I believe that will work, although there are some parts of the implementation that I don't understand. For instance, do you ever need to check `` name `elem` exprNames smaller1 ``? I was under the impression that if you had `local "name" = exprLocal in ...`, then `name` could never appear in `exprLocal`.

I do wonder: rather than using dead-code elimination, could `majority` itself be tweaked to avoid generating dead local variables? Looking at the implementation of `majority` under the hood:

https://github.com/Copilot-Language/copilot/blob/34b454513d43db86fa1b0b1c648bf05a7b233949/copilot-libraries/src/Copilot/Library/Voting.hs#L50-L61

It looks like the issue arises when calling `inCan` with an empty `xs`. That will cause `local` to bind a local variable, which is named `cnt'` in the `inCnt` function. When `majority'` is recursively called with an empty `xs`, however, then the value of `cnt'` is never used.

To avoid this issue, what if `majority'` did this instead?

```diff
diff --git a/copilot-libraries/src/Copilot/Library/Voting.hs b/copilot-libraries/src/Copilot/Library/Voting.hs
index 84ebec19..d67c89e6 100644
--- a/copilot-libraries/src/Copilot/Library/Voting.hs
+++ b/copilot-libraries/src/Copilot/Library/Voting.hs
@@ -56,7 +56,10 @@ majority' (x:xs) can cnt =
   where
   inZero zero    = local (if zero then x else can) inCan
     where
-    inCan can'   = local (if zero || x == can then cnt+1 else cnt-1) inCnt
+    inCan can'   =
+        case xs of
+          [] -> can'
+          _  -> local (if zero || x == can then cnt+1 else cnt-1) inCnt
       where
       inCnt cnt' = majority' xs can' cnt'

```

This avoids calling `local` when `xs` is empty. I've confirmed that both the `maj` example in https://github.com/Copilot-Language/copilot/issues/408#issue-1529810135 and the `voting` example in https://github.com/Copilot-Language/copilot/issues/408#issuecomment-1379572568 compile warning-free with `gcc -Wall` with this change applied. (The dead-code elimination pass may be worth implementing separately, however.)

ivanperez-keera, 2023-03-09:
> do you ever need to check ```name `elem` exprNames smaller1```

You're right. The thought crossed my mind but I had not bothered to check. I just checked and it's not needed.

I thought perhaps it could be used to create some sort of recursion. `Copilot.Core` will accept it, but the C99 backend produces incorrect C code if it's used that way. The reification process in `copilot-language` does not allow for recursion in `local` definitions.

> could `majority` itself be tweaked to avoid generating dead local variables?

Simpler is better. Let's add a change to `Voting.hs`, and consider the dead code elimination bit later.

I'll file a "normative" issue.

ivanperez-keera, 2023-03-28:
Technical Lead: Confirmed that the issue should be addressed.

RyanGlScott, 2023-07-27:
I'll submit a patch for this tomorrow.

RyanGlScott, 2023-07-28:
Implementor: Solution implemented, review requested.

ivanperez-keera, 2023-08-10:
Change Manager: Verified that:
- Solution is implemented:
  - [X] The code proposed compiles and passes all tests. Details:
    Build log: https://app.travis-ci.com/github/Copilot-Language/copilot/builds/265134221

  - [X] The solution proposed produces the expected result. Details:
  
    The following Dockerfile checks that the implementation of `majority` produces code without unused local variables (it checks that it produces code that compiles with no warnings, and the GCC compiler installed by this docker script produces warnings by default for unused local variables):

    ```Dockerfile
    --- Dockerfile
    FROM ubuntu:trusty
    
    RUN apt-get update
    
    RUN apt-get install --yes software-properties-common
    RUN add-apt-repository ppa:hvr/ghc
    RUN apt-get update
    
    RUN apt-get install --yes ghc-8.6.5 cabal-install-2.4
    RUN apt-get install --yes libz-dev
    
    ENV PATH=/opt/ghc/8.6.5/bin:/opt/cabal/2.4/bin:$PWD/.cabal-sandbox/bin:$PATH
    
    RUN cabal update
    RUN cabal v1-sandbox init
    RUN apt-get install --yes git
    
    ADD Spec.hs /tmp/Spec.hs
    
    SHELL ["/bin/bash", "-c"]
    CMD git clone $REPO && cd $NAME && git checkout $COMMIT && cd .. \
      && cabal v1-install alex happy \
      && cabal v1-install $NAME/copilot**/ \
      && cabal v1-exec -- runhaskell /tmp/Spec.hs \
      && gcc -Werror -Wall maj.c -c -o maj.o \
      && echo "Success"

    --- Spec.hs
    module Main where
    
    import Language.Copilot
    import Copilot.Compile.C99
    import qualified Copilot.PrettyPrint as PP
    
    spec :: Spec
    spec = trigger "maj" (majority [true, true]) []
    
    main :: IO ()
    main =
      reify spec >>= compile "maj"
    ```
    Command (substitute variables based on new path after merge):
    ```sh
    $ docker run -e "REPO=https://github.com/GaloisInc/copilot-1" -e "NAME=copilot" -e "COMMIT=f07e2d4bc6911e1d3133636a72811a808db8fc5e" -it copilot-verify-408
    ```
- [X] Implementation is documented. Details:
  The API does not change, so no changes needed. The decision behind this implementation choice is documented in the code.
- [X] Change history is clear.
- [X] Commit messages are clear.
- [X] Changelogs are updated.
- [X] Examples are updated. Details:
  No updates needed.
- [X] Required version bumps are evaluated. Details:
  Bump not needed. This change is internal (optimization of code generated; no observable difference in behavior once the C code is compiled).

ivanperez-keera, 2023-08-10:
Change Manager: Implementation ready to be merged.

commit eeed69c3 (2023-08-10): Merge branch 'develop-majority'. Close #408.

**Description**

The function `Copilot.Library.Voting.majority'`, used to implement majority
voting, produces code that may potentially contain unused variables.

In particular, when the argument list of that function, with signature:

```haskell
majority' :: (P.Eq a, Typed a) => [Stream a] -> Stream a -> Stream Word32 -> Stream a
```

only has one element, Copilot will generate code with unused local variables.
This happens in almost every case, since the function is called recursively
with a smaller list at every step.

The C code generated by Co [... cut]
closed 2023-08-10 by ivanperez-keera