Copilot-Language/copilot #409: `copilot-core`: Remove module `Copilot.Core.PrettyDot`

ivanperez-keera, 2023-01-18:
**Description**

The module `Copilot.Core.PrettyDot` is not being used by any other part of Copilot.

It was deprecated in Copilot 3.11 and no messages have been received requesting that it be kept in the implementation.

As per our internal policy of waiting 3 versions from deprecation until a public interface declaration can be removed, this module can now be removed.

**Type**

- Bug: unused code included in the implementation.

**Additional context**

- Issue #359 , addressed in Copilot 3.11, deprecated the module.

**Requester**

- Ivan Perez

**Method to check presence of bug**

The following expression finds uses of `Copilot.Core.PrettyDot`, ignoring appearances in comments or strings.

```
$ grep -nHre '\<Copilot.Core.PrettyDot\>' --include='*.hs' | grep -ve '^{-#' | grep -ve '--.*Copilot.Core.PrettyDot' | grep -ve '"[^"]*Copilot.Core.PrettyDot[^"]*"'
copilot-core/src/Copilot/Core/PrettyDot.hs:8:module Copilot.Core.PrettyDot
```

The result obtained is just the module. There are no uses elsewhere.

**Expected result**

The string returned by the command above should be empty (module is not present) or list further uses. If the former, attempting to import `Copilot.Core.PrettyDot`, or any entities previously defined in it, should result in an error.

**Desired result**

The string returned by the command above should be empty. Attempting to import `Copilot.Core.PrettyDot`, or any entities previously defined in it, should result in an error.

**Proposed solution**

Remove the module `Copilot.Core.PrettyDot`.

Review the dependencies in case some are no longer needed.

**Further notes**

None.

ivanperez-keera, 2023-02-07:
Technical Lead: Issue scheduled for fixing in Copilot 3.14.

Fix assigned to: @willpog .

willpog, 2023-02-10:
Implementor: Solution implemented, review requested.

ivanperez-keera, 2023-02-18:
Change Manager: Verified that:
- Solution is implemented:
  - [X] The code proposed compiles and passes all tests. Details:
    Build log: https://app.travis-ci.com/github/Copilot-Language/copilot/builds/260644770

  - [X] The solution proposed produces the expected result. Details:
    The following Dockerfile checks that the module can no longer be imported, in which case it prints the message `Success`:
    ```Dockerfile
    FROM ubuntu:trusty
    
    RUN apt-get update
    
    RUN apt-get install --yes software-properties-common
    RUN add-apt-repository ppa:hvr/ghc
    RUN apt-get update
    
    RUN apt-get install --yes ghc-8.6.5 cabal-install-2.4
    RUN apt-get install --yes libz-dev
    
    ENV PATH=/opt/ghc/8.6.5/bin:/opt/cabal/2.4/bin:$PWD/.cabal-sandbox/bin:$PATH
    
    RUN cabal update
    RUN cabal v1-sandbox init
    RUN cabal v1-install alex happy
    RUN apt-get install --yes git
    
    SHELL ["/bin/bash", "-c"]
    CMD git clone $REPO && cd $NAME && git checkout $COMMIT && cd .. \
      && cabal v1-install copilot/copilot-core/ --enable-tests --run-tests \
      && ! cabal v1-exec -- runhaskell <<< "import Copilot.Core.PrettyDot; main :: IO (); main = return ()" \
      && echo "Success"
    ```
    Command (substitute variables based on new path after merge):
    ```
    $ docker run -e "REPO=https://github.com/willpog/copilot" -e "NAME=copilot" -e "COMMIT=12342630c85ce9a292b65bf23f90c7c24215932d" -it copilot-verify-409
    ```
- [X] Implementation is documented. Details:
  The documentation is removed with the module itself.
- [X] Change history is clear.
- [X] Commit messages are clear.
- [X] Changelogs are updated.
- [X] Examples are updated. Details:
  No updates needed.
- [X] Required version bumps are evaluated. Details:
  Bump needed (module removed).

ivanperez-keera, 2023-02-18:
Change Manager: Implementation ready to be merged.

commit 61df2b2f (2023-02-18): Merge branch 'develop-remove-prettydot'. Close #409.

**Description**

The module `Copilot.Core.PrettyDot` is not being used by any other part of
Copilot.

It was deprecated in Copilot 3.11 and no messages have been received requesting
that it be kept in the implementation.

As per our internal policy of waiting 3 versions from deprecation until a
public interface declaration can be removed, this module can now be removed.

**Type**

- Bug: unused code included in the implementation.

**Additional context**

- Issue 359, addressed in Copilot 3.11, deprecated the module.

**Requester**

- Ivan  [... cut]
closed 2023-02-18 by ivanperez-keera