Copilot-Language/copilot #412: `copilot-language`: Remove `Copilot.Language.prettyPrint`

ivanperez-keera, 2023-01-18:
**Description**

The function `Copilot.Language.prettyPrint` is not being used by any other part of Copilot.

It was deprecated in Copilot 3.11 and no messages have been received requesting that it be kept in the implementation.

As per our internal policy of waiting 3 versions from deprecation until a public interface declaration can be removed, this module can now be removed.

**Type**

- Bug: unused code included in the implementation.

**Additional context**

- Issue #362  , addressed in Copilot 3.11, deprecated the function.

**Requester**

- Ivan Perez

**Method to check presence of bug**

The function is exposed at the highest-level of the interface. The absence of dependencies does not indicate a bug in this case.

As per our proposed design, the function should not be accessible from the `Copilot.Language` module. Running the following should trigger an error:
```sh
$ ghci -e 'import Copilot.Language(prettyPrint)'
```

**Expected result**

Running the following should trigger an error:
```sh
$ ghci -e 'import Copilot.Language(prettyPrint)'
```

**Desired result**

Running the following should trigger an error:
```sh
$ ghci -e 'import Copilot.Language(prettyPrint)'
```

**Proposed solution**

Remove the function `Copilot.Language.prettyPrint` from `copilot-language`.

Review the dependencies in case some are no longer needed.

**Further notes**

Due to the possible reduction in dependencies, some Copilot packages may no longer be installed automatically when installing the package `copilot`. Care should be taken to add the necessary dependencies into the `copilot` library to make it install all packages in the project.

ivanperez-keera, 2023-02-07:
Technical Lead: Issue scheduled for fixing in Copilot 3.14.

Fix assigned to: @willpog .

willpog, 2023-03-06:
Implementor: Solution implemented, review requested.

ivanperez-keera, 2023-03-07:
Change Manager: Verified that:
- Solution is implemented:
  - [X] The code proposed compiles and passes all tests. Details:
    Build log: https://app.travis-ci.com/github/Copilot-Language/copilot/builds/261217774

  - [X] The solution proposed produces the expected result. Details:
    The following Dockerfile checks that the function can no longer be imported, in which case it prints the message `Success`:
    ```Dockerfile
    FROM ubuntu:trusty
    
    RUN apt-get update
    
    RUN apt-get install --yes software-properties-common
    RUN add-apt-repository ppa:hvr/ghc
    RUN apt-get update
    
    RUN apt-get install --yes ghc-8.6.5 cabal-install-2.4
    RUN apt-get install --yes libz-dev
    
    ENV PATH=/opt/ghc/8.6.5/bin:/opt/cabal/2.4/bin:$PWD/.cabal-sandbox/bin:$PATH
    
    RUN cabal update
    RUN cabal v1-sandbox init
    RUN cabal v1-install alex happy
    RUN apt-get install --yes git
    
    SHELL ["/bin/bash", "-c"]
    CMD git clone $REPO && cd $NAME && git checkout $COMMIT && cd .. \
      && cabal v1-install copilot/copilot-core/ copilot/copilot-interpreter/ copilot-prettyprinter/ copilot-theorem/ copilot-language/ --enable-tests --run-tests \
      && ! cabal v1-exec -- runhaskell <<< "import Copilot.Language (prettyPrint); main :: IO (); main = return ()" \
      && echo "Success"
    ```
    Command (substitute variables based on new path after merge):
    ```
    $ docker run -e "REPO=https://github.com/willpog/copilot" -e "NAME=copilot" -e "COMMIT=ead6ba56ea80cf95b226c7f99f87325213362298" -it copilot-verify-412
    ```
- [X] Implementation is documented. Details:
  A function is removed with its documentation.
- [X] Change history is clear.
- [X] Commit messages are clear.
- [X] Changelogs are updated.
- [X] Examples are updated. Details:
  No updates needed.
- [X] Required version bumps are evaluated. Details:
  Bump needed (public definition removed).

ivanperez-keera, 2023-03-07:
Change Manager: Implementation ready to be merged.

commit 7267735e (2023-03-07): Merge branch 'develop-remove-prettyprint'. Close #412.

**Description**

The function `Copilot.Language.prettyPrint` is not needed by any other
part of Copilot. Re-exporting a pretty-printer, which is a backend,
directly from copilot-language introduces an asymmetry in our design,
since other backends (e.g., C99) are not re-exported by
copilot-language. Consequently, Copilot.Language.prettyPrint was
deprecated in Copilot 3.11 and no messages have been received requesting
that it be kept in the implementation. As per our internal policy of
waiting 3 versions from deprecation until a public inte [... cut]
closed 2023-03-07 by ivanperez-keera