Copilot-Language/copilot #431: `copilot-c99`: Constant array streams result in undefined behavior when used as trigger guards or arguments

RyanGlScott, 2023-03-31:
In #386, it was noticed that `copilot-c99` would sometimes compile Copilot specifications involving streams of arrays into C code that exhibits undefined behavior. The test case in #386 only considered the case where the _generator_ function for a stream (e.g., `s0_gen`) produces an array value, but the same underlying issue also affects accessor functions (e.g., `f_arg0`) and guard functions (e.g., `f_guard`). Here is an example that shows both of these occurring with Copilot 3.14:

```hs
-- Array.hs
{-# LANGUAGE DataKinds #-}
module Main where

import Copilot.Compile.C99
import Language.Copilot
import qualified Prelude as P

spec :: Spec
spec = trigger "f" (stream .!! 0 >= 3) [arg stream]
  where
    stream :: Stream (Array 2 Int16)
    stream = constant (array [3,4])

main :: IO ()
main = reify spec >>= compile "array"
```

If you run this, it will generate the following C monitor files:

```c
// array_types.h
```
```c
// array.h
void f(int16_t f_arg0[(2)]);
void step(void);
```
```c
// array.c
#include <stdint.h>
#include <stdbool.h>
#include <string.h>
#include <stdlib.h>
#include <math.h>

#include "array_types.h"
#include "array.h"

bool f_guard(void) {
  return (((int16_t [(2)]){((int16_t)(3)), ((int16_t)(4))})[(uint32_t)(0)]) >= ((int16_t)(3));
}

int16_t * f_arg0(void) {
  return (int16_t [(2)]){((int16_t)(3)), ((int16_t)(4))};
}

void step(void) {
  int16_t * f_arg_temp0;
  if ((f_guard)()) {
    {(f_arg_temp0) = ((f_arg0)());
     (f)((f_arg_temp0));}
  };
}
```

Note that both `f_guard` and `f_arg0` attempt to return references to arrays on the stack, just like in #386. This is undefined behavior in C, and if you link this code against the following program:

```c
// array_driver.c
#include <stdint.h>
#include <stdio.h>

#include "array_types.h"
#include "array.h"

void f (int16_t f_arg0[(2)]) {
  printf("([%d,%d])\n", f_arg0[0], f_arg0[1]);
}

int main(void) {
  int i = 0;

  printf("f:\n");

  for (i=0; i<5; i++) {
    step();
  }
  return 0;
}
```

Then it will produce different results depending on the level of optimization:

```
$ gcc -O0 array_driver.c array.c -o main && ./main # Without optimizations
f:
([3,4])
([3,4])
([3,4])
([3,4])
([3,4])

$ gcc -O array_driver.c array.c -o main && ./main # With optimizations
f:
([21877,0])
([21877,0])
([21877,0])
([21877,0])
([21877,0])
```

The fix adopted in #386 was to change `copilot-c99`'s codegen to use `memcpy`, and I believe a similar fix would work here as well.

ivanperez-keera, 2023-08-10:
Hi @RyanGlScott . I'm confused about something in this issue. You say that "Note that both `f_guard` and `f_arg0` attempt to return references to arrays on the stack". However, `f_guard` returns a `bool`, so it's not really returning a reference to an array on the stack. It is using an array to build the return value, which is a boolean.

Would you consider that there's anything wrong with the implementation of `f_guard`?

Guards always take streams of booleans, so I don't know if this issue will manifest for guards. Is there a different spec that would generate more complex C code that would exhibit that bug also in the implementation of `f_guard`, or does this only manifest in the trigger arguments?

RyanGlScott, 2023-08-10:
Good catch, I must not have looked carefully enough at `f_guard`'s generated code. You are correct: `f_guard` is valid C99 code, as is simply indexing into a compound literal expression, and it doesn't return a reference to something allocated on the stack.

In that case, I think the only issue is in the generated code for `f_arg0`, which _does_ return a reference to a stack-allocated array. Here is a revised test case which tests this property alone:

```hs
-- Array.hs
{-# LANGUAGE DataKinds #-}
module Main where

import Copilot.Compile.C99
import Language.Copilot
import qualified Prelude as P

spec :: Spec
spec = trigger "f" true [arg stream]
  where
    stream :: Stream (Array 2 Int16)
    stream = constant (array [3,4])

main :: IO ()
main = reify spec >>= compile "array"
```

Which generates this C code:

```c
// array.c
#include <stdint.h>
#include <stdbool.h>
#include <string.h>
#include <stdlib.h>
#include <math.h>

#include "array_types.h"
#include "array.h"

bool f_guard(void) {
  return true;
}

int16_t * f_arg0(void) {
  return (int16_t [(2)]){((int16_t)(3)), ((int16_t)(4))};
}

void step(void) {
  int16_t * f_arg_temp0;
  if ((f_guard)()) {
    {(f_arg_temp0) = ((f_arg0)());
     (f)((f_arg_temp0));}
  };
}
```

That also exhibits undefined behavior with the same `array_driver.c` from https://github.com/Copilot-Language/copilot/issues/431#issue-1649623890:

```
$ gcc -O0 array_driver.c array.c -o main && ./main # Without optimizations
f:
([3,4])
([3,4])
([3,4])
([3,4])
([3,4])

$ gcc -O array_driver.c array.c -o main && ./main # With optimizations
f:
([32764,0])
([32764,0])
([32764,0])
([32764,0])
([32764,0])
```

ivanperez-keera, 2023-08-15:
**Description**

C99 backend generates incorrect code. In the generated code, the functions that generate values passed as arguments to trigger functions may return pointers to stack-allocated arrays, so the memory may be used for something else (or freed).

Instead, the structure of the implementation should guarantee that the arrays are always allocated.

**Type**

- Bug: generated C code may be incorrect.

**Additional context**

- Bug #386 was initially opened for the same reason, but only addressed the `_gen` functions generated by Copilot. Other functions generated by Copilot also return pointers to arrays and can fail for the same reason.

**Requester**

- Ryan Scott (Galois)

**Method to check presence of bug**

Given this Copilot spec:

```hs
--- Array.hs
{-# LANGUAGE DataKinds #-}
module Main where

import           Copilot.Compile.C99
import           Language.Copilot
import qualified Prelude             as P

spec :: Spec
spec = trigger "f" true [arg stream]
  where
    stream :: Stream (Array 2 Int16)
    stream = constant (array [3,4])

main :: IO ()
main = reify spec >>= compile "array"
```

If you link the generated code against the following program:

```c
// array_driver.c
#include <stdint.h>
#include <stdio.h>

#include "array_types.h"
#include "array.h"

void f (int16_t f_arg0[(2)]) {
  printf("([%d,%d])\n", f_arg0[0], f_arg0[1]);
}

int main(void) {
  int i = 0;

  printf("f:\n");

  for (i=0; i<5; i++) {
    step();
  }
  return 0;
}
```

Then it will produce different results depending on what optimization level you compile with. The observed behavior of the program should be independent of the optimization level.

```
$ gcc -O0 array_driver.c array.c -o main && ./main # Without optimizations
f:
([3,4])
([3,4])
([3,4])
([3,4])
([3,4])

$ gcc -O array_driver.c array.c -o main && ./main # Without optimizations
f:
([32767,0])
([32767,0])
([32767,0])
([32767,0])
([32767,0])
```

**Expected result**

The two executions above after compiling code with and without optimizations should return the same results.

**Desired result**

The two executions above after compiling code with and without optimizations should return the same results.

**Proposed solution**

Modify generated C99 backend so that the output array in a function producing values passed to triggers as arguments, as well as any they call that also produces arrays, take the returned array as input. That way, all memory is preallocated and there's no need for mallocs either (which we want to avoid).

**Further notes**

None.

ivanperez-keera, 2023-12-12:
Technical Lead: Bug scheduled for fixing in Copilot 3.18.

Fix assigned to: @RyanGlScott .

RyanGlScott, 2023-12-19:
Implementor: Solution implemented, review requested.

ivanperez-keera, 2023-12-20:
Change Manager: Verified that:
- Solution is implemented:
  - [X] The code proposed compiles and passes all tests. Details:
    Build log: https://app.travis-ci.com/github/Copilot-Language/copilot/builds/267952512

  - [X] The solution proposed produces the expected result. Details:
    The following Dockerfile checks that passing a stream of arrays as an argument to a trigger, when compiled with and without optimizations, produces the same results, in which case it prints the message `Success`. The docker image produces a diff of the two outputs if not.
    ```Dockerfile
    --- Dockerfile-verify-431
    FROM ubuntu:trusty
    
    RUN apt-get update
    
    RUN apt-get install --yes software-properties-common
    RUN add-apt-repository ppa:hvr/ghc
    RUN apt-get update
    
    RUN apt-get install --yes ghc-8.6.5 cabal-install-2.4
    RUN apt-get install --yes libz-dev
    
    ENV PATH=/opt/ghc/8.6.5/bin:/opt/cabal/2.4/bin:$PWD/.cabal-sandbox/bin:$PATH
    
    RUN cabal update
    RUN cabal v1-sandbox init
    RUN cabal v1-install alex happy
    RUN apt-get install --yes git
    
    ADD Array.hs /tmp/Array.hs
    ADD array_driver.c /tmp/array_driver.c
    
    SHELL ["/bin/bash", "-c"]
    CMD git clone $REPO && cd $NAME && git checkout $COMMIT && cd .. \
      && cabal v1-install $NAME/copilot**/ \
      && cabal v1-exec -- runhaskell /tmp/Array.hs \
      && gcc -O0 /tmp/array_driver.c array.c -I. -o main-no-op && ./main-no-op > no-optimizations \
      && gcc -O /tmp/array_driver.c array.c -I. -o main-op && ./main-op > with-optimizations \
      && diff no-optimizations with-optimizations \
      && echo "Success"
            
    --- Array.hs
    {-# LANGUAGE DataKinds #-}
    module Main where
    
    import           Copilot.Compile.C99
    import           Language.Copilot
    import qualified Prelude             as P
    
    spec :: Spec
    spec = trigger "f" true [arg stream]
      where
        stream :: Stream (Array 2 Int16)
        stream = constant (array [3,4])
    
    main :: IO ()
    main = reify spec >>= compile "array"      

    --- array_driver.c
    #include <stdint.h>
    #include <stdio.h>
    
    #include "array_types.h"
    #include "array.h"
    
    void f (int16_t f_arg0[(2)]) {
      printf("([%d,%d])\n", f_arg0[0], f_arg0[1]);
    }
    
    int main(void) {
      int i = 0;
    
      printf("f:\n");
    
      for (i=0; i<5; i++) {
        step();
      }
      return 0;
    }
    ```
    Command (adjust repo as necessary):
    ```
    $ docker run -e "REPO=https://github.com/GaloisInc/copilot-1" -e "NAME=copilot-1" -e "COMMIT=e975b4a0e007241f14e1c1032f46111ec4d69e83" -it copilot-verify-431
    ```
- [X] Implementation is documented. Details:
  Internal change. No changes to the public documentation are needed. Comments are added to new internal functions.
- [X] Change history is clear.
- [X] Commit messages are clear.
- [X] Changelogs are updated.
- [X] Examples are updated. Details:
  No updates needed.
- [X] Required version bumps are evaluated. Details:
  Bump not needed. This change is internal. The bug fix is in a function in a hidden module, the changes do not affect how to interface with the C code produced, and any new functions introduced are not exposed.

ivanperez-keera, 2023-12-20:
Change Manager: Implementation ready to be merged.

commit 4620006d (2023-12-20): Merge branch 'develop-fix-trigger-arg-arrays'. Close #431.

**Description**

Copilot's C99 backend generates incorrect code. In the generated code,
the functions that generate values passed as arguments to trigger
functions may return pointers to stack-allocated arrays, so the memory
may be used for something else (or freed).

Instead, the structure of the implementation should guarantee that the
arrays are always allocated.

**Type**

- Bug: generated C code may be incorrect.

**Additional context**

- Bug #386 was initially opened for the same reason, but only addressed
  the `_gen` function [... cut]
closed 2023-12-20 by ivanperez-keera