Copilot-Language/copilot #442: Paid for yearly copilot, but can't access it

quantumdump, 2023-06-06:
I can access my copilot from laptop (I was using it during trial), but when I tried to access copilot from my pc, it says that I don't have access to it. Same goes for my account. This account email is where all the letters from github were coming from so this is the account I bought yearly access with. How can I deal with this as soon as possible?

simtheverse, 2023-06-06:
This is not GitHub Copilot. This project is completely unrelated. Suggest reaching out to GitHub support.

quantumdump, 2023-06-06:
> This is not GitHub Copilot. This project is completely unrelated. Suggest reaching out to GitHub support.

I'm sorry Google gave me this as a third link about GitHub customer support. My bad. I was in a hurry

closed 2023-06-06 by ivanperez-keera