Copilot-Language/copilot #444: Copilot does not working

BehnazR, 2023-07-04:
Hello,

I installed the Copilot on my Mint system following the steps. The output of testing is:

GHCi, version 9.2.8: https://www.haskell.org/ghc/  :? for help
ghci> ghci> Leaving GHCi.

However, it seems it is not able to import other modules such as Copilot.Compile.C99, Copilot.Library.LTL, etc.
Could you help me with this issue please?

Thanks,

agoodloe, 2023-07-04:
Could you let us know the version of Cabal being used and the method used to install Copilot.That might help.

ivanperez-keera, 2023-07-04:
What you are seeing here is correct.

Try just running ghci without arguments and then importing the module once you are in the REPL.

```shell
$ ghci
> import Copilot.Compile.C99
```

BehnazR, 2023-07-05:
The version of Cabal is 3.6.2.0 and I use the following command as the instructions suggested:
cabal v2-install --lib copilot

I import the module in the REPL, however, I cannot make the .hs file to run. Have any idea what I should do?

BehnazR, 2023-07-05:
BTW, the GHC version I am using is 9.2.8 which should be fine.

ivanperez-keera, 2023-07-06:
Download this file:

https://raw.githubusercontent.com/Copilot-Language/copilot/master/copilot/examples/AddMult.hs

Save it in the same directory where you are running ghci from.

Start the repl and load that module with the `:load` command (`:l` for short):
```sh
Prelude> :l AddMult.hs
[1 of 1] Compiling Main             ( AddMult.hs, interpreted )
Ok, one module loaded.
*Main> main
```

Running main inside should show the values of that spec using the interpreter.

BehnazR, 2023-07-06:
So here is what I did in REPL to make it to work:
:set -package copilot-c99
:set -package copilot-language
And then I load the file created by OGMA and it worked.

I wish I could do something to avoid doing it every time, but I am happy it works anyway.

Thank you for your help!

ivanperez-keera, 2023-07-07:
Yes, that is ok. That file contains any custom data declarations needed. It is empty if none are needed.

Glad we figured it out!

closed 2023-07-07 by ivanperez-keera

Rezenders, 2024-12-04:
Hello,

Sorry for necrobumping, but I am facing the same issue described by @BehnazR. I didn't understand how @BehnazR got it to work.

I installed copilot then I tried running the [Heater](https://github.com/Copilot-Language/copilot/blob/master/copilot/examples/Heater.hs) example as described in the [Examples section](https://github.com/Copilot-Language/copilot?tab=readme-ov-file#examples) in the README as follows:

```Bash
$ runhaskell Heater.hs
Loaded package environment from /home/gus/.ghc/x86_64-linux-8.8.4/environments/default

Heater.hs:11:1: error:
    Could not load module ‘Copilot.Compile.C99’
    It is a member of the hidden package ‘copilot-c99-4.1’.
    You can run ‘:set -package copilot-c99’ to expose it.
    (Note: this unloads all the modules in the current scope.)
    Use -v (or `:set -v` in ghci) to see a list of the files searched for.
   |
11 | import Copilot.Compile.C99
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^

```

From what I understood from the previous posts, I need to start ghci, set the copilot packages, then load the Heater.hs file like:

```Bash
$ ghci
GHCi, version 8.8.4: https://www.haskell.org/ghc/  :? for help
Loaded package environment from /home/gus/.ghc/x86_64-linux-8.8.4/environments/default
Prelude> :set -package copilot-c99
package flags have changed, resetting and loading new packages...
Prelude> :set -package copilot-language
package flags have changed, resetting and loading new packages...
Prelude> :l Heater.hs 
[1 of 1] Compiling Main             ( Heater.hs, interpreted )
Ok, one module loaded.
*Main> 
```

From the ghci output, it seems that everything went well. However, no files were generated. What else do I need to do to generate the files?
I am sorry if this is a noob question, this is my very first time using Haskell.

## Extra info

OS: Ubuntu 22.04

Build method:
```Bash
$ apt install ghc cabal-install
$ cabal update
$ cabal v2-install --lib copilot
```

Build test:
```Bash
$ ghci <<< 'import Language.Copilot'
GHCi, version 8.8.4: https://www.haskell.org/ghc/  :? for help
Loaded package environment from /home/gus/.ghc/x86_64-linux-8.8.4/environments/default
Prelude> Prelude Language.Copilot> Leaving GHCi.
```

Cabal version:
```Bash
$ cabal --version
cabal-install version 3.0.0.0
compiled using version 3.0.1.0 of the Cabal library
```

ghci version:
```bash
$ ghci --version
The Glorious Glasgow Haskell Compilation System, version 8.8.4
```

Rezenders, 2024-12-04:
I managed to generate the files, I just had to run `:main` in ghci after loading Heater.hs.

```Bash
$ ghci
GHCi, version 8.8.4: https://www.haskell.org/ghc/  :? for help
Loaded package environment from /home/gus/.ghc/x86_64-linux-8.8.4/environments/default
Prelude> :set -package copilot-c99
package flags have changed, resetting and loading new packages...
Prelude> :set -package copilot-language
package flags have changed, resetting and loading new packages...
Prelude> :l Heater.hs 
[1 of 1] Compiling Main             ( Heater.hs, interpreted )
Ok, one module loaded.
*Main> :main
```

ivanperez-keera, 2024-12-04:
Hi @Rezenders . Yes, you normally need to run `main`.

Perhaps you could help us in https://github.com/Copilot-Language/copilot/discussions/543 to improve the installation instructions and initial instructions.

Also, if you are willing to upgrade Ubuntu, you can install copilot using `apt`. That will also keep it up-to-date with newer versions.

Rezenders, 2024-12-12:
Unfortunately, I can't upgrade Ubuntu as I need to have Ubuntu 22.04 to run ROS 2 Humble.
I posted a comment in the discussion y