Copilot-Language/copilot #452: `copilot-theorem`: README mentions inexisting `Driver.hs`

ivanperez-keera, 2023-08-16:
**Description**

`copilot-theorem`'s README mentions a file `Driver.hs` that is not part of the library:
```
As an example, the following prover is used in Driver.hs:
```
and later
```
Some examples are in the examples folder. The Driver.hs contains the main function to run any example. 
```

 These mentions should be adjusted or removed.

**Type**

- Bug: incorrect documentation.

**Additional context**

None.

**Requester**

- Max Fan (NASA)

**Method to check presence of bug**

Running a search through the tree brings up several mentions:
```
$ grep -nIHre 'Driver.hs'
copilot-theorem/README.md:205:As an example, the following prover is used in `Driver.hs`:
copilot-theorem/README.md:254:Some examples are in the *examples* folder. The `Driver.hs` contains the `main`
copilot-theorem/README.md:257:by changing one *import* directive in `Driver.hs`.
```

**Expected result**

The output of the above `grep` command should be empty, indicating that that inexisting file is not mentioned.

**Desired result**

The output of the above `grep` command should be empty, indicating that that inexisting file is not mentioned.

**Proposed solution**

The claims should be adjusted to match the existing file tree, either by updating the reference to point to their current locations, or by removing the claim altogether.

**Further notes**

None.

ivanperez-keera, 2023-10-29:
Change Manager: Confirmed that the README is outdated and mentions in-existing functions, modules and files.

ivanperez-keera, 2023-10-29:
Technical Lead: Confirmed that the issue should be addressed.

RyanGlScott, 2024-01-29:
I investigated these parts of `copilot-theorem` recently as well. I believe what the `README` refers to as `lightProver` has been superseded by the new `kInduction` prover, which is separate from the Kind2 and What4 provers. See also https://github.com/Copilot-Language/copilot/commit/976bc5f3c862bd36d67840741d675c0d208657d3, which removes `lightProver` in favor of `kInduction`.

fdedden, 2024-04-29:
I have picked up fixing the README.

@RyanGlScott yes, it seems that `lightProver` has been superseded by `kInduction` (and the code has been refactored a bit since then). `copilot-theorem` is pretty much the only part of Copilot where I have nearly no knowledge about. What is the difference between `kInduction`:
https://github.com/Copilot-Language/copilot/blob/f8239019b9c83315a78e2a92221b862945aa3b7c/copilot-theorem/src/Copilot/Theorem/Prover/SMT.hs#L110-L116

and `kind2Prover`:
https://github.com/Copilot-Language/copilot/blob/f8239019b9c83315a78e2a92221b862945aa3b7c/copilot-theorem/src/Copilot/Theorem/Kind2/Prover.hs#L44-L49

Sounds like they solve the same problem with a different backend?

The README is rather outdated and incomplete, so maybe we just want to remove all parts mentioning `lightProver` and do a complete rewrite later on?

RyanGlScott, 2024-04-29:
> Sounds like they solve the same problem with a different backend?

Yes, that is my understanding as well.

fdedden, 2024-04-29:
Yeah, I think we may want to revisit and clean the library. @ivanperez-keera I propose we simply remove any obsolete information from the README, so that it is at least not plain incorrect about things. Later on we can investigate if `kInduction` and `kind2Prover` are actually equal, and deprecate one of them (in the end both are based on Yices). After this (and possibly other cleanup work) we can rewrite the README. What do you think?

fdedden, 2024-05-01:
@ivanperez-keera I have made a branch (https://github.com/Copilot-Language/copilot/compare/master...fdedden:copilot:develop-fix-nonexisting-readme) that removes any mention to `Driver.hs` and the Light Prover. I have not tested the equivalence of the Light Prover with the k-induction prover yet, but it is a start. Let me know what you think.

BTW: I have seen numerous other typos and mistakes in the file, so it is in dire need of a cleanup.

ivanperez-keera, 2024-05-05:
Technical Lead: Issue scheduled for fixing in Copilot 3.20.

Fix assigned to: @fdedden .

fdedden, 2024-05-07:
Implementor: Solution implemented, review requested.

ivanperez-keera, 2024-05-08:
Change Manager: Verified that:
- Solution is implemented:
  - [X] The code proposed compiles and passes all tests. Details:
    Build log: https://app.travis-ci.com/github/Copilot-Language/copilot/builds/270352630
  - [X] The solution proposed produces the expected result. Details:
    The change is to the README, so it's been evaluated by manual inspection. I checked some of the examples with kind2-0.7.2 to make sure they produce the expected results.
- [X] Implementation is documented. Details:
  The change is to the documetnation.
- [X] Change history is clear.
- [X] Commit messages are clear.
- [X] Changelogs are updated.
- [X] Examples are updated. Details:
   No updates needed.
- [X] Required version bumps are evaluated. Details:
  Bump not needed. Changes affect README only.

ivanperez-keera, 2024-05-08:
Change Manager: Implementation ready to be merged.

commit 2fc1cbde (2024-05-08): Merge remote-tracking branch 'fdedden/develop-fix-nonexisting-readme'. Close #452.

**Description**

`copilot-theorem`'s README mentions a file `Driver.hs` that is not part
of the library:
```
As an example, the following prover is used in Driver.hs:
```
and later
```
Some examples are in the examples folder. The Driver.hs contains the
main function to run any example.
```

It also says that there are two provers: one based on `Kind2`, and one
called `lightProver`. The latter no longer exists, but we do have a
`What4` module now.

These mentions should be adjusted or removed.

**Type**

- Bug: [... cut]
closed 2024-05-08 by ivanperez-keera