Copilot-Language/copilot #469: copilot-language: test compile error

swt2c, 2023-11-14:
While upgrading to copilot 3.17, we noticed that `copilot-language`'s tests fail to compile with the below error:

```
Building test suite 'unit-tests' for copilot-language-3.17..
[1 of 3] Compiling Test.Extra       ( tests/Test/Extra.hs, dist-ghc/build/unit-tests/unit-tests-tmp/Test/Extra.o )
[2 of 3] Compiling Test.Copilot.Language.Reify ( tests/Test/Copilot/Language/Reify.hs, dist-ghc/build/unit-tests/unit-tests-tmp/Test/Copilot/Language/Reify.o )

tests/Test/Copilot/Language/Reify.hs:735:16: error:
    • Could not deduce (Copilot.Core.Type.Struct t)
        arising from a superclass required to satisfy ‘Typed t’,
        arising from a use of ‘observer’
      from the context: (Read t, Eq t, Typed t, Arbitrary t)
        bound by a pattern with constructor:
                   SemanticsP :: forall t.
                                 (Typeable t, Read t, Eq t, Show t, Typed t, Arbitrary t) =>
                                 (Stream t, [t]) -> SemanticsP,
                 in an equation for ‘checkSemanticsP’
        at tests/Test/Copilot/Language/Reify.hs:733:33-59
      Possible fix:
        add (Copilot.Core.Type.Struct t) to the context of
          the data constructor ‘SemanticsP’
    • In the expression: observer testObserverName expr
      In an equation for ‘spec’: spec = observer testObserverName expr
      In the expression:
        do let spec = observer testObserverName expr
           llSpec <- reify spec
           let trace = eval Haskell steps llSpec
           let expectation = take steps exprList
           ....
    |
735 |     let spec = observer testObserverName expr
    |                ^^^^^^^^
```
Unsure if this is possibly related to GHC 9.4 updates, or some other package updates in Debian?

Full compile logs here (for amd64): https://buildd.debian.org/status/fetch.php?pkg=haskell-copilot-language&arch=amd64&ver=3.17-1&stamp=1699932749&raw=0

Let me know if I can provide any additional info.

ivanperez-keera, 2023-11-14:
Thanks a lot! Will review it and get back to you.

swt2c, 2023-11-15:
Are you able to reproduce this one locally, or do you need any more info from me on it?

ivanperez-keera, 2023-11-15:
I managed to reproduce the problem.

swt2c, 2023-11-27:
Hey Ivan, just checking in to see if this was looking relatively easy to fix, or whether it might take a bit to sort out.  I think this is the last (known) blocker in getting copilot updated in Debian.  Obviously not the most urgent issue in the world, though.  :-)

ivanperez-keera, 2023-11-27:
Hi. I'm hoping we'll be able to fix this soon. I don't know if I'll be able to find time this week but I'll everything I can. If not, I'll prioritize fixing this by next week.

ivanperez-keera, 2023-11-28:
@swt2c I was able to find the solution. I'll try to push this soon.

If you want to backport this change to publish 3.17, here's the diff:
```patch
diff --git a/copilot-language/tests/Test/Copilot/Language/Reify.hs b/copilot-language/tests/Test/Copilot/Language/Reify.hs
index e056d946..d724df73 100644
--- a/copilot-language/tests/Test/Copilot/Language/Reify.hs
+++ b/copilot-language/tests/Test/Copilot/Language/Reify.hs
@@ -31,7 +31,7 @@ import qualified Copilot.Language.Operators.Integral as Copilot
 import qualified Copilot.Language.Operators.Mux      as Copilot
 import qualified Copilot.Language.Operators.Ord      as Copilot
 import           Copilot.Language.Reify              (reify)
-import           Copilot.Language.Spec               (observer)
+import           Copilot.Language.Spec               (Spec, observer)
 import           Copilot.Language.Stream             (Stream)
 import qualified Copilot.Language.Stream             as Copilot
 
@@ -732,7 +732,8 @@ semanticsShowK steps (SemanticsP (expr, exprList)) =
 checkSemanticsP :: Int -> [a] -> SemanticsP -> IO Bool
 checkSemanticsP steps _streams (SemanticsP (expr, exprList)) = do
     -- Spec with just one observer of one expression.
-    let spec = observer testObserverName expr
+    let spec :: Spec
+        spec = observer testObserverName expr
 
     -- Reified stream (low-level)
     llSpec <- reify spec
```

swt2c, 2023-11-28:
Thanks @ivanperez-keera I'll try putting that patch in tonight.

ivanperez-keera, 2023-11-28:
Technical Lead: Confirmed that the issue should be addressed.

swt2c, 2023-11-30:
Thanks @ivanperez-keera, I can confirm this patch gets copilot back to a green board in Debian (ignore mips64el).  :-)

[image]

ivanperez-keera, 2023-12-12:
Technical Lead: Bug scheduled for fixing in Copilot 3.18.

Fix assigned to: @ivanperez-keera.

ivanperez-keera, 2023-12-17:
Implementor: Solution implemented, review requested.

ivanperez-keera, 2023-12-17:
Change Manager: Verified that:
- Solution is implemented:
  - [X] The code proposed compiles and passes all tests. Details:
    Build log: https://app.travis-ci.com/github/Copilot-Language/copilot/builds/267903584
  - [X] The solution proposed produces the expected result. Details:
  The following Dockerfile checks that copilot-language compiles with GHC 9.4, in which case it prints the message Success:
    ```Dockerfile
    FROM ubuntu:focal
    
    RUN apt-get update
    
    RUN apt-get install --yes libz-dev
    RUN apt-get install --yes git
    
    RUN apt-get install --yes wget
    RUN mkdir -p $HOME/.ghcup/bin
    RUN wget https://downloads.haskell.org/~ghcup/0.1.19.2/x86_64-linux-ghcup-0.1.19.2 -O $HOME/.ghcup/bin/ghcup
    
    RUN chmod a+x $HOME/.ghcup/bin/ghcup
    ENV PATH=$PATH:/root/.ghcup/bin/
    ENV PATH=$PATH:/root/.cabal/bin/
    RUN apt-get install --yes curl
    RUN apt-get install --yes gcc make libgmp3-dev g++
    
    SHELL ["/bin/bash", "-c"]
    
    RUN ghcup install ghc 9.4.7
    RUN ghcup install cabal 3.8
    RUN ghcup set ghc 9.4.7
    RUN cabal update
    
    CMD git clone $REPO && \
        cd $NAME && \
        git checkout $COMMIT && \
        cd copilot-language && \
        cabal test . && \
        echo Success
    ```
    Command (substitute variables based on new path after merge): 
    ```
    docker run -e "REPO=https://github.com/ivanperez-keera/copilot" -e "NAME=copilot" -e "COMMIT=e25c7afaf743a5fdc7a9638516f59b006383bef0" -it copilot-verify-469
    ```
- [X] Implementation is documented. Details:
  Internal change; no changes to the documentation are needed, and a comment is added to ensure that future revisions do not remove the change accidentally.
- [X] Change history is clear.
- [X] Commit messages are clear.
- [X] Changelogs are updated.
- [X] Examples are updated. Details:
  No updates needed.
- [X] Required version bumps are evaluated. Details:
  Bump not needed (internal change to tests; does not affect API).

ivanperez-keera, 2023-12-17:
Change Manager: Implementation ready to be merged.

commit 60c44849 (2023-12-17): Merge branch 'develop-language-test-typeError'. Close #469.

**Description**

`copilot-language`'s tests fail to compile with GHC 9.4. Prior versions
sometimes made arbitrary decisions during type inference, whereas the
new type checker will simply reject such programs.

For example, compiling `copilot-language`'s tests with GHC 9.4.7 results
in the following error message:

```
Build profile: -w ghc-9.4.7 -O1
In order, the following will be built (use -v for more details):
 - copilot-language-3.17 (test:unit-tests) (first run)
Preprocessing test suite 'unit-tests' for copilot-language-3.17..
 [... cut]
closed 2023-12-17 by ivanperez-keera