Copilot-Language/copilot #471: `copilot-c99`: compiler warnings during tests

swt2c, 2023-11-15:
While building `copilot-c99` on Debian, we observed lots of these compiler warnings while running the tests:

```
main.c:9:23: warning: integer constant is so large that it is unsigned
    9 | uint64_t input_s[] = {10613134964919841005};
      |                       ^~~~~~~~~~~~~~~~~~~~
```

It seems that probably all that needs to be done is postfix uint64_t literals with 'u' to make the compiler aware that these are unsigned literals.

ivanperez-keera, 2023-11-15:
The way these values are printed is controlled by the following instances:

https://github.com/Copilot-Language/copilot/blob/cfc565bcf52dd359eb7ee768549459402c1785df/copilot-c99/tests/Test/Copilot/Compile/C99.hs#L884-L916

Are there any other instances apart from those for `uint64_t` that should be post-fixed?

swt2c, 2023-11-15:
> Are there any other instances apart from those for `uint64_t` that should be post-fixed?

I think it's only `uint64_t` as that seems to be what all the compiler warnings are referencing.  If you want me to try out a change, let me know.  (Sorry, my Haskell knowledge is VERY beginner.)

ivanperez-keera, 2023-11-27:
Pinging @fdedden for awareness.

ivanperez-keera, 2023-11-28:
**Description**

The code generated by `copilot-c99` during tests produces warnings during compilation. The reason is that some constants included in the code are too big and the compiler is making assumptions about their necessary types. If those constants are annotated, their intention will be made clearer.

**Type**

- Bug: test produce warnings during compilation.

**Additional context**

None.

**Requester**

- Scott Talbert (Debian Developer & Debian Haskell Group)

**Method to check presence of bug**

Compiling `copilot-c99`'s tests can produce warnings:

```
main.c:9:23: warning: integer constant is so large that it is unsigned
    9 | uint64_t input_s[] = {10613134964919841005};
      |                       ^~~~~~~~~~~~~~~~~~~~
```

**Expected result**

The above tests should compile without warnings.

**Desired result**

The above tests should compile without warnings.

**Proposed solution**

Modify the [`CShow` instances for different types](https://github.com/Copilot-Language/copilot/blob/cfc565bcf52dd359eb7ee768549459402c1785df/copilot-c99/tests/Test/Copilot/Compile/C99.hs#L884-L916) so that constants are printed with the necessary annotations.

**Further notes**

- Tests are randomly generated, so there's no guarantee that code with such warnings will be produced by a specific run.  This issue is therefore not always replicable unless we fix the number generator seed.

ivanperez-keera, 2023-11-28:
Change Manager: Confirmed that the issue manifests with GHC 9.4.

ivanperez-keera, 2023-12-12:
Technical Lead: Bug scheduled for fixing in Copilot 3.18.

Fix assigned to: @ivanperez-keera.

ivanperez-keera, 2023-12-18:
@swt2c I left a PR that you can check to see if it works for you. Please let me know before I merge it. Thanks!

ivanperez-keera, 2023-12-19:
Implementor: Solution implemented, review requested.

ivanperez-keera, 2023-12-19:
Change Manager: Verified that:
- Solution is implemented:
  - [X] The code proposed compiles and passes all tests. Details:
    Build log: https://app.travis-ci.com/github/Copilot-Language/copilot/builds/267932537
    
  - [X] The solution proposed produces the expected result. Details:
    The issue cannot be easily reproduced because it is dependent on the architecture and compiler. The correct behavior has been verified by asking the user who reported the issue to confirm that they no longer experience the problem with the solution proposed. See https://github.com/Copilot-Language/copilot/pull/479#issuecomment-1862077333 for details.
- [X] Implementation is documented. Details:
  Internal change. No changes to the public documentation are needed. The new code is documented.
- [X] Change history is clear.
- [X] Commit messages are clear.
- [X] Changelogs are updated.
- [X] Examples are updated. Details:
  No updates needed.
- [X] Required version bumps are evaluated. Details:
  Bump not needed (internal change to tests; does not affect API).

ivanperez-keera, 2023-12-19:
Change Manager: Implementation ready to be merged.

commit 907d5bbf (2023-12-19): Merge branch 'develop-test-integerSizes'. Close #471.

**Description**

The code generated by `copilot-c99` during tests produces warnings
during compilation. The reason is that some constants included in the
code are too big and the compiler is making assumptions about their
necessary types. If those constants are annotated, their intention will
be made clearer.

**Type**

- Bug: test produce warnings during compilation.

**Additional context**

None.

**Requester**

- Scott Talbert (Debian Developer & Debian Haskell Group)

**Method to check presence of bug**

Compiling `copilot-c99`'s tests [... cut]
closed 2023-12-19 by ivanperez-keera