Copilot-Language/copilot #497: ' ' is not a nondigit error

simondlevy, 2024-02-07:
I'm encountering a strange error message (appears to be a parsing failure) when I try to compile my Copilot code.  Here's a minimal reproducible [example](https://github.com/simondlevy/Copilot/blob/main/problem/Example.hs), but first I'll explain what I'm trying to do:

Following Example 1 on pp. 13-14 of the [tutorial](https://copilot-language.github.io/downloads/copilot_tutorial.pdf), I create a structure to represent the data coming from my intended C program, and I declare an instance of this structure as a Stream to support the interaction with C.  Then I mirror this structure in a datatype each of whose components is a Stream, so that I can write algorithms to work with the components individually, along with a "lifting" function to translate the first structure into the second.  So far so good; however, once the number of components grows large enough (three seems to be sufficient) I see the following when I call runhaskell on my example:

```
runhaskell Example.hs
Loaded package environment from /home/levys@ad.wlu.edu/.ghc/x86_64-linux-8.6.5/environments/default
Example.hs: ' ' is not a nondigit
CallStack (from HasCallStack):
  error, called at src/Language/C99/Util/Expr.hs:78:10 in language-c99-util-0.1.1-6f5020c06c32a7361dc125749bf511b92317cd868fa7d0bbd69d9cb1d86b8612:Language.C99.Util.Expr
```

I believe my approach is sound, but I would be happy to be shown a better way that avoids this error.

ivanperez-keera, 2024-02-07:
Tagging @fdedden for awareness.

ivanperez-keera, 2024-02-07:
I think the problem is that this line:

https://github.com/simondlevy/Copilot/blob/main/problem/Example.hs#L12C22-L12C26

Has a space in the field name `"y "`.

closed 2024-02-07 by simondlevy

ivanperez-keera, 2024-02-07:
No problem. I still think Copilot could deliver better error messages, but I'm glad we at least found the issue.

Something to think about for future releases.