Copilot-Language/copilot #514: `copilot-theorem`: Extend range of versions `what4` to include 1.6

ivanperez-keera, 2024-05-16:
**Description**

`what4` has seen a new release 1.6, but `copilot-theorem` needs versions strictly lower than 1.6. This version is API-compatible for all features used by `copilot-theorem`, but it includes support for GHC 9.8, which we want to also support.

**Type**

- Management: update versions of dependencies.

**Additional context**

None.

**Requester**

- Ryan Scott.

**Method to check presence of bug**

Not applicable (not a bug).

**Expected result**

Copilot can be installed with `what4-1.6`.
 
**Desired result**

Copilot can be installed with `what4-1.6`.

**Proposed solution**

Modify `copilot-theorem`'s cabal file to accept `what4` versions `>= 1.6` and `<1.7`.

Adjust `copilot-theorem` as needed to work with that version of `what4`.

**Further notes**

None.

fdedden, 2024-07-01:
I made a preliminary solution for this issue: https://github.com/fdedden/copilot/tree/develop-bump-what4-dependency.

ivanperez-keera, 2024-07-06:
Technical Lead: Issue scheduled for fixing in Copilot 3.20.

Fix assigned to: @ivanperez-keera .

ivanperez-keera, 2024-07-07:
Implementor: Solution implemented, review requested.

ivanperez-keera, 2024-07-07:
Change Manager: Verified that:
- Solution is implemented:
  - [X] The code proposed compiles and passes all tests. Details:
    Build log: https://app.travis-ci.com/github/Copilot-Language/copilot/builds/271304528

  - [X] The solution proposed produces the expected result. Details:
    The following Dockerfile installs `copilot` forcing the version of `what4` to be 1.6:
    ```Dockerfile
    FROM ubuntu:focal
    
    RUN apt-get update
    
    RUN apt-get install --yes libz-dev
    RUN apt-get install --yes git
    
    RUN apt-get install --yes wget
    RUN mkdir -p $HOME/.ghcup/bin
    RUN wget https://downloads.haskell.org/~ghcup/0.1.19.2/x86_64-linux-ghcup-0.1.19.2 -O $HOME/.ghcup/bin/ghcup
    
    RUN chmod a+x $HOME/.ghcup/bin/ghcup
    ENV PATH=$PATH:/root/.ghcup/bin/
    ENV PATH=$PATH:/root/.cabal/bin/
    RUN apt-get install --yes curl
    RUN apt-get install --yes gcc g++ make libgmp3-dev
    
    SHELL ["/bin/bash", "-c"]
    
    RUN ghcup install ghc 9.4
    RUN ghcup install cabal 3.8
    RUN ghcup set ghc 9.4.8
    RUN cabal update
    
    CMD git clone $REPO \
        && cd $NAME \
        && git checkout $COMMIT \
        && cabal install --lib copilot**/ \
             --constraint="what4>=1.6" \
        && echo Success
    ```
    Command (substitute variables based on new path after merge):
    ```
    $ docker run -e "REPO=https://github.com/ivanperez-keera/copilot" -e "NAME=copilot" -e "COMMIT=4cf2dce291b7e111f76cba26c89a39d1b197f34a" -it copilot-verify-514
    ```
- [X] Implementation is documented. Details:
  No updates needed.
- [X] Change history is clear.
- [X] Commit messages are clear.
- [X] Changelogs are updated.
- [X] Examples are updated. Details:
  No updates needed.
- [X] Required version bumps are evaluated. Details:
  Bump not needed (relaxes version constraints).

ivanperez-keera, 2024-07-07:
Change Manager: Implementation ready to be merged.

commit f5a96ceb (2024-07-07): Merge branch 'develop-bump-what4-dependency'. Close #514.

**Description**

`what4` has seen a new release 1.6, but `copilot-theorem` needs versions
strictly lower than 1.6. This version is API-compatible for all features
used by `copilot-theorem`, but it includes support for GHC 9.8, which we
want to also support.

**Type**

- Management: update versions of dependencies.

**Additional context**

None.

**Requester**

- Ryan Scott.

**Method to check presence of bug**

The following dockerfile installs copilot forcing the use of what4-1.6,
in which case it prints the message "Success".

```
FR [... cut]
closed 2024-07-07 by ivanperez-keera