Copilot-Language/copilot #525: `copilot-core`: Add documentation for `updateField` describing how (and why) to implement it

RyanGlScott, 2024-07-08:
[Copilot 3.20](https://github.com/Copilot-Language/copilot/releases/tag/v3.20) added support for struct updates via the new [`UpdateField`](https://hackage.haskell.org/package/copilot-core-3.20/docs/Copilot-Core-Operators.html#v:UpdateField) operation. One slightly non-obvious aspect of `UpdateField` is that `copilot-interpreter` doesn't support interpreting it unless you make sure to implement the [`updateField`](https://hackage.haskell.org/package/copilot-core-3.20/docs/Copilot-Core-Type.html#v:updateField) method in the `Struct` instance of any struct that whose field you update. For example, here is a naïve attempt at interpreting a Copilot specification involving a struct update:

```hs
{-# LANGUAGE DataKinds #-}
{-# LANGUAGE NoImplicitPrelude #-}
module Main (main) where

import Language.Copilot

data S = S
  { unS :: Field "unS" Word32
  }

instance Struct S where
  typeName _ = "s"
  toValues s = [Value typeOf (unS s)]

instance Typed S where
  typeOf = Struct (S (Field 0))

spec :: Spec
spec = do
  let externS :: Stream S
      externS = extern "extern_s" (Just [S (Field 0)])

      example :: Stream Word32
      example = (externS ## unS =: 42) # unS

  trigger "example" (example == example) [arg externS, arg example]

main :: IO ()
main = interpret 1 spec
```

This will fail with:

```
$ runghc UpdateFieldInterpreterFail.hs 
UpdateFieldInterpreterFail.hs: Field updates not supported for this type.
CallStack (from HasCallStack):
  error, called at src/Copilot/Core/Type.hs:64:17 in copilot-core-3.20-inplace:Copilot.Core.Type
```

To make this work, you need to implement `updateField` in the `Struct S` instance. Here is an example of how to do that:

```hs
{-# LANGUAGE DataKinds #-}
{-# LANGUAGE GADTs #-}
{-# LANGUAGE NoImplicitPrelude #-}
{-# LANGUAGE ScopedTypeVariables #-}
module Main (main) where

import Data.Proxy (Proxy(..))
import Data.Type.Equality (TestEquality(..), (:~:)(..))
import GHC.TypeLits (sameSymbol)
import qualified Prelude as P

import Language.Copilot

data S = S
  { unS :: Field "unS" Word32
  }

instance Struct S where
  typeName _ = "s"
  toValues s = [Value typeOf (unS s)]
  updateField s (Value fieldTy (field :: Field s a)) =
    case (sameSymbol (Proxy @s) (Proxy @"unS"), testEquality fieldTy Word32) of
      (Just Refl, Just Refl) -> s { unS = field }
      _ -> error $ "Unexpected field: " P.++ show field

instance Typed S where
  typeOf = Struct (S (Field 0))

spec :: Spec
spec = do
  let externS :: Stream S
      externS = extern "extern_s" (Just [S (Field 0)])

      example :: Stream Word32
      example = (externS ## unS =: 42) # unS

  trigger "example" (example == example) [arg externS, arg example]

main :: IO ()
main = interpret 1 spec
```
```
$ runghc UpdateFieldInterpreter.hs 
example:     
(<unS:0>,42)
```

The catch here is that I only knew how to do this because I was involved in the design of the `updateField` method. Unfortunately, there are currently no Haddocks for `updateField` itself, nor is there any kind of indication in the Copilot documentation that you would need to implement `updateField` like shown above in order to make `copilot-interpreter` work. Given how subtle this process is, I think there _should_ be some kind of tutorial in the documentation about how to do this.

It might also be helpful if the `Field updates not supported for this type` error message indicated that you need to implement `updateField`.

RyanGlScott, 2024-09-13:
Perhaps the best way to tutorialize this would be to extend the existing struct-related examples in the Copilot repo (e.g., [this one](https://github.com/Copilot-Language/copilot/blob/068c06dd7ab6e900e2e8728ecb1c3b6e94ba9ccb/copilot/examples/Structs.hs) and [this one](https://github.com/Copilot-Language/copilot/blob/068c06dd7ab6e900e2e8728ecb1c3b6e94ba9ccb/copilot/examples/what4/Structs.hs)) to demonstrate how to implement `updateField` for them.

(Note that if we implement https://github.com/Copilot-Language/copilot/issues/539, we might also want to update the examples to automatically derive `updateField` implementations rather than implement them explicitly. If that happens, it might also be worth having separate examples that show how to implement `updateField` manually rather than using deriving.)

ivanperez-keera, 2024-09-20:
**Description**

Copilot 3.20 added support for struct updates via a new `UpdateField` operation.

For struct updates to be usable in the interpreter, the creator of the spec must implement the `updateField` method in the `Struct` instance of any structs that need to be updated. The implementation of that function is non-trivial; we must include examples and documentation to help users understand how to write those definitions.

**Type**

- Feature: include additional documentation to aid understanding.

**Additional context**

None.

**Requester**

- Ryan Scott (Galois)

**Method to check presence of bug**

Not applicable (not a bug).

**Expected result**

There is documentation available to help users understand how to write those instances.

**Desired result**

There is documentation available to help users understand how to write those instances.

**Proposed solution**

Add an example that includes several structs that are updated.

Add haddocs for the type class documenting the implementation of the operation, possibly pointing to the example in the repository if the explanation is too long.

**Further notes**

None.

ivanperez-keera, 2024-09-27:
Technical Lead: Issue scheduled for fixing in Copilot 4.1.

Fix assigned to: @RyanGlScott .

RyanGlScott, 2024-09-30:
Implementor: Solution implemented, review requested.

ivanperez-keera, 2024-10-19:
Change Manager: Verified that:
- Solution is implemented:
  - [X] The code proposed compiles and passes all tests. Details:
    Build log: https://app.travis-ci.com/github/Copilot-Language/copilot/builds/272801025
  - [X] The solution proposed produces the expected result. Details:
    Evaluated by manual inspection. Running the following dockerfile checks that the new example and all examples modified continue running:
    ```Dockerfile
    FROM ubuntu:focal
    
    ENV DEBIAN_FRONTEND=noninteractive
    RUN apt-get update
    
    RUN apt-get install --yes libz-dev
    RUN apt-get install --yes git
    
    RUN apt-get install --yes wget
    RUN mkdir -p $HOME/.ghcup/bin
    RUN wget https://downloads.haskell.org/~ghcup/0.1.19.2/x86_64-linux-ghcup-0.1.19.2 -O $HOME/.ghcup/bin/ghcup
    
    RUN chmod a+x $HOME/.ghcup/bin/ghcup
    ENV PATH=$PATH:/root/.ghcup/bin/
    ENV PATH=$PATH:/root/.cabal/bin/
    RUN apt-get install --yes curl
    RUN apt-get install --yes gcc g++ make libgmp3-dev
    RUN apt-get install --yes pkg-config
    RUN apt-get install --yes z3
    
    SHELL ["/bin/bash", "-c"]
    
    RUN ghcup install ghc 9.4
    RUN ghcup install cabal 3.2
    RUN ghcup set ghc 9.4.8
    RUN cabal update
    
    CMD git clone $REPO \
        && cd $NAME \
        && git checkout $COMMIT \
        && cabal v1-sandbox init \
        && cabal v1-install alex happy --constraint='happy<2' \
        && cabal v1-install copilot**/ \
        && cabal v1-exec -- runhaskell copilot/examples/what4/Structs.hs \
        && cabal v1-exec -- runhaskell copilot/examples/Structs.hs \
        && cabal v1-exec -- runhaskell copilot/examples/StructsUpdateField.hs \
        && echo Success
    ```
    Command (substitute variables based on new path after merge):
    ```
    $ docker run -e "REPO=https://github.com/GaloisInc/copilot-1" -e "NAME=copilot-1" -e "COMMIT=75a3d9045dd6da368a2105ab8686129f71577075" -it copilot-verify-525
    ```
- [X] Implementation is documented. Details:
  The change is to the documentation of several modules and the examples. A new example is introduced and it is documented.
- [X] Change history is clear.
- [X] Commit messages are clear.
- [X] Changelogs are updated.
- [X] Examples are updated. Details:
  A new example is introduced, and other examples are updated to point to it as needed.
- [X] Required version bumps are evaluated. Details:
  No bump needed. Change is to the examples.

ivanperez-keera, 2024-10-19:
Change Manager: Implementation ready to be merged.

commit ae851b00 (2024-10-19): Merge branch 'develop-updateField-documentation'. Close #525.

**Description**

Copilot 3.20 added support for struct updates via a new `UpdateField`
operation.

For struct updates to be usable in the interpreter, the creator of the
spec must implement the `updateField` method in the `Struct` instance of
any structs that need to be updated. The implementation of that function
is non-trivial; we must include examples and documentation to help users
understand how to write those definitions.

**Type**

- Feature: include additional documentation to aid understanding.

**Additional context**

Non [... cut]
PR #547 (2024-10-19): Add Haddocks for `updateField` and describe how to implement it. Refs #525.
This adds an `examples/StructsUpdateField.hs` example that (1) describes how to implement `updateField` for some example `Struct` instances, and (2) demonstrates that `updateField` works by using it with the Copilot interpreter. This also adds Haddocks to `updateField` that briefly describes how one might implement `updateField` in custom `Struct` instances. Because the full details of implementing `updateField` are somewhat intricate, these Haddocks refer to the `copilot/examples/StructsUpdateField.hs` example for fur [... cut]
closed 2024-10-19 by ivanperez-keera