Copilot-Language/copilot #565: `copilot-core`: record fields of `UExpr` are unnecessary

ivanperez-keera, 2024-11-11:
**Description**

The record fields `uExprType` and `uExprExpr`, part of `Copilot.Core.Expr.UExpr`, are never really used within the module and only rarely used elsewhere in Copilot or, as far as we know, by any user of Copilot. As a matter of fact, we cannot use them to obtain the value inside the `UExpr` due to the existential it encloses.

This fields are not necessary parts of the interface to use Copilot, and not required by Copilot's requirements. Therefore, it would be suitable to remove that definitions from the module.

As per our internal policy of waiting 3 versions from deprecation until a public interface declaration can be removed, those definitions should be deprecated.

**Type**

- Bug: unused code included in the implementation.

**Additional context**

None.

**Requester**

- Ivan Perez

**Method to check presence of bug**

The field is mentioned in the record definition, but making it a non-record datatype, or not exporting that field in the module exports, does not change the behavior and does not otherwise affect Copilot.

The code compiles without errors and behaves in the same way.

**Expected result**

The following elements are deprecated:
- `Copilot.Core.Expr.UExpr.uExprExpr`
- `Copilot.Core.Expr.UExpr.uExprType`

**Desired result**

The following elements are deprecated:
- `Copilot.Core.Expr.UExpr.uExprExpr`
- `Copilot.Core.Expr.UExpr.uExprType`

**Proposed solution**

Deprecate the following functions:
- `Copilot.Core.Expr.UExpr.uExprExpr`
- `Copilot.Core.Expr.UExpr.uExprType`

Modify any uses of those fields anywhere in Copilot to use pattern matching instead.

**Further notes**

None.

ivanperez-keera, 2024-11-14:
Technical Lead: Issue scheduled for fixing in Copilot 4.2.

Fix assigned to: @kaBeech.

kaBeech, 2024-11-15:
Implementor: Solution implemented, review requested.

ivanperez-keera, 2024-12-18:
Change Manager: Verified that:
- Solution is implemented: 
  - [X] The code proposed compiles and passes all tests. Details:
    Build log: https://app.travis-ci.com/github/Copilot-Language/copilot/builds/273541802
  - [X] The solution proposed produces the expected result. Details:
    The following Dockerfile tries to import and use `uExprType`, and `uExprExpr`, printing "Success" if the imports are deprecated, and "Failure" if either of them is not deprecated:
    ```Dockerfile
    FROM ubuntu:focal
    
    ENV DEBIAN_FRONTEND=noninteractive
    RUN apt-get update
    
    RUN apt-get install --yes libz-dev
    RUN apt-get install --yes git
    
    RUN apt-get install --yes wget
    RUN mkdir -p $HOME/.ghcup/bin
    RUN wget https://downloads.haskell.org/~ghcup/0.1.19.2/x86_64-linux-ghcup-0.1.19.2 -O $HOME/.ghcup/bin/ghcup
    
    RUN chmod a+x $HOME/.ghcup/bin/ghcup
    ENV PATH=$PATH:/root/.ghcup/bin/
    ENV PATH=$PATH:/root/.cabal/bin/
    RUN apt-get install --yes curl
    RUN apt-get install --yes gcc g++ make libgmp3-dev
    RUN apt-get install --yes pkg-config
    
    SHELL ["/bin/bash", "-c"]
    
    RUN ghcup install ghc 9.8.2
    RUN ghcup install cabal 3.2
    RUN ghcup set ghc 9.8.2
    RUN cabal update
    
    SHELL ["/bin/bash", "-c"]
    CMD git clone $REPO && cd $NAME && git checkout $COMMIT && cd .. \
      && cabal v1-sandbox init \
      && cabal v1-install $NAME/copilot-core/ \
      && ! cabal v1-exec -- runhaskell -XGADTs -Wall -Werror <<< 'import Copilot.Core.Expr(UExpr(UExpr, uExprType),Expr(Const)); import Copilot.Core.Type(Type(Int8)); main :: IO (); main = case UExpr Int8 (Const Int8 0) of; (UExpr { uExprType = Int8}) -> putStrLn "Failure"' \
      && ! cabal v1-exec -- runhaskell -XGADTs -Wall -Werror <<< 'import Copilot.Core.Expr(UExpr(UExpr, uExprExpr),Expr(Const)); import Copilot.Core.Type(Type(Int8)); main :: IO (); main = case UExpr Int8 (Const Int8 0) of; (UExpr { uExprExpr = _}) -> putStrLn "Failure"' \
      && echo "Success"
    ```
    Command (substitute variables based on new path after merge):
    ```
    $ docker run -e "REPO=https://github.com/kaBeech/copilot" -e "NAME=copilot" -e  "COMMIT=4d8431d77632f4ed91cdab97bcc80f7331ce09cd" -it copilot-verify-565
    ```
- [X] Implementation is documented. Details:
   No need for additional documentation (deprecation).
- [X] Change history is clear.
- [X] Commit messages are clear.
- [X] Changelogs are updated.
- [X] Examples are updated. Details:
  No updates needed (deprecation).
- [X] Author is internal or has provided signed CLA.
- [X] Required version bumps are evaluated. Details:
  Major bump required. Public element being deprecated is an API-breaking change.

ivanperez-keera, 2024-12-18:
Change Manager: Implementation ready to be merged.

commit cdf807da (2024-12-18): Merge branch 'develop-deprecate-UExpr-fields'. Close #565.

**Description**

The record fields `uExprType` and `uExprExpr`, part of
`Copilot.Core.Expr.UExpr`, are never really used within the module and only
rarely used elsewhere in Copilot or, as far as we know, by any user of Copilot.
As a matter of fact, we cannot use them to obtain the value inside the `UExpr`
due to the existential it encloses.

This fields are not necessary parts of the interface to use Copilot, and not
required by Copilot's requirements. Therefore, it would be suitable to remove
that definitions from the module.

As per [... cut]
closed 2024-12-18 by ivanperez-keera