Copilot-Language/copilot #570: bounds issues made copilot drop out of ghc-9.10 Stackage Nightly

juhp, 2024-12-12:
- copilot < 0 # tried copilot-4.1, but its *library* requires filepath >=1.4 && < 1.5 and the snapshot contains filepath-1.5.2.0
        - copilot-c99 < 0 # tried copilot-c99-4.1, but its *library* requires filepath >=1.4 && < 1.5 and the snapshot contains filepath-1.5.2.0
        - copilot-language < 0 # tried copilot-language-4.1, but its *library* requires containers >=0.4 && < 0.7 and the snapshot contains containers-0.7
        - copilot-libraries < 0 # tried copilot-libraries-4.1, but its *library* requires containers >=0.4 && < 0.7 and the snapshot contains containers-0.7
        - copilot-theorem < 0 # tried copilot-theorem-4.1, but its *library* requires containers >=0.4 && < 0.7 and the snapshot contains containers-0.7
        - copilot-theorem < 0 # tried copilot-theorem-4.1, but its *library* requires data-default >=0.7 && < 0.8 and the snapshot contains data-default-0.8.0.0

kaBeech, 2024-12-13:
@ivanperez-keera I'm happy to fix this!

ivanperez-keera, 2024-12-14:
**Description**

In order to keep Copilot effectively working in the current Haskell ecosystem, as well as new versions of GHC, we need to extend the versions of dependencies that Copilot can be installed with. For example, `file` has seen releases in the 1.5.* series, but `copilot` and `copilot-c99` need versions strictly lower than 1.5. The same is true for `containers`, of which we depend on versions prior to 0.7 but version 0.7 has been released, and `data-default`, of which we depend on versions prior to 0.8 but version 0.8.0.0 has been released.

For those libraries that have introduced no breaking changes affecting Copilot, we should extend the version bounds.

**Type**

- Management: update versions of dependencies.

**Additional context**

None.

**Requester**

- Jens Petersen.

**Method to check presence of bug**

Not applicable (not a bug).

**Expected result**

Copilot can be installed with the most recent version of all dependencies published on hackage that do not require a change to the Haskell code.
 
**Desired result**

Copilot can be installed with the most recent version of all dependencies published on hackage that do not require a change to the Haskell code.

**Proposed solution**

Copilot can be installed with the most recent version of all dependencies published on hackage that do not require a change to the Haskell code.

**Further notes**

None.

ivanperez-keera, 2024-12-30:
Technical Lead: Thanks for volunteering to fix this! We have a release coming up very soon. @kaBeech If assigned to you, would you be able to complete this by this Wednesday and meet with me on Thursday if anything needs to be reviewed?

ivanperez-keera, 2025-01-02:
@kaBeech Gentle nudge :smiley:

kaBeech, 2025-01-02:
Goodness! My apologies @ivanperez-keera, I've been on vacation and didn't notice this. I'll start working on this now.

Happy New Year!

ivanperez-keera, 2025-01-02:
Technical Lead: Issue scheduled for fixing in Copilot 4.2.

Fix assigned to: @kaBeech .

kaBeech, 2025-01-02:
Built and tested everything locally, will use the verbiage that pings review once Travis confirms.

@ivanperez-keera I tried to match your commits in e.g. #354 but I'm always happy to meet with you!

kaBeech, 2025-01-03:
Implementor: fix implemented, review requested.

ivanperez-keera, 2025-01-03:
Change Manager: Verified that:
- Solution is implemented:
  - [X] The code proposed compiles and passes all tests. Details:
    Build log: https://github.com/Copilot-Language/copilot/runs/35130533360
  - [X] The solution proposed produces the expected result. Details:
    The following Dockerfile checks that Copilot can be installed with the recent versions of `filepath`, `containers` and `data-default`, as well as GHC (9.10 since 9.12 is not currently available in GHCUP), after which it prints the message "Success":
    ```Dockerfile
    FROM ubuntu:focal
    
    ENV DEBIAN_FRONTEND=noninteractive
    RUN apt-get update
    
    RUN apt-get install --yes libz-dev
    RUN apt-get install --yes git
    
    RUN apt-get install --yes wget
    RUN mkdir -p $HOME/.ghcup/bin
    RUN wget https://downloads.haskell.org/~ghcup/0.1.19.2/x86_64-linux-ghcup-0.1.19.2 -O $HOME/.ghcup/bin/ghcup
    
    RUN chmod a+x $HOME/.ghcup/bin/ghcup
    ENV PATH=$PATH:/root/.ghcup/bin/
    ENV PATH=$PATH:/root/.cabal/bin/
    RUN apt-get install --yes curl
    RUN apt-get install --yes gcc g++ make libgmp3-dev
    RUN apt-get install --yes pkg-config
    
    SHELL ["/bin/bash", "-c"]
    
    RUN ghcup install ghc 9.10
    RUN ghcup install cabal 3.2
    RUN ghcup set ghc 9.10
    RUN cabal update
    
    SHELL ["/bin/bash", "-c"]
    CMD git clone $REPO && cd $NAME && git checkout $COMMIT && cd .. \
      && cabal v1-sandbox init \
      && cabal v1-install alex happy --constraint='happy <= 2' \
      && cabal v1-install $NAME/copilot**/ --constraint='filepath>=1.5' --constraint='containers>=0.7' --constraint='data-default>=0.8'  \
      && echo "Success"
    ```
    Command (substitute variables based on new path after merge):
    ```
    $ docker run -e "REPO=https://github.com/kaBeech/copilot" -e "NAME=copilot" -e  "COMMIT=a31eb047e5d7e9d1f1d79d42898a667d05fd2389" -it copilot-verify-570
    ```
- [X] Implementation is documented. Details:
  No updates needed; change is to version numbers only.
- [X] Change history is clear.
- [X] Commit messages are clear.
- [X] Changelogs are updated.
- [X] Examples are updated. Details:
  No updates needed; change is to version numbers only.  
- [X] Required version bumps are evaluated. Details:
  Bump not needed; change extends, but does not restrict, the range of versions compatible with Copilot.

ivanperez-keera, 2025-01-03:
Change Manager: Implementation ready to be merged.

commit 8f783ee0 (2025-01-03): Merge remote-tracking branch 'kaBeech/develop-version-bounds'. Close #570.

**Description**

In order to keep Copilot effectively working in the current Haskell ecosystem,
as well as new versions of GHC, we need to extend the versions of dependencies
that Copilot can be installed with. For example, `filepath` has seen releases
in the 1.5.* series, but `copilot` and `copilot-c99` need versions strictly
lower than 1.5. The same is true for `containers`, of which we depend on
versions prior to 0.7 but version 0.7 has been released, and `data-default`, of
which we depend on versions prior to 0.8 b [... cut]
closed 2025-01-03 by ivanperez-keera